pauldemarco/flutter_blue #286: withDevices: in scan method

ericstephen, 2019-06-20:
If not a device uuid what is the purpose of this parameter? No matter what I put here the scan method will always scan for all devices. What am I misunderstanding?

bpillon, 2019-06-21:
Hello Eric.
For the moment, withDevices seems to be useless. You can look at the scan method implementation, the parameter withDevices is never user;

bpillon, 2019-06-21:
After a fast look at native documentations. It seems that iOS doesn't allow to scan only some devices (it's possible on Android). But we could may be implement a filter on the result on iOS to send to Flutter only the post filtered list.

ericstephen, 2019-06-21:
Yes, and I have been working on the iOS platform with no testing on Android at the moment. It would be nice to use this on iOS to filter out the noise of other bluetooth devices in the scan and I know I can add a service to the advertisement data, but I am unable to change that at the moment so was hoping for an alternative means to filter out the noise.