pauldemarco/flutter_blue #287: "Another scan is already in progress" exception (v.0.6.0+1)

diso73, 2019-06-21:
Hi,
I'm using:

```
_scanSubscription = _flutterBlue.scan(timeout: SCAN_TIMEOUT, withServices: withServices)
.listen((oneResult) {
        Utils.printDebug('BTDeviceConnection startScan: ScanResult received oneResult: device.id=' + oneResult.device.name);
        if (oneResult.device.id.id == deviceId) {
          // do stuffs with this device 
      _flutterBlue.stopScan();
      _scanSubscription.cancel();
        }
      }, onDone: () {
        Utils.printDebug('BTDeviceConnection startScan: onDone invoked');
      });
```

It seems to me tha the scan is never stopped before the timeout.
If I try to repeat this action before SCAN_TIMEOUT (20 seconds) it always causes: 

Unhandled Exception: Exception:  Another scan is already in progress.
 FlutterBlue.scan (package:flutter_blue/src/flutter_blue.dart:80:7)

I tried both stopScan() and/or _scanSubscription.cancel(), but no effect. 

Could you please help me?

Thanks

diso73, 2019-06-22:
I cannot let the scan to stop programmatically... Simply canceling the scan subscriptions seems not to be enough. Rolling back to 0.5 works fine.

My app shows known devices in a tab card widget and at each card-change I try to connect making a "scan_for_the_device_and_then_connect" process. Users can change cards quite quicly, so I can't wait the scan timeout before restarting the process, I need to force the scan to stop and then restart it.

Thank you

VaivanOmmeren, 2019-06-24:
stopScan() seems to work for me in (0.6.0 +1)
Have you tried calling the stopScan in the dispose of the card to ensure the scan get's stopped when a user switches card?

frey1esm, 2019-06-29:
Neither `_flutterBlue.stopScan()` nor `_scanSubscription.cancel()` are preventing this error for me on 0.6.0 +1.

`_Exception (Exception: Another scan is already in progress.)`

And according to [this issue](https://github.com/pauldemarco/flutter_blue/pull/17) `.stopScan()` shouldn't be used directly anyway.

bretie, 2019-07-18:
Same issue here.

raphael-bmec-co, 2019-09-26:
Same issue. +1 for fix proposed by @bretie

Madhukaran, 2022-11-09:
+1 is it resolved?