pauldemarco/flutter_blue #289: Over the Air (OTA) Device Firmware Updates (DFU)

LiveRock, 2019-06-27:
Over the Air (OTA) Device Firmware Updates (DFU)

Hi

Is there an example where I can use flutter_blue to perform OTA firmware updates to a device?

jthansen, 2019-06-27:
This is working for me on Android but I am having some difficulty with iOS. I believe it has to do with how iOS creates a UUID instead of using the MAC address like Android. Any 
ideas on how to fix that?  https://github.com/fengqiangboy/flutter-nordic-dfu

LiveRock, 2019-07-09:
Who has experience using the Cypress BLE chipset or BLE module?
https://www.cypress.com/part/cyble-214015-01

LiangLuDev, 2019-07-23:
I am the same question

pauloamaral93, 2019-10-31:
You can send the update firmware file in multiple parts, in my case my file has 740k bytes.
I divided into multiples parts of 448 bytes (payload) + protocol packets. You need to configure MTU mine is 500) and for > Android 8 devices, you need the Connection Priority. This OTA process in the best conditions, take me 3 minutes max.

WallySciAdmin, 2021-12-10:
i have ,bin file that has to be updated on nrf device using flutter.. can some one help me on this. 
The link below requires .zip file 
https://github.com/fengqiangboy/flutter-nordic-dfu

there is "nrfutil" tool to convert hex to .zip file in order to update but that is not working. 

nrf device manager facilitates to upload .bin file however the same needs to be used for custom app, how can i go about implementing the same..

MinhalWaheed2001, 2024-02-19:
Hi there, you can use flutter_ota package on pub dev : https://pub.dev/packages/flutter_ota

MinhalWaheed2001, 2024-03-02:
@faizannazir You can use the example app mentioned in the README.md file of flutter_ota package to test it. Already developed app using this package isnt available till now.