pauldemarco/flutter_blue #290: Less Devices Scanned than IOS BT Scan

sajal-sh, 2019-06-28:
Help: Why are less devices being searched than IOS BT Scan?
90% device names are not displayed?

Is anything related to Bluetooth LE?

alextran1502, 2019-07-01:
Have you tried turn on and off bluetooth on your Android phone? sometimes Android BLE chip has problem with retrieving scan results.

sajal-sh, 2019-07-01:
Actually I am using an iPad. (_No Android Device_)
1. Turn on BT from _Settings_.
2. Search/Scan -> Result is _n_ devices listed with proper names.

But if I use "_example_" app using this plugin. Press Find Devices.
1. significantly less BT devices are discovered.
2. only 1 or 2 get proper names. (Bose Headphones are the lucky ones).
3. All others get UUID sort of thing.

Basic question is still-> why less devices are being searched by app, in comparison to IOS settings BT Scan.

pauldemarco, 2020-04-02:
@sajal-sh Do you have any scan filters in place?