pauldemarco/flutter_blue #292: flutter run not working

dannyalbuquerque, 2019-07-01:
I created a simple flutter project and just added flutter_blue dependency.

flutter run is not working with an error on pod install because the path of !ProtoCompiler is wrong (missing the /ios/ folder)

```
Error output from CocoaPods:
↳
       /bin/bash: line 3: /Users/be.caresa/Workspace/Flutter/hrm_demo/Pods/!ProtoCompiler/protoc: No
       such file or directory 
```

I can run the app using XCode and Runner.xcworkspace

aitch-3, 2019-07-08:
I get the same error message, 
change to `0.5.0` works fine

FWeissenb, 2019-07-11:
Do you have a "normal" flutter app? Or have you migrated to androidX?

dannyalbuquerque, 2019-07-11:
@aitch-3 Same error with 0.5.0
@FWeissenb  I have a "normal" flutter app

I use https://pub.dev/packages/flutter_bluetooth_serial instead of flutter_blue for the moment

FWeissenb, 2019-07-11:
flutter_blue > 0.5 is androidX afaik.

Do you use Bluetooth classic or Bluetooth low energy? I think flutter_blue
is BLE only and fluuter_bluetooth_serial is bluetooth classic

FWeissenb, 2019-07-11:
0.5.0 # <https://pub.dev/packages/flutter_blue/versions/0.5.0#050>

   - *Breaking change*. Migrate from the deprecated original Android
   Support Library to AndroidX. This shouldn't result in any functional
   changes, but it requires any Android apps using this plugin to also
   migrate <https://developer.android.com/jetpack/androidx/migrate> if
   they're using the original support library.

sorry. 0.5.0 is also androidx. So try to use a lower version (0.4.2+1) or
upgrade your app to androidx
https://flutter.dev/docs/development/packages-and-plugins/androidx-compatibility

frank.weissenborn@gmail.com>:

dannyalbuquerque, 2019-07-11:
> […](#)

I need BLE. For the moment, I'm trying to run flutter_blue only on iOS.

dannyalbuquerque, 2019-07-11:
I can run it using this:
```
  flutter_blue:      
    git:
       url: https://github.com/pauldemarco/flutter_blue.git
       ref: 38ae679fbf445fecac36bc5d45265f946e0990d7
```
I needed[ this fix ](https://github.com/pauldemarco/flutter_blue/commit/38ae679fbf445fecac36bc5d45265f946e0990d7)but it was reverted https://github.com/pauldemarco/flutter_blue/commit/38ae679fbf445fecac36bc5d45265f946e0990d7

closed 2019-07-11 by dannyalbuquerque

TimmyGzw, 2019-07-13:
@dannyalbuquerque thank you!!