pauldemarco/flutter_blue #295: PlatformException(set_notification_error, error when writing the descriptor, null) on setNotifyValue

rma6, 2019-07-03:
I'm trying to turn on notifications for a characteristic, but every time that I run `characteristic(true)` I get this error `PlatformException(set_notification_error, error when writing the descriptor, null)`. Here's a piece of my code:

```
BluetoothCharacteristic ss;
List<BluetoothService> services = await connectedDevice.discoverServices();
        services.forEach((service) {
          if(service.uuid.toString().compareTo("0d3448ae-12ec-4eee-8adf-e4330cfcfd5d") == 0)
          {
            service.characteristics.forEach((c) {
              if(c.uuid.toString().compareTo("31b3a584-9c79-438d-8b41-c033a9089e8b") == 0)
              {
                ss = c;
              }
            });
          }
        });
await ss.setNotifyValue(true);
ss.value.listen((value) {
_status = value.elementAt(0);
});
```

andretietz, 2019-07-16:
I am experiencing this too, but only with a characteristic that is not readable. 
Here's what I mean:
I can setNotifyValue to true here, without a problem:
[image]
But here, I am getting exactly the same error as @rma6.
[image]

It seems like a bug.

aytunch, 2019-11-05:
@rma6 most probably
`await ss.setNotifyValue(true);`
runs before you finish iterating through the services/characteristics. If you run this it would work:
```
BluetoothCharacteristic ss;
List<BluetoothService> services = await connectedDevice.discoverServices();
        services.forEach((service) {
          if(service.uuid.toString().compareTo("0d3448ae-12ec-4eee-8adf-e4330cfcfd5d") == 0)
          {
            service.characteristics.forEach((c) {
              if(c.uuid.toString().compareTo("31b3a584-9c79-438d-8b41-c033a9089e8b") == 0)
              {
                ss = c;
                await ss.setNotifyValue(true);
                ss.value.listen((value) {
                _status = value.elementAt(0);
              }
            });
          }
        });
```

});

boskokg, 2020-05-13:
@aytunch I think that your code example is wrong. You should not call await inside forEach method

itsgk92, 2020-05-13:
The issue is solved if we have a delay between consecutive setNotifyValue.

jiyarong, 2020-06-13:
> The issue is solved if we have a delay between consecutive setNotifyValue.

could you paste your code, i have the same problem, thank you

saramonteiro, 2020-06-26:
@aytunch The snippet you provided did not solve the problem

itsgk92, 2020-06-26:
In the example above, await inside a forEach doesn't work. Separate discovery of service/characteristics and setNotifyValue call rather than making it in a single loop.

For example 

  ```
 accelerometerCharacteristic = _getAccelerometerCharacteristic(sensorService);
      gyroscopeCharacteristic = _getGyroscopeCharacteristic(sensorService);

if(accelerometerCharacteristic` == null || gyroscopeCharacteristic == null){

      return false;
    }else {
      bool accelerometerSubscribed;
      bool gyroscopeSubsribed;

      if (!accelerometerCharacteristic.isNotifying) {
        accelerometerSubscribed =
        await accelerometerCharacteristic.setNotifyValue(true);
      }

      await Future.delayed(const Duration(milliseconds: 500));

      if (!gyroscopeCharacteristic.isNotifying) {
        gyroscopeSubsribed = await gyroscopeCharacteristic.setNotifyValue(true);
      }
```

saramonteiro, 2020-06-26:
@itsgk92 I did like this, and it continues throwing the exception:

`
  Future\<void\> _connect (BluetoothDevice dev) async {

    String envSensServiceUUID = '0000181a-0000-1000-8000-00805f9b34fb';
    String tempCharUUID = '00002a6e-0000-1000-8000-00805f9b34fb';
    List<int> tempValue;
    List<BluetoothCharacteristic> characteristics;
    BluetoothCharacteristic c, MyChar;
    BluetoothService s, Myservice;
    bool tempCharSubscribed = false;
    await dev.connect();
    print("CONNECTED");

    List<BluetoothService> services = await dev.discoverServices();

    for (s in services) {
      if (s.uuid.toString() == envSensServiceUUID) {
        Myservice = s;
      }
    }
    for (c in Myservice.characteristics) {
      if (c.uuid.toString() == tempCharUUID) {
        MyChar = c;
      }
    }
      tempCharSubscribed = await MyChar.setNotifyValue(true);
 }
`

maqoo, 2020-08-13:
For me a synchronized call to `setNotifyValue` sufficed. I used [synchronized](https://pub.dev/packages/synchronized) package to achive that:

`await _lock.synchronized(
      () async {
        await characteristic.setNotifyValue(true);
      },
    );`

maks, 2021-02-08:
I think this issue can be closed because its working as expected in current version `0.7.3` as long as you `await`  calls to `setNotifyValue(true)` *AND* as per the previous comment, the key is to *not* use async `forEach()`! 

A very good explaination as to why async `forEach()` causes a problem [is documented in this SO question](https://stackoverflow.com/questions/51106934/my-async-call-is-returning-before-list-is-populated-in-foreach-loop).

So for example I used `await Future.forEach()` instead.

jimykhan, 2021-02-20:
Hi guys i face this for two days this is my solution it work for me i hope it will solve ur problem i think we need await every function return future 

discoverService(List<BluetoothDevice> device){
    device.forEach((d) async {
      List<BluetoothService> ser = await d.discoverServices();
      print("this is llist of servieces");
      for(BluetoothService service in ser){
        for(BluetoothCharacteristic chr in service.characteristics){
          if(chr.properties.notify){
            await chr.setNotifyValue(true);
            print("this notify bit now after setNotifiction true ${chr.isNotifying}");
            chr.value.listen((event) {
              print("this is last value of chr = ${event.toString()}");
            });
            await Future.delayed(Duration(seconds: 3));

            print("${chr.uuid}");
          }
        }
      }
      print("this is llist of servieces");
    });}

yuriboeira11tx, 2021-05-17:
> For me a synchronized call to `setNotifyValue` sufficed. I used [synchronized](https://pub.dev/packages/synchronized) package to achive that:
> 
> ```
> await _lock.synchronized(() async {
>         await characteristic.setNotifyValue(true);
>       },
>     );
> ```
> 
> Also funny thing is that the exception only happend on Android.

It's works!

stefan-sherwood, 2023-05-23:
The fact that adding a synchronized call suggests that you were making a second  `setNotifyValue( true )` call before the first one had completed. Note that the conditional `!gyroscopeCharacteristic.isNotifying()` doesn't prevent multiple simultaneous asynchronous calls since the second one starts before the first has completed.

In @saramonteiro 's example, this should work:

```
bool subscribeAlreadyCalled = false;

Future<void> _connect (BluetoothDevice dev) async {
	
	String envSensServiceUUID = '0000181a-0000-1000-8000-00805f9b34fb';
	String tempCharUUID = '00002a6e-0000-1000-8000-00805f9b34fb';
	List<int> tempValue;
	List<BluetoothCharacteristic> characteristics;
	BluetoothCharacteristic c, MyChar;
	BluetoothService s, Myservice;
	bool tempCharSubscribed = false;
	await dev.connect();
	print("CONNECTED");
	
	List<BluetoothService> services = await dev.discoverServices();
	
	for (s in services) {
	  if (s.uuid.toString() == envSensServiceUUID) {
	    Myservice = s;
	  }
	}
	for (c in Myservice.characteristics) {
	  if (c.uuid.toString() == tempCharUUID) {
	    MyChar = c;
	  }
	}
	
	if (!subscribeAlreadyCalled) {
	  subscribeAlreadyCalled = true;
	  tempCharSubscribed = await MyChar.setNotifyValue(true);
	}
}

```

shubhamsinghmutualmobile, 2023-06-29:
> @rma6 most probably `await ss.setNotifyValue(true);` runs before you finish iterating through the services/characteristics. If you run this it would work:
>
> ```
> BluetoothCharacteristic ss; [...]

This helped me 👍 Thank you

MO-Lewis, 2024-03-21:
For anyone looking into this in the future, I decided to switch this library out (3 years out of date now) for a fork of this one, called [flutter blue plus](https://github.com/boskokg/flutter_blue_plus).

It'll take some rewriting in your code to get set up with, but this library is _**much**_ more reliable than the latest release of the original Flutter Blue library.