pauldemarco/flutter_blue #299: How to handle errors

FWeissenb, 2019-07-05:
I tried to catch a Timeout Exception from FlutterBlue connect. But it doesn't work.

I tried:

```
try {
      await scanResult.device.connect(timeout: Duration(seconds: 5));
    } on TimeoutException catch (e) {
      print(debugPrefix + " $e");
    }
```

And I tried

```
await r.device.connect(timeout: Duration(seconds: 5)).catchError((e){

       print(debugPrefix + " $e");

     });
```

All I get is the message in debug console and my app stops :/

```
E/flutter (11807): [ERROR:flutter/lib/ui/ui_dart_state.cc(148)] Unhandled Exception: TimeoutException after 0:00:05.000000: Failed to connect in time.
E/flutter (11807): #0      BluetoothDevice.connect.<anonymous closure> 
package:flutter_blue/src/bluetooth_device.dart:33
E/flutter (11807): #1      _rootRun  (dart:async/zone.dart:1120:38)
E/flutter (11807): #2      _CustomZone.run  (dart:async/zone.dart:1021:19)
E/flutter (11807): #3      _CustomZone.runGuarded  (dart:async/zone.dart:923:7)
E/flutter (11807): #4      _CustomZone.bindCallbackGuarded.<anonymous closure>  (dart:async/zone.dart:963:23)
E/flutter (11807): #5      _rootRun  (dart:async/zone.dart:1124:13)
E/flutter (11807): #6      _CustomZone.run  (dart:async/zone.dart:1021:19)
E/flutter (11807): #7      _CustomZone.bindCallback.<anonymous closure>  (dart:async/zone.dart:947:23)
E/flutter (11807): #8      Timer._createTimer.<anonymous closure>  (dart:async-patch/timer_patch.dart:21:15)
E/flutter (11807): #9      _Timer._runTimers  (dart:isolate-patch/timer_impl.dart:382:19)
E/flutter (11807): #10     _Timer._handleMessage  (dart:isolate-patch/timer_impl.dart:416:5)
E/flutter (11807): #11     _RawReceivePortImpl._handleMessage  (dart:isolate-patch/isolate_patch.dart:171:12)
```

Is this a flutterblue, a dart or a user problem?

FWeissenb, 2019-07-08:
if I handle it this way 

```
runZoned(() async {
      await r.device.connect(timeout: Duration(seconds: 5), autoConnect: false);
    }, onError: (e){
              print(debugPrefix + " Error: $e");
          error = true;
    });
```

the error message is printed but it will not leave the zone :/

Any suggestions @pauldemarco

sierraRhoades, 2019-07-31:
I was having the same problem. I found that doing my device connection call this way will properly handle the timeout occurrence. Its a little frustrating that if you declare the timeout in the parameters that you can't then declare an error handler. Maybe this could be updated where it gives you an on error option if you declare the timeout when first calling device.connect() ? @pauldemarco 

```
  Future<bool> _connectToDevice() async
  {
    await device.connect().timeout(Duration(seconds:10), onTimeout: (){
      debugPrint('timeout occured');
      return Future.value(false);
    });
    
    debugPrint('got outside of the connect');
    return Future.value((true));
  }
```

sierraRhoades, 2019-07-31:
Or at least update the usage documentation to reflect this is the proper way to handle the timeouts?

FWeissenb, 2019-08-14:
You can try this @sierraRhoades 

```
 Future<bool> _connectToDevice(BluetoothDevice device,int timeout) async
  {
    Future<bool> returnValue;
    await device.connect(autoConnect: false).timeout(Duration(seconds:timeout), onTimeout: (){
      debugPrint('timeout occured');
      returnValue = Future.value(false);
      device.disconnect();
    }).then((data) {
      if(returnValue == null)
        {
          debugPrint('connection successful');
          returnValue = Future.value(true);
        }
    });

    return returnValue;
  }
```

So it will not reconnect and timeout is not handled in flutter_blue.

@pauldemarco is ther an other way ?

sierraRhoades, 2019-08-21:
@FWeissenb 
That worked thank you very much!

FWeissenb, 2019-09-10:
@pauldemarco  Can you please merge the PR to fix this?

SilentCatD, 2022-05-16:
any update on this?
I can try the way [FWeissenb](https://github.com/FWeissenb) suggest and it works, though it felt improper

jasaw, 2023-06-05:
> You can try this @sierraRhoades
>
> ```
>  Future<bool> _connectToDevice(BluetoothDevice device,int timeout) async
>   {
>     Future<bool> returnValue;
>     await device.connect(autoConnect: [...]

This method has a serious flaw where the caller's timeout kicks in and it doesn't cancel the underlying `connect` call.