pauldemarco/flutter_blue #304: flutter_blue-0.6.0+1 ... FlutterBluePlugin.java uses or overrides a deprecated API

mtin79, 2019-07-09:
Hi there,
i am getting this error message:

```
Note: /Users/.../flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.6.0+1/android/src/main/java/com/pauldemarco/flutterblue/FlutterBluePlugin.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details. 
```

Does that in any way indicate that flutter blue is not working properly with flutter channels "stable, beta or dev" or is this just a warning message?

Thanks for help!

SanjiKir, 2020-07-25:
@mtin79 This means that on Android `flutter_blue` uses some deprecated methods which can be deleted in the future. @pauldemarco, hey man, sorry for pinging you directly, but this might be an issue in the future, any plans on fixing it? :)

TheMaverickProgrammer, 2021-07-28:
We'd like this to be fixed for the future of the package as well. We really like this package.