pauldemarco/flutter_blue #305: Flutter_blue 0.6.0 - Absolute unusable

woozbooz, 2019-07-10:
@pauldemarco, please, change your mind about using RxDart.
RxDart use brainless and unusable logic.
Examle. With old code i can do next thing:
flutterBlue=FlutterBlue.instance;
```
_conn=flutterBlue.connect(scanResult.device, timeout: Duration(seconds: 5),
                                                autoConnect: false).listen(
                                                  (x){
                                                    if (x==BluetoothDeviceState.connected) {
                                                       _conn?.cancel();
                                                       _conn=null;
                                                       if (mounted) {
                                                         setState(() {
                                                           ..............
                                                         });
                                                       }
                                                    }
                                                  },
                                                  cancelOnError: true
                                                );
```
With this code i can have both - connection and state listener. And i can use one variable to store state and connection, so i can just use _conn.cancel() to disconnect device and stop to listen device state.
But with new API i need to have atleast two variables to store device and device state, also i have to change all my program code to use StreamBuilders everywhere to see if there is errors etc.

Please, change your mind about DartRx, ty.

bpillon, 2019-07-10:
@woozbooz If you want you still can use the old (0.5.0) version.
But with the new flutter stable (1.7.8), 0.5.0 doesn't work anymore in debug on Android (due to changes on FlutterJNI) . So I made a 0.5.0 fork compatible version if you want : https://github.com/bpillon/flutter_blue

woozbooz, 2019-07-10:
@bpillon, TY, i will check it later.
Did it compatible with 1.7.8?

casiff, 2019-07-16:
Yes the fork is working with 1.7.8!

khainke, 2019-07-23:
It looks like (at least for me) the fork also suffers from issue #317 which makes use also problematic wrt. reconnecting devices.

ekuleshov, 2019-07-29:
Agree with the original poster. With such major API changes so far down the road this package should have been renamed, or at very least some clear migration guide should have been provided.

khainke, 2019-07-29:
The current version has some major issues with #311 and #317 and therefore fails for anything but the most basic use cases. Otherwise, the idea of using RxDart is not a bad one, imo, its easier to read and clearer. But thats not worth losing functionality ...

shinayser, 2019-08-01:
The changes on the api were AWESOME, rxdart is amazing and makes everything easier. BUT! the basic functionallity simply stopped working. I just can't write characteristics aymore #331 . 

:'(

rdehouss, 2020-04-02:
Unfortunately, we have all been warned of possible breaking changes [in the README](https://github.com/pauldemarco/flutter_blue/blob/master/README.md#alpha-version)