pauldemarco/flutter_blue #313: device.state.listen does not receive connecting or disconnecting events

khainke, 2019-07-16:
```dart
device.state.listen(
  (state) {print(state);}
)
await device.connect()
await device.disconnect()
```
Expected Output:
```dart
BluetoothDeviceState.disconnected (initial)
BluetoothDeviceState.connecting
BluetoothDeviceState.connected
BluetoothDeviceState.disconnected
BluetoothDeviceState.disconnected
```

Actual:
```dart
BluetoothDeviceState.disconnected
BluetoothDeviceState.connected
BluetoothDeviceState.disconnected
```

Mojachieee, 2019-07-29:
I dug into this a bit (at least for Android).

Currently the method `deviceState` in FlutterBluePlugin.java handles changes in the device state. 
This uses the Android function `BluetoothManager.getConnectionState` to get the new state.

However, that function only returns either the connected or disconnected state, despite the documentation claiming otherwise.
Here is its implementation ([source](https://android.googlesource.com/platform/frameworks/base/+/master/core/java/android/bluetooth/BluetoothManager.java)):
```java
public int getConnectionState(BluetoothDevice device, int profile) {
    if (DBG) Log.d(TAG, "getConnectionState()");
    List<BluetoothDevice> connectedDevices = getConnectedDevices(profile);
    for (BluetoothDevice connectedDevice : connectedDevices) {
        if (device.equals(connectedDevice)) {
            return BluetoothProfile.STATE_CONNECTED;
        }
    }
    return BluetoothProfile.STATE_DISCONNECTED;
}
```

So to fix this the plugin will need to manually change the state after a connect or disconnect call. 

  @pauldemarco what are you thoughts on this approach to a fix?

alejo9719, 2020-02-26:
I'm still having this issue. And what's worst, the state listener gets a "disconnected" state just after I execute the device connect method. When the device finally connects, the listener gets the "connected" state.

khainke, 2020-02-26:
Yeah, looks like the device.state gets updated with a value, whenever the current value gets read/known, and is not distinct by default, one could however use the Stream.distinct method to achieve this. Also this issue was not targeted in the last months, i personally use wrappers around the connect and disconnect methods to produce my own device state stream alias with intermediate states (connecting, disconnecting).

alejo9719, 2020-02-26:
Yes, the only solution I found was using wrappers to produce my own states...

philos3, 2020-04-28:
> Yes, the only solution I found was using wrappers to produce my own states...

i have the same issue. can you show me the example code of your solution

ElHyperion, 2020-08-25:
This issue still persists. If disconnecting and connecting states are not working, they should be either implemented or removed from the plugin completely.