pauldemarco/flutter_blue #321: BLE Device sending packet of data every second, iOS missing packets regularly, Android performs better.

Aluzja, 2019-07-22:
So we have a bluetooth enabled sensor that is sending out a data packet every second, app receives the raw packet and runs calculations etc on the data to display.

I rarely see a missed packet in Android, and if I do it's only a second maybe 2, with iOS I'm regularly seeing missing packets and can be anywhere from 3 seconds worth up to around 20 seconds of missed packets.

Is there something in the plugin that I can enable to catch more of these packets or is it an iOS thing where Android is handling bluetooth better?

bretie, 2019-08-07:
This is hard to believe. Android seems to have way more issues.

Aluzja, 2019-08-10:
The code isn't indicative of the problem. Proof is in the pudding as they say.
Here is a google sheet showing the packet readings between the devices. Both were ready to read when I reset the sensor that is sending out the packets. 

[iOS vs Android Packet Log](https://docs.google.com/spreadsheets/d/1uQ0lYWcwvYfVCt17HonuboxK4YOq8XzNp7hrutKClvs/edit?usp=sharing)