pauldemarco/flutter_blue #322: writeCharacteristic failed

xuzhengquan, 2019-07-25:
`
E/flutter (30742): [ERROR:flutter/lib/ui/ui_dart_state.cc(148)] Unhandled Exception: PlatformException(write_characteristic_error, writeCharacteristic failed, null)
E/flutter (30742): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:564:7)
E/flutter (30742): #1      MethodChannel.invokeMethod (package:flutter/src/services/platform_channel.dart:302:33)
E/flutter (30742): <asynchronous suspension>
E/flutter (30742): #2      BluetoothCharacteristic.write (package:flutter_blue/src/bluetooth_characteristic.dart:123:10)
E/flutter (30742): <asynchronous suspension>
E/flutter (30742): #3      InnoPen.write.<anonymous closure> (package:wzx/service/blue_service/inno_pen.device.dart:67:57)
E/flutter (30742): #4      new Future.delayed.<anonymous closure> (dart:async/future.dart:316:39)
E/flutter (30742): #5      _rootRun (dart:async/zone.dart:1120:38)
E/flutter (30742): #6      _CustomZone.run (dart:async/zone.dart:1021:19)
E/flutter (30742): #7      _CustomZone.runGuarded (dart:async/zone.dart:923:7)
E/flutter (30742): #8      _CustomZone.bindCallbackGuarded.<anonymous closure> (dart:async/zone.dart:963:23)
E/flutter (30742): #9      _rootRun (dart:async/zone.dart:1124:13)
E/flutter (30742): #10     _CustomZone.run (dart:async/zone.dart:1021:19)
E/flutter (30742): #11     _CustomZone.bindCallback.<anonymous closure> (dart:async/zone.dart:947:23)
E/flutter (30742): #12     Timer._createTimer.<anonymous closure> (dart:async-patch/timer_patch.dart:21:15)
E/flutter (30742): #13     _Timer._runTimers (dart:isolate-patch/timer_impl.dart:382:19)
E/flutter (30742): #14     _Timer._handleMessage (dart:isolate-patch/timer_impl.dart:416:5)
E/flutter (30742): #15     _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:171:12)
E/flutter (30742): 

`

xuzhengquan, 2019-07-26:
gatt need to  queue requests to resolve multiple requests.

siddharthshah87, 2019-07-28:
I think I am having the same issue for which I opened an issue. @pauldemarco - have you seen this issue in the past. Thanks for the help

shinayser, 2019-08-01:
I am having this same issue right now with 0.6.0+2 :'(

vachelcai, 2019-09-04:
the same issue with 0.6.0+4

JonYun, 2020-08-05:
> E / flutter（30742）：[ERROR：flutter / lib / ui / ui_dart_state.cc（148）]未处理的异常：PlatformException（write_characteristic_error，writeCharacteristic失败，为null）
> E / flutter（30742）：＃0 StandardMethodCodec.decodeEnvelope（软件包： flutter / src / services / message_codecs.dart：564：7）
> E / flutter（ 30742 ）：[＃1](https://github.com/pauldemarco/flutter_blue/issues/1) MethodChannel.invokeMethod（package：flutter / src / services / platform_channel.dart：302：33）
> E / flutter（30742）：
> E / flutter（30742）：[＃2](https://github.com/pauldemarco/flutter_blue/issues/2) BluetoothCharacteristic.write（package：flutter_blue / src / bluetooth_characteristic.dart：123：10）
> E / flutter（30742）：
> E / flutter（ 30742 ）：[＃3](https://github.com/pauldemarco/flutter_blue/issues/3) InnoPen.write。（包裹：wzx / service / blue_service / inno_pen.device.dart：67：57）
> E / flutter（ 30742 ）：[＃4](https://github.com/pauldemarco/flutter_blue/issues/4)新的Future.delayed。（dart：async / future.dart：316：39）
> E / flutter（ 30742 ）：[＃5](https://github.com/pauldemarco/flutter_blue/issues/5) _rootRun（dart：async / zone.dart：1120：38）
> E / flutter（ 30742 ）：[＃6](https://github.com/pauldemarco/flutter_blue/issues/6) _CustomZone.run（dart ：async / zone.dart：1021：19）
> E / flutter（ 30742 ）：[＃7](https://github.com/pauldemarco/flutter_blue/issues/7) _CustomZone.runGuarded（dart：async / zone.dart：923：7）
> E / flutter（ 30742 ）：[＃8](https://github.com/pauldemarco/flutter_blue/issues/8) _CustomZone.bindCallbackGuarded。（dart：async / zone.dart：963：23）
> E / flutter（ 30742 ）：[＃9](https://github.com/pauldemarco/flutter_blue/issues/9) _rootRun（dart：async / zone.dart：1124：13）
> E / flutter（ 30742 ）：[＃10](https://github.com/pauldemarco/flutter_blue/issues/10) _CustomZone.run（dart ：async / zone.dart：1021：19）
> E / flutter（30742）：[＃11](https://github.com/pauldemarco/flutter_blue/issues/11) _CustomZone.bindCallback。（dart：async / zone.dart：947：23）
> E / flutter（ 30742 ）：[＃12](https://github.com/pauldemarco/flutter_blue/pull/12) Timer._createTimer。（dart：async-patch / timer_patch.dart：21：15）
> E / flutter（ 30742 ）：[＃13](https://github.com/pauldemarco/flutter_blue/issues/13) _Timer._runTimers（dart：isolate-patch / timer_impl.dart：382：19）
> E / flutter（30742）：[＃ 14](https://github.com/pauldemarco/flutter_blue/issues/14) _Timer._handleMessage（dart：isolate-patch / timer_impl.dart：416：5）
> E / flutter（ 30742 ）：[＃15](https://github.com/pauldemarco/flutter_blue/issues/15) _RawReceivePortImpl._handleMessage（dart：isolate-patch / isolate_patch.dart：171：12）
> E / flutter（ 30742）：

I also encountered the same error as you. Have you solved this problem now?

iw2lsi, 2021-11-18:
Hi all,

  I'm having this error on v0.7.2... is this still unresolved  or I am doing something wrong ?

          tnx,

                   Giampaolo