pauldemarco/flutter_blue #326: Provide API to check if BLE scan is in progress

ekuleshov, 2019-07-29:
Internally `_isScanning.value == true` is used to check if BLE scan is in progress. Please provide API to do the same check in the app code

GauravPatni, 2019-07-29:
you can subscribe to "FlutterBlue.instance.isScanning" stream to get state / change in state .

Refer following code ( from flutter_blue example )

```
floatingActionButton: StreamBuilder<bool>(
        stream: FlutterBlue.instance.isScanning,
        initialData: false,
        builder: (c, snapshot) {
          if (snapshot.data) {
            return FloatingActionButton(
              child: Icon(Icons.stop),
              onPressed: () => FlutterBlue.instance.stopScan(),
              backgroundColor: Colors.red,
            );
          } else {
            return FloatingActionButton(
                child: Icon(Icons.search),
                onPressed: () => FlutterBlue.instance
                    .startScan(timeout: Duration(seconds: 8)));
          }
        },
      ),
```

ekuleshov, 2019-07-29:
In my case I can't use StreamBuilder and need to check if scan initiated on a previous screen is running when user switched to the next screen

shinayser, 2019-08-01:
> In my case I can't use StreamBuilder and need to check if scan initiated on a previous screen is running when user switched to the next screen

``` dart
await FlutterBlue.instance.isScanning.first
```

ekuleshov, 2019-08-02:
> > In my case I can't use StreamBuilder and need to check if scan initiated on a previous screen is running when user switched to the next screen
> 
> ```dart
> await FlutterBlue.instance.isScanning.first
> ```

It would be nice an easy if that worked. But that throws a stream state error when no scan is running. Besides, the library internally is using different code to check if scan is running. So, I'm asking to expose the same API to allow library consumers to use