pauldemarco/flutter_blue #327: 0.6.0 flutter connect

victoryeo, 2019-07-29:
in 0.3.3, i use the code below to connect to BT devices:
FlutterBlue flutterBlue = FlutterBlue.instance;
flutterBlue.connect(device.device, timeout: Duration(seconds: 60)).listen((s)

In 0.6.0, the connect method is changed to BluetoothDevice connect.
May i know how to use the connect method in 0.6.0 ?

VaivanOmmeren, 2019-07-29:
You can scan for devices by using the example mentioned in the [Readme](https://github.com/pauldemarco/flutter_blue/blob/master/README.md)

```
/// Start scanning
var scanSubscription = flutterBlue.scan().listen((scanResult) {
    // do something with scan result
    device = scanResult.device;
    print('${device.name} found! rssi: ${scanResult.rssi}');
});

/// Stop scanning
scanSubscription.cancel();
```

Once you have found the device you want to connect to you can connect to the device like this
```
/// Connect to the device
await device.connect();

/// Disconnect from device
device.disconnect();
```

Overall the [Readme](https://github.com/pauldemarco/flutter_blue/blob/master/README.md) does a great way of the basic usage of the package.

victoryeo, 2019-07-30:
In https://github.com/pauldemarco/flutter_blue/blob/master/lib/src/bluetooth_device.dart
Future<void> connect({ Duration timeout,    bool autoConnect = true, })

It is different from the function prototype in Readme.
await device.connect();

Furthermore, the connect method does not return any value, how to know the connect method is successful or not ?

VaivanOmmeren, 2019-07-30:
You can await the connecting.
```dart
await device.connect()
```

You can also listen to the StreamSubscription of the device state

```dart
var streamSubscription = device.state.listen((state){
print(state)
});
```

This will print the state of the device to the console every time it changes. Such as disconnect, connecting and connect.

victoryeo, 2019-07-31:
Thank you. For the benefit of others, i share the code.  And how to integrate this code to my `code?`
`var streamSubscription = device.state.listen((state){
  print(state)
});`

My code:
    `_scanSubscription = flutterBlue.scan(timeout: const Duration(seconds: 5),)
        .listen((scanResult) async {
      // do something with scan result
      device = scanResult.device;
      print('${device.name} found! rssi: ${scanResult.rssi}');
      devices.add(device);
      try {
        await device.connect();
        print("connected");
      } on PlatformException {
        print("Error");
      }
    }, onDone: () => _stopScan());`