pauldemarco/flutter_blue #330: Windows, MacOs and Linux support

WolfgangFahl, 2019-07-31:
To be truly multi-platform this library should be extensible to the major operating systems.

khainke, 2019-08-04:
Flutter for web is currently available as a technical PREVIEW. Right now, 99.9% of flutter applications run on ios and android.

pauldemarco, 2020-04-02:
Tracking macOS here #463

maks, 2021-02-07:
@robert-ancell I'm interested in getting Linux support so happy to collaborate if thats something you've already started working on?

robert-ancell, 2021-02-08:
@maks I did start looking at this. I was trying to connect up [bluez.dart](https://pub.dev/packages/bluez) to flutter_blue. Since bluez.dart is all written in Dart, this would require flutter_blue to be restructured to use a platform interface like [path_provider](https://github.com/flutter/plugins/tree/master/packages/path_provider) does. This would mean some plugins can use the platform channels, and others (i.e. Linux in this case) could just use native code. @pauldemarco have you had any thoughts on this?

I've currently shelved that work and am working on the connectivity plugin instead (connecting to NetworkManager). @maks would love to collaborate - feel free to ping me for reviews or help!

gocal, 2021-03-25:
We've working on the implementation at https://github.com/allbrightio/flutter_blue/tree/federated_plugin, linux plugin uses internally bluez.dart and the results are quite promising. we'd also like to port windows implementation of the other blue library https://github.com/woodemi/quick_blue to be compatible with flutter_blue.

pauldemarco, 2021-03-25:
@gocal Awesome work. What’s current state? We can push for an experimental release whenever you feel it’s ready.

maks, 2021-03-25:
@gocal That's great news! I'm guessing you already saw that Robert recently added [GATT support in bluez.dart](https://github.com/canonical/bluez.dart/issues/10)

sachaarbonel, 2021-04-12:
Also worth to mention that windows can be added via the [newly added win32 Bluetooth API](https://github.com/timsneath/win32/issues/57#issuecomment-815449104)

FXschwartz, 2021-05-17:
Is anyone currently working on this feature? Any idea on when it might be implemented?

bradx954, 2021-09-06:
Bumping because I am also interested in this. The Pinephone will count as a linux desktop in terms of flutter support once they get the the arm64 cross build issue figured out.

codercq, 2022-02-08:
Is there any Bluetooth plug-in for Windows platform? It may be a long time before Flutter_Blue supports Windows

fotiDim, 2022-02-09:
@codercq there is https://pub.dev/packages/quick_blue but it has been giving nothing but crashes to me.

jyashi, 2022-04-11:
It's April people... any updates on this? Or do i need to resort to a python hack?

GameTec-live, 2023-07-24:
any news? linux and windows support would be great!

tototo23, 2024-01-02:
Hi ! Does anyone know if Windows support in on the pipe for this lib or if we should an  other alternative ?