pauldemarco/flutter_blue #331: Error trying to write to chracteristc

shinayser, 2019-08-01:
The package I am trying to send is 20 bytes long:

```
/flutter (21603): [ERROR:flutter/lib/ui/ui_dart_state.cc(148)] Unhandled Exception: PlatformException(write_characteristic_error, writeCharacteristic failed, null)
E/flutter (21603): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:564:7)
E/flutter (21603): #1      MethodChannel.invokeMethod (package:flutter/src/services/platform_channel.dart:316:33)
E/flutter (21603): <asynchronous suspension>
E/flutter (21603): #2      BluetoothCharacteristic.write (package:flutter_blue/src/bluetooth_characteristic.dart:123:10)
E/flutter (21603): <asynchronous suspension>
```

I am using `  flutter_blue: ^0.6.0+2`

There is any workaround for this?

bretie, 2019-08-07:
And where is your code at?

chronocurl, 2019-08-07:
I have the same problem and I am really interested in the solution too! 
I do 
await bluetoothCharacteristic.write(utf8.encode(command)), withoutResponse: true ); 
and the error occurs after some writings.
And in this way it is every time  
await bluetoothCharacteristic.write(utf8.encode(command)));

shinayser, 2019-08-07:
> And where is your code at?

I've updated the question.

chronocurl, 2019-08-07:
I did a test and I pass 1 or 2 characters to the write command and in my case it does not seem to be the size that causes the error. The bluetooth device returns information during writing, could a collision between messages explain the error?

shinayser, 2019-08-07:
> I did a test and I pass 1 or 2 characters to the write command and in my case it does not seem to be the size that causes the error. The bluetooth device returns information during writing, could a collision between messages explain the error?

I am chunking the packages because our hardware is the limiting part here. He can't hadle packages more than 20 bytes. But doesn't matter how much bytes we send to the hardware, it always gives the error above.

Again, in the version 0.4.1 everything was working fine.
@pauldemarco Could give us some light?

chronocurl, 2019-08-07:
Did you try the write with the withoutResponse : true. In my case it works for some write and it crashes after.

pauldemarco, 2019-08-07:
Please make sure to ‘await’ any calls to the plugin, and the outer functions as well.
Example)
‘await  _characteristic.write(X);’
...
‘await sendPackage(service, package);’

chronocurl, 2019-08-07:
In my case I already have await before the write instruction and I still have the error. I notice that in my Characteristic listener I receive the writes that I send and also the returns of the bluetooth device. Is it possible that there is a problem when I send a message and return a message at the same time?

pauldemarco, 2019-08-07:
Awaiting with delay is not the same as awaiting the actual function call. Unless you’re referring to different code, which I would ask that you update the post.

shinayser, 2019-08-07:
After setting "withoutResponse" to false I was able to send packages to the hardware. But, now I am receiving duplicate characteristic updates, like https://github.com/pauldemarco/flutter_blue/issues/317.

khainke, 2019-08-07:
> 
> 
> I just set "withoutResponse" to `true` and now I am able to send packages to the hardware.
> But, now I am receiving duplicate characteristic updates, like #317.
> 
> _@pauldemarco The API design got awesome BTW, congratulations! Just need to fx those small issues and it will be AWESOME._

Just be careful with this, when the documentation is still correct in this case, writing without response will NOT guarantee a successful write/send. Which means, you could still drop some packages you are sending out, but you will not notice it, when writing without response.

shinayser, 2019-08-07:
> > I just set "withoutResponse" to `true` and now I am able to send packages to the hardware.
> > But, now I am receiving duplicate characteristic updates, like #317.
> 
> Just be careful with this, when the documentation is still correct in this case, writing without response will NOT guarantee a successful write/send. Which means, you could still drop some packages you are sending out, but you will not notice it, when writing without response.

I understand, but when sending with response, the error mentioned on top happens =(

chronocurl, 2019-08-07:
Here the code to write :
  @override
  void sendMessageToDevice(String command) async {

    if (this.isConnected) {
      await bluetoothCharacteristic.write(utf8.encode(command), withoutResponse: true );
    }
  }

  @override
  void calibration() {
      sendMessageToDevice("I");
  }

If necessary the code to set the characteristics listener : 
  void _setServiceCharistic() async{

    if (device==null) return;

    device.discoverServices().then((s) {
      services = s;
      for(var service in services) {
        if (service.uuid == Guid("0000ffe0-0000-1000-8000-00805F9B34FB")) {
          for (var c in service.characteristics) {
            if (c.uuid == Guid("0000ffe1-0000-1000-8000-00805F9B34FB")) {
              bluetoothCharacteristic=c;
              bluetoothCharacteristic.setNotifyValue(true);
              Future.delayed(const Duration(milliseconds: 100), () {});
              inConnectionStatus.add(ConnectionMessage(deviceState: deviceState, macAdresse: device.id.id));

              valueChangedSubscriptions = bluetoothCharacteristic.value.listen((d){
                ...
              });
            }
          }
        }
      }
    });
  }

khainke, 2019-08-08:
@shinayser If it's working without response, that's fine. But if you ever feel the need to write with response, you could still try what paul suggested and guarantee a single write at a time by awaiting the write calls, awaiting the outer function (sendPackage) calls and maybe adding a mutex or something like this.

chronocurl, 2019-08-08:
Currently I send only 1 write at a time to my device, there is no concurrency at this point. The first situation is that everything is fine when the device does not return anything. The second is that when a write is done and a return is expected the error will occur after a few iterations. I suppose there might be a problem when the send comes in at the same time as the receive, but I'm not sure. I see how to block the writes, but how to block the return of the device so that it does not conflict with the writes?

bretie, 2019-08-08:
> I just set "withoutResponse" to `true` and now I am able to send packages to the hardware.
> But, now I am receiving duplicate characteristic updates, like #317.
> 
> _@pauldemarco The API design got awesome BTW, congratulations! Just need to fx those small issues and it will be AWESOME._

You're getting duplicate characteristic updates? So on the device side that is broadcasting bluetooth you're getting duplicate writes? To elaborate... your write characteristic is being called multiple times?

shinayser, 2019-08-08:
The `characterist.value` is being called twice:

```dart
 charac.value.listen((values) {
         //This is getting called twice
});
```

chronocurl, 2019-08-08:
I did several tests today and I confirm that there is a real problem when a write message is sent along with a message from the bluetooth device.

write ---> crash * <------- BT Device
* PlatformException (write_characteristic_error, writeCharacteristic failed, null)

If the write and read messages are sent sequentially there is no problem.
Is there a way around the problem or is there a possible fix for the API?

Another question, is it normal that the bluetoothCharacteristic.value.listen recovers as much the messages coming from the write and the messages coming from bluetooth device? Normally we should not only receive messages from the bluetooth device?

khainke, 2019-08-08:
> Another question, is it normal that the bluetoothCharacteristic.value.listen recovers as much the messages coming from the write and the messages coming from bluetooth device? Normally we should not only receive messages from the bluetooth device?

See issue #311

Prathik-Jain, 2019-08-26:
I am not able to write even after setting `withoutResponse` to `true`.

```dart
  write(List<int> data) async {
    BluetoothDevice device = locator<BluetoothConnectionService>().device;
    if (device != null) {
      List<BluetoothService> services = await device.discoverServices();
      print(await services[0].characteristics[1]
          .write([0, 0], withoutResponse: true));
    }
  }
```
Gives me the same error as mentioned in the first comment.

z8a, 2019-11-17:
I've write characteristic error also in the plugin example. Is there a release in witch is it working?

Prathik-Jain, 2019-11-18:
@z8a I had similar issue, can you please make sure if the characteristic you are writing to is not read-only?

z8a, 2019-11-27:
@Prathik-Jain I checked writing with an HM-10 device and it worked. Then I changed the type of my device characteristic from 'Write without response' to 'Write' and it's working.

Mavrap, 2020-06-14:
Hi, @z8a , how did you set device characteristics from 'Write without response' to 'Write' on HM-10 ?
I have the same problem with flutter blue 0.7.2 on Android device. On IOS it works fine...

espresso3389, 2020-06-14:
I had similar issues with certain BLE devices but I recently found that the root cause of these issues are just the device latency.

I mean, we should have some delay like `await Future.delay(Duration(milliseconds: 100)` just before such write calls.

Of course, with certain Android/iOS devices, we don't need them but they don't harm anything.

Mavrap, 2020-06-14:
Thanks but I have already try the delay without success.
I don't what I can do!!! my app is finished and deployed on AppStore, just the write function don't works on Android...

Mavrap, 2020-06-14:
Furthermore, i have a function that writes only data without any exchange before when I click on a button, the pb is the same...

rdehouss, 2020-07-01:
I ended up with a retry loop on the write:
```
import 'package:quiver/iterables.dart'; // for "partition" function

...

int mtu = await device.mtu.first; // default 20 on Android?

partition(bytes, mtu).forEach((bytes) async {
  log.info('Sending chunk of ${bytes.length}: $bytes');

  var retry = 0;
  do {
    try {
      await characteristicWrite.write(bytes, withoutResponse: true);
      break;
    } on PlatformException {
      await Future.delayed(Duration(milliseconds: 100));
    }
    // 3 retries to have a failsafe and avoid potential infinite loop
    // TODO: Check if we cannot have a 'real/decent' max retry...
  } while (retry < 3);
});
```

SethKitchen, 2020-10-12:
I changed a couple things to get mine to work -- 

1) set minSdkVersion to 21 in build.gradle
2) targetSdkVersion 28
3) Add
```xml
    <uses-permission android:name="android.permission.BLUETOOTH" />
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"/>
```
to debug AND main Android.manifest.xml

4) Change all forEach statements (run asynchronously which causes overlap) to synchronous for loops. For example:

```dart
          List<BluetoothService> services = await r.device.discoverServices();
          print('got services');
          for (int j = 0; j < services.length; j++) {
           }
```
from

```dart
       services.forEach
```

brendabullorini, 2020-11-11:
> 
> 1. Change all forEach statements (run asynchronously which causes overlap) to synchronous for loops. For example:
> 
> ```dart
>           List<BluetoothService> services = await r.device.discoverServices();
>           print('got services');
>           for (int j = 0; j < services.length; j++) {
>            }
> ```
> 
> from
> 
> ```dart
>        services.forEach
> ```
> 

@SethKitchen Changing the forEach statements worked for me. Thank you

ugifractal, 2021-01-27:
```writeWithoutResponse``` on IOS will be ignored, if characteristic doesn't have property ```BLECharacteristic::PROPERTY_WRITE_NR```

here sample warning message:
```
does not specify the "Write Without Response" property - ignoring response-less write**
``` 

So make sure the characteristic has ```BLECharacteristic::PROPERTY_WRITE_NR``` to make ```writeWithoutResponse``` working on IOS

Ignacio1110, 2022-02-25:
In my situation, I did not cancel the stream subscription( for write value to the device) correctly when disconnecting the device.
so when it auto reconnecting to the device, will occur the error.

anas-iqbal, 2022-05-23:
Hey any update on this? This plugin works great, but I am also getting write errors sometimes due to which my app misbehaves, I am already using await with write calls also tried with future delay but still it gives me error. I just send 1 write cmd at a time, any help please?

karansingla007, 2023-02-23:
@channel I am stuck here, Anybody found any solution for read and write?

```
E/flutter ( 2750): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(read_characteristic_error, unknown reason, may occur if readCharacteristic was called before last read finished., null, null)
E/flutter ( 2750): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:653:7)
E/flutter ( 2750): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:315:18)
E/flutter ( 2750): <asynchronous suspension>
E/flutter ( 2750): #2      BluetoothCharacteristic.read (package:flutter_blue_plus/src/bluetooth_characteristic.dart:80:5)
E/flutter ( 2750): <asynchronous suspension>
E/flutter ( 2750): #3      DeviceScreen._buildServiceTiles.<anonymous closure>.<anonymous closure>.<anonymous closure> (package:flutter_blue_plus_example/main.dart:201:40)
```

```
E/flutter ( 2750): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(write_characteristic_error, writeCharacteristic failed, null, null)
E/flutter ( 2750): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:653:7)
E/flutter ( 2750): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:315:18)
E/flutter ( 2750): <asynchronous suspension>
E/flutter ( 2750): #2      BluetoothCharacteristic.write (package:flutter_blue_plus/src/bluetooth_characteristic.dart:117:18)
E/flutter ( 2750): <asynchronous suspension>
E/flutter ( 2750): #3      DeviceScreen._buildServiceTiles.<anonymous closure>.<anonymous closure>.<anonymous closure> (package:flutter_blue_plus_example/main.dart:208:29)
```

anas-iqbal, 2023-02-23:
@karansingla007 looks like this packg isn't being maintained anymore u can use this one https://pub.dev/packages/flutter_blue_plus

In my case making a singleton class of ble instance helped.

automationsky, 2023-07-29:
> I have the same problem and I am really interested in the solution too! I do await bluetoothCharacteristic.write(utf8.encode(command)), withoutResponse: true ); and the error occurs after some writings. And in this way it is every time await bluetoothCharacteristic.write(utf8.encode(command)));

Did you. find any solution ?