pauldemarco/flutter_blue #332: Gradle complains about minSdkVersion in AndroidManifest.xml

espresso3389, 2019-08-05:
Gradle complains about minSdkVersion in AndroidManifest.xml:

> ERROR: The minSdk version should not be declared in the android manifest file. You can move the version from the manifest to the defaultConfig in the build.gradle file.
> Remove minSdkVersion and sync project

* Android Studio 3.4.2
* flutter_blue-0.6.0+2
* Gradle 5.1.1
* Android Gradle Plugin 3.4.2

[√] Flutter (Channel master, v1.8.4-pre.91, on Microsoft Windows [Version 10.0.18950.1000], locale ja-JP)
[√] Android toolchain - develop for Android devices (Android SDK version 29.0.1)
[√] Android Studio (version 3.4)
[√] Connected device (1 available)

LiangLuDev, 2019-08-06:
edit app build.gradle minSdkVersion 19

espresso3389, 2019-08-11:
Yes, it's easy to edit the code manually.
But when I build my own app, the build process stops with the error.
And it seems that [AndroidManifest.xml](https://github.com/pauldemarco/flutter_blue/commit/761566a1d74be3895710f59e00784d56843050e0) should not have hard-coded minSdk version.

espresso3389, 2019-08-18:
@lidongze91 I'm not sure why do you hard-coded minSdkVersion on AndroidManifest.xml?
My understanding is that Gradle is responsible for minSdkVersion and minSdkVersion must not be in AndroidManifest.xml.

pauldemarco, 2019-08-18:
cc @mehmetf

lidongze91, 2019-08-19:
@espresso3389 It seems that the minSdkVersion defined in manifest needs to be in sync with the defaultConfig of build.gradle file. The reason is that Flutter supports min sdk version of 16 and for this plugin, certain API needs version 18. Anything lower than that won't work in this case.

mehmetf, 2019-08-20:
@lidongze91 does the sample app for the this plugin run via `flutter run`? That uses gradle to compile. If that works, then I am not sure why @espresso3389's environment is complaining.

If this becomes an issue, we might need to move the minSdk to build.config and parse it internally to populate the documentation.

lidongze91, 2019-08-20:
@mehmetf It runs just fine. I think the build.gradle just has to have the same minimum version on default config. Let me try out some other plugins to confirm on this. @espresso3389 Please let us know what error you encountered during your app build.

espresso3389, 2019-08-29:
Strangely, it seems that `flutter run` does not complain about this. It occurs only on Android Studio.

My app is using `minSdkVersion 21` so 18 is not the problem.

My current env:

```
Doctor summary (to see all details, run flutter doctor -v):
[√] Flutter (Channel stable, v1.7.8+hotfix.4, on Microsoft Windows [Version 10.0.18965.1005], locale ja-JP)

[√] Android toolchain - develop for Android devices (Android SDK version 29.0.2)
[√] Android Studio (version 3.5)
[√] Connected device (1 available)

• No issues found!
PS C:\Work\onboard> flutter doctor -v
[√] Flutter (Channel stable, v1.7.8+hotfix.4, on Microsoft Windows [Version 10.0.18965.1005], locale ja-JP)
    • Flutter version 1.7.8+hotfix.4 at C:\Work\flutter
    • Framework revision 20e59316b8 (6 weeks ago), 2019-07-18 20:04:33 -0700
    • Engine revision fee001c93f
    • Dart version 2.4.0

[√] Android toolchain - develop for Android devices (Android SDK version 29.0.2)
    • Android SDK at C:\Users\kawasaki\AppData\Local\Android\sdk
    • Android NDK location not configured (optional; useful for native profiling support)
    • Platform android-29, build-tools 29.0.2
    • Java binary at: C:\Program Files\Android\Android Studio\jre\bin\java
    • Java version OpenJDK Runtime Environment (build 1.8.0_202-release-1483-b03)
    • All Android licenses accepted.

[√] Android Studio (version 3.5)
    • Android Studio at C:\Program Files\Android\Android Studio
    • Flutter plugin version 38.2.3
    • Dart plugin version 191.8423
    • Java version OpenJDK Runtime Environment (build 1.8.0_202-release-1483-b03)

[√] Connected device (1 available)
    • HW 01K • LPF0219121000507 • android-arm64 • Android 8.1.0 (API 27)
```

ened, 2020-03-02:
It causes a lint error in conjunction with flutter 1.14.6 (current beta):

```
> Lint infrastructure error
  Caused by: java.lang.reflect.InvocationTargetException
  	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
  	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
  	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
  	at java.lang.reflect.Method.invoke(Method.java:498)
  	at com.android.tools.lint.gradle.api.ReflectiveLintRunner.runLint(ReflectiveLintRunner.kt:38)
  	at com.android.build.gradle.tasks.LintBaseTask.runLint(LintBaseTask.java:114)
  	at com.android.build.gradle.tasks.LintPerVariantTask.lint(LintPerVariantTask.java:63)
  	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
  	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
  	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
  	at java.lang.reflect.Method.invoke(Method.java:498)
  	at org.gradle.internal.reflect.JavaMethod.invoke(JavaMethod.java:103)
  	at org.gradle.api.internal.project.taskfactory.StandardTaskAction.doExecute(StandardTaskAction.java:49)
  	at org.gradle.api.internal.project.taskfactory.StandardTaskAction.execute(StandardTaskAction.java:42)
  	at org.gradle.api.internal.project.taskfactory.StandardTaskAction.execute(StandardTaskAction.java:28)
  	at org.gradle.api.internal.AbstractTask$TaskActionWrapper.execute(AbstractTask.java:717)
  	at org.gradle.api.internal.AbstractTask$TaskActionWrapper.execute(AbstractTask.java:684)
  	at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter$5.run(ExecuteActionsTaskExecuter.java:476)
  	at org.gradle.internal.operations.DefaultBuildOperationExecutor$RunnableBuildOperationWorker.execute(DefaultBuildOperationExecutor.java:402)
  	at org.gradle.internal.operations.DefaultBuildOperationExecutor$RunnableBuildOperationWorker.execute(DefaultBuildOperationExecutor.java:394)
  	at org.gradle.internal.operations.DefaultBuildOperationExecutor$1.execute(DefaultBuildOperationExecutor.java:165)
  	at org.gradle.internal.operations.DefaultBuildOperationExecutor.execute(DefaultBuildOperationExecutor.java:250)
  	at org.gradle.internal.operations.DefaultBuildOperationExecutor.execute(DefaultBuildOperationExecutor.java:158)
  	at org.gradle.internal.operations.DefaultBuildOperationExecutor.run(DefaultBuildOperationExecutor.java:92)
  	at org.gradle.internal.operations.DelegatingBuildOperationExecutor.run(DelegatingBuildOperationExecutor.java:31)
  	at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter.executeAction(ExecuteActionsTaskExecuter.java:461)
  	at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter.executeActions(ExecuteActionsTaskExecuter.java:444)
  	at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter.access$200(ExecuteActionsTaskExecuter.java:93)
  	at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter$TaskExecution.execute(ExecuteActionsTaskExecuter.java:237)
  	at org.gradle.internal.execution.steps.ExecuteStep.lambda$execute$1(ExecuteStep.java:33)
  	at java.util.Optional.orElseGet(Optional.java:267)
  	at org.gradle.internal.execution.steps.ExecuteStep.execute(ExecuteStep.java:33)
  	at org.gradle.internal.execution.steps.ExecuteStep.execute(ExecuteStep.java:26)
  	at org.gradle.internal.execution.steps.CleanupOutputsStep.execute(CleanupOutputsStep.java:58)
  	at org.gradle.internal.execution.steps.CleanupOutputsStep.execute(CleanupOutputsStep.java:35)
  	at org.gradle.internal.execution.steps.ResolveInputChangesStep.execute(ResolveInputChangesStep.java:48)
  	at org.gradle.internal.execution.steps.ResolveInputChangesStep.execute(ResolveInputChangesStep.java:33)
  	at org.gradle.internal.execution.steps.CancelExecutionStep.execute(CancelExecutionStep.java:39)
  	at org.gradle.internal.execution.steps.TimeoutStep.executeWithoutTimeout(TimeoutStep.java:73)
  	at org.gradle.internal.execution.steps.TimeoutStep.execute(TimeoutStep.java:54)
  	at org.gradle.internal.execution.steps.CatchExceptionStep.execute(CatchExceptionStep.java:35)
  	at org.gradle.internal.execution.steps.CreateOutputsStep.execute(CreateOutputsStep.java:51)
  	at org.gradle.internal.execution.steps.SnapshotOutputsStep.execute(SnapshotOutputsStep.java:45)
  	at org.gradle.internal.execution.steps.SnapshotOutputsStep.execute(SnapshotOutputsStep.java:31)
  	at org.gradle.internal.execution.steps.CacheStep.executeWithoutCache(CacheStep.java:208)
  	at org.gradle.internal.execution.steps.CacheStep.execute(CacheStep.java:70)
  	at org.gradle.internal.execution.steps.CacheStep.execute(CacheStep.java:45)
  	at org.gradle.internal.execution.steps.BroadcastChangingOutputsStep.execute(BroadcastChangingOutputsStep.java:49)
  	at org.gradle.internal.execution.steps.StoreSnapshotsStep.execute(StoreSnapshotsStep.java:43)
  	at org.gradle.internal.execution.steps.StoreSnapshotsStep.execute(StoreSnapshotsStep.java:32)
  	at org.gradle.internal.execution.steps.RecordOutputsStep.execute(RecordOutputsStep.java:38)
  	at org.gradle.internal.execution.steps.RecordOutputsStep.execute(RecordOutputsStep.java:24)
  	at org.gradle.internal.execution.steps.SkipUpToDateStep.executeBecause(SkipUpToDateStep.java:96)
  	at org.gradle.internal.execution.steps.SkipUpToDateStep.lambda$execute$0(SkipUpToDateStep.java:89)
  	at java.util.Optional.map(Optional.java:215)
[... cut ...]
  	at com.android.builder.errors.EvalIssueReporter.reportError(EvalIssueReporter.kt)
  	at com.android.build.gradle.internal.ide.ModelBuilder.createVariant(ModelBuilder.java:601)
  	at com.android.build.gradle.internal.ide.ModelBuilder.buildAndroidProject(ModelBuilder.java:399)
  	at com.android.build.gradle.internal.ide.ModelBuilder.buildAll(ModelBuilder.java:193)
```

From a developer POV, I need to fork the project, and refer to that in pubspec.yaml to remove the warning. The lint seems to be very explicit about only specifying minSdk in gradle files now.

alekskuzmin, 2020-03-30:
I hope this gets merged ASAP.

closed 2020-04-02 by pauldemarco