pauldemarco/flutter_blue #333: Help! connect to device

coduy96, 2019-08-07:
How can I connect to device if I only have mac address value? Thanks a lot!

Fleximex, 2019-08-08:
If you know the information of the device beforehand then you can create a `BluetoothDevice` object yourself and use that to connect.
Simple example:
```
BluetoothDevice myDevice = BluetoothDevice(
   id: DeviceIdentifier("MAC address"),
   type: BluetoothDeviceType.le,
   name: "Device Name",
),
FlutterBlue.instance.connect(mydevice);
```

coduy96, 2019-08-09:
Thanks for your answer but I can not create a BluetoothDevice object by your ways! 
This is why: DeviceIdentifier("MAC address") - Only static members can be accessed in initializers.

coduy96, 2019-08-09:
Thanks for your support <3

votruk, 2019-08-09:
On iOS you cannot connect to a device via mac address - they are not exposed to the users (https://stackoverflow.com/a/2971250/4312184). As I know, the only way to do it is to store device mac address inside an advertisement manufacturer data. At least we do it with our devices and everything working good.

louwy001, 2019-08-12:
If I got MAC address from advertisement manufacture data and then saved that address, I still need scan and filter results through comparing advertisement manufacture data with the saved MAC address, and then connect device on iOS? I can’t connect device directly with MAC address? Thanks!

JUREXT, 2020-08-05:
I am using ^0.7.2 version and when I run on apple device my app, the logs don't show me the mac address, but I know the device mac address to connect to, the code above doesn't work for me. How can in create device in connect to it? Thank you.

yuriboeira11tx, 2021-04-20:
> I am using ^0.7.2 version and when I run on apple device my app, the logs don't show me the mac address, but I know the device mac address to connect to, the code above doesn't work for me. How can in create device in connect to it?
> Problem: BluetoothDevice doesn't have a default constructor.
> Thank you.

I'm also in trouble

WanKapef, 2022-01-12:
Does anyone have any idea how to solve this issue, I also want to connect to a specific device, and there is no longer the constructor in the BluetoothDevice class.