pauldemarco/flutter_blue #334: No onStateChanged() method in BluetoothDevice

daniels20000, 2019-08-07:
Similarly to issue #278 state instead of onStateChanged()

khainke, 2019-08-07:
I guess Paul did not have time on hand to reflect all of the latest changes in the readme, but the package documentation helps, listen to the state stream of the BluetoothDevice: https://pub.dev/documentation/flutter_blue/latest/flutter_blue/BluetoothDevice/state.html

daniels20000, 2019-08-07:
Paul is a great developer and I'm thankful for the work he've put into this repo. I just wanted to point this out in case someone else would struggle to make this work. :)

Fleximex, 2019-08-09:
Don't if this is just about the documentation or a question on how to do it as well.

In Flutter Blue 0.6.0 you can do this:
```
BluetoothDeviceState deviceState = BluetoothDeviceState.disconnected;
List<BluetoothService> services = List();

deviceStateSubscription = device.state.listen((s) {
   deviceState = s;
   if (s == BluetoothDeviceState.connected) {
      device.discoverServices().then((s) {
         services = s;
      });
   }
});
```
Of course the above example is simplified. In my projects I had a lot of fluff. One of which is adding an `else if` statement for when the state is `disconnected`, to properly clean up my code.

daniels20000, 2019-08-09:
Thank You for your explanation @Fleximex :) I was migrating form 0.5 and I faced small difficulties with it :D

ringonotts, 2019-09-06:
Glad i saw this issue as the documentation is out of date.
This seems to have changed again (0.6.0+2) unless I am mistaken - device.state.listen now returns a stream.