pauldemarco/flutter_blue #336: Cant find any devices while flutter blue scanning

luissouza, 2019-08-08:
I try to find the devices with de oficial example:
https://github.com/pauldemarco/flutter_blue/tree/master/example

I have several devices (BLE) that should be listed, but they NEVER appear. What am i doing wrong? Please, help me! 

All the devices are (BLE).

gumbypp, 2019-08-09:
Have you used another app (ex: LightBlue) to verify that your device can discover these devices?

luissouza, 2019-08-09:
> Have you used another app (ex: LightBlue) to verify that your device can discover these devices?

luissouza, 2019-08-09:
> Have you used another app (ex: LightBlue) to verify that your device can discover these devices?

@gumbypp yes! Not appear to, why flutter_blue only appear BLE devices? I r

SzymonGalazka, 2019-08-19:
I have the same issue. I tried with LG G6 with the example code and it shows nothing. Same if I search for paired devices. My devices are listed on LightBlue and on app built on top of flutter_bluetooth_serial.

   **scanForDevices() {**

    print('list of paired devices');
    flutterBlue.connectedDevices.asStream().listen((paired) {
      print('paired device: $paired');
    });
    setState(() {
      _isScanning = true;
    });
    scanSubscription =
        flutterBlue.scan(timeout: Duration(seconds: 20)).listen((scanResult) {
      var device = scanResult.device;
      print('${device.name} found! rssi: ${scanResult.rssi}');
    }, onDone: _stopScan);
    scanSubscription.cancel();
  **}**
_logs:
`I/flutter (21110): list of paired devices
D/BluetoothManager(21110): getConnectedDevices
D/FlutterBluePlugin(21110): mGattServers size: 0
D/BluetoothAdapter(21110): isLeEnabled(): ON
D/BluetoothLeScanner(21110): onScannerRegistered() - status=0 scannerId=5 mScannerId=0
I/flutter (21110): paired device: []
`

RuijinYao, 2019-09-10:
I also encountered the same situation, but I can search for the device after I open the location of the phone.

hicnar, 2019-11-29:
Same thing here, just one bluetooth device gets detected although plenty are on in detectable mode. Tried all available ScanModes but no progress  @RuijinYao by opening the location.. do you mean giving your app the location permission?

alejo9719, 2019-12-04:
Same problem

Skalate, 2020-02-03:
Me too, and I try agin by other bluetooth packages. but no one can work normally.

KushalRathore09, 2020-02-22:
HI Team ,  I am also facing the same problem. My use case is I am trying to connect bluetooth scanner device (Socket 7XI[68481F]) but can’t find it with this plugin, though other devices are visible . I also tried the other flutter bluetooth serial plugin but in my case I’ll be needing this functionality in my IOS app .  Does any one has got any work around for this , it will be much appreciate . Thanks in advance .

Skalate, 2020-02-22:
On Android devices, turn on GPS, and use all the functions in Flutter-blue. If not, phone can discover the device but cannot connect. Maybe you can turn on GPS. I haven't compiled the IOS program, maybe I can try it in a few days.
I am a student studying automation, I don’t know much about android, but I have gained a lot of experience in the process of learning flutter. I am happy to share with you, I hope to keep in touch. 

 

李峦
掌上理工

 联系我

KushalRathore09, 2020-02-26:
@Skalate thanks for reverting back , but unfortunately your solution didn't work for me . Have anyone got any solutions for this . Thanks in advance .

Skalate, 2020-02-27:
I don't know how to solve your problem, maybe you can try to contact the author of flutter-blue. Or use mixed development, it will take you some days to study the swift language. 

李峦
掌上理工

 联系我

GreeceNg, 2020-03-18:
Hi I also have the same problem with trying to connect my arduino bluetooth using this example code. Is there any solution for this problem?

luissouza, 2020-03-18:
@GreeceNg I have this problem. Some devices are not listed in this bluetooth plugin because they are different tecnologies. I had to change the plugin and now im using this: https://pub.dev/packages/blue_thermal_printer.

theali13, 2020-03-22:
Same Issue
FlutterBluePlugin( 7582): mDevices size: 0

ride4sun, 2020-03-24:
I have the same problem for me. I tried that 6 months ago with the same problem and I am surprised how little changed. 
@pauldemarco Hi Paul, do you have any input in this problem.

 @luissouza I need basic write and read functionality with BLE. Would the package u use work fort that? https://pub.dev/packages/blue_thermal_printer.

Any alternative packages which are better maintained would be highly appreciated. Any input?

theali13, 2020-03-24:
> I have the same problem for me. I tried that 6 months ago with the same problem and I am surprised how little changed.
> @pauldemarco Hi Paul, do you have any input in this problem.
> 
> @luissouza I need basic write and read functionality with BLE. Would the package u use work fort that? Seems like this package works only for Android: https://pub.dev/packages/blue_thermal_printer.
> 
> Any alternative packages which are better maintained and work with ios would be highly appreciated. Any input?

That package didn't work for me either it, it doesn't connect

Skalate, 2020-03-29:
Hello, I checked some information and sorted out the following possible solutions. 
1. Try programming with Object-c. Create a new empty directory and use 'flutter create -i objc my-flutter-app' to create a new project. 
2. Use a lower version of the API. 'use_frameworks! platform :ios, '9.0''
If none of the above solutions can be solved, I highly recommend you to learn IOS development, Mixed development seems to be the only way

李峦
掌上理工

 联系我

ride4sun, 2020-03-29:
Please try to use this branch. Defects got fixed on this branch which might fix ur problems. Keep track of this #145 to see if a new package get created to get better support.

use this in your yaml for now:
```
flutter_blue:
git:
url: https://github.com/boskokg/flutter_blue.git
ref: 0.7.1
```

OmiWakode, 2020-03-30:
> I try to find the devices with de oficial example:
> https://github.com/pauldemarco/flutter_blue/tree/master/example
> 
> I have several devices (BLE) that should be listed, but they NEVER appear. What am i doing wrong? Please, help me!
> 
> All the devices are (BLE).

I also have the same issue. I don't know whats wrong

dirkbo, 2020-03-30:
Same here. I should be able to see a lot of devices around me, but all i can see is my connected SPEN (Galaxy Note10), which i actually can't see when i open my phones bluetooth settings and scan for devices...

luissouza, 2020-04-04:
> I have the same problem for me. I tried that 6 months ago with the same problem and I am surprised how little changed.
> @pauldemarco Hi Paul, do you have any input in this problem.
>
> @luissouza I need [...]

Im using now esc_pos_bluetooth plugin.
This plugin (esc_pos_bluetooth) cant fin my device too. 
I get this plugin example, and i change a bit, in the connection code, like this.
```
 printerManager.selectPrinter(PrinterBluetooth(
        flutterBluetoothBasic.BluetoothDevice.fromDocument(
            printerName, printerAddress, 0, true)));
```

Hope it helps!

luissouza, 2020-04-04:
@ride4sun 

Im using now **esc_pos_bluetooth** plugin.
This plugin (**esc_pos_bluetooth**) cant fin my device too. 
I get this plugin example, and i change a bit, in the connection code, like this.

```
 printerManager.selectPrinter(PrinterBluetooth(
        flutterBluetoothBasic.BluetoothDevice.fromDocument(printerName, printerAddress, 0,true)));
```

Hope it helps!

Akilesh30, 2020-08-06:
Hii 
I'm working on social distancing app 
I'm facing same issue couldn't get any mobile devices in scanresult

javiervalero31, 2020-09-04:
I Have the same problem ! 
When I start the scanning, the stream scanResults don't return anything...
This feature is the most basic and don't work !
Any idea ?

luissouza, 2020-09-14:
@javiervalero31, i think, is not return anything because all the devices that you are looking for are not BLE, and this plugin just show BLE devices.. Look in play store for BLE app, this app shows to you if your device is BLE (Bluetooh low energy).

nadjwarazali, 2020-09-14:
I have the same problem before and all I did is turning on the location/ GPS and try to re-scan again and it works for my ESP32 BLE!

napong, 2020-09-18:
I'm facing same problem too.
It works on many device to connect my thermal printer.  e.g. iphone, ipad, redmi, huewei p30.
But it's not working for my samsung note10lite and note10+.
Any idea?

Detlev1, 2020-09-26:
@nadjwarazali Thank you it worked

afarre, 2020-10-19:
Had the same problem with https://github.com/lupyuen/flutter-blue-sample example (based on pauldemarco's) and as others mentioned, turing on GPS (as well as bluetooth) solved this problem. I find it odd that you need location ON in order to find bluetooth devices. Anyone knows why?

ganeshlg, 2020-10-28:
Turn on GPS it will work fine...!!!

BananaNeil, 2021-01-08:
I was experiencing this issue when testing on a device with the --release flag.

Apparently there is an issue with the minification of flutter blue. There are two ways of working around the problem, as listed in this issue: https://github.com/pauldemarco/flutter_blue/issues/662

In android/app/build.gradle you can disable minifaction:
```
android {
    ...
    buildTypes {
        release {
            signingConfig signingConfigs.debug
            shrinkResources false
            minifyEnabled false
        }
    }
```

OR you can create a proguard rule that prevents flutter_blue from minimizing:
In android/app/build.gradle:
```
android {
    ...
    buildTypes {
        release {
            signingConfig signingConfigs.debug
            minifyEnabled true
            useProguard true
            proguardFiles getDefaultProguardFile('proguard-android.txt'), 'proguard-rules.pro' 
        }
    }
```
and create a file at android/app/proguard-rules.pro with the following:
```
-keep class com.pauldemarco.flutter_blue.** { *; }
```

dprophete, 2021-04-06:
Has anybody tried to connect to an epson TM-T88VI ? It just doesn't show up in the list. Tried to enable GPS and all but still can't detect it (the android/ios app provided by Epson does detect it though...)

FlynnDerek, 2021-05-17:
Hi all, I'm also having this issue, the scan only returns two devices named 'gDevice-beacon' which are not connectable. I did make sure that the ESP32 was discovered on lightblue, and it shows up. I also enabled location and bluetooth scanning on the app. I also tried disabling minimizing with no luck. I'm just wondering if anyone else has something I could try?

AndyTanuC, 2021-05-18:
> Hi all, I'm also having this issue, the scan only returns two devices named 'gDevice-beacon' which are not connectable. I did make sure that the ESP32 was discovered on lightblue, and it shows up. I [...]

Are you running your app on your emulator ? If yes, then it's to be expected, since the emulators are not supporting the bluetooth yet (cmiiw). I faced the same issue, but when I try to run it on my android phone, it works just fine.

FlynnDerek, 2021-05-19:
Hey @AndyTanuC, you're right. I was using an emulator, but decided to buy a cheap android tablet and its working great now. Thanks for your help!

islom9797, 2022-05-25:
As anywone used this package after change o null safety?
i Have same issue
I tried so many types ,turn on location ,mifinify enablaed false,shrinkresourses false, but no reaction
D/FlutterBluePlugin(20287): mDevices size: 0
@pauldemarco  plase help. I tried all points which have there

islom9797, 2022-05-25:
at the moment i am using Galaxy s21 and Galaxy fold for test

Dyaamohamed123, 2022-08-30:
i had the same problem and it was in the permissions

UthraGomatam, 2022-10-12:
@DyaaElabasiry how did you resolve it? pls help

vytautas-pranskunas-, 2022-12-01:
Any updates on this? Why this packages is so bad maitained?

AlphaRosh, 2022-12-03:
hi there i tried all the above added proguard,disabled shrinkResources false
minifyEnabled false and advices from others too , but the scan results dont show the devices which are active

RafCoder90, 2023-01-14:
> hi there i tried all the above added proguard,disabled shrinkResources false minifyEnabled false and advices from others too , but the scan results dont show the devices which are active

Sto avendo lo stesso problema ma vedo che il dispositivo bluetooth non me lo trovano nemmeno altre app, ma se dal mio smartphone faccio una normale scansione lo trova

KattlerJoe, 2023-03-05:
Got the same problem, turning in GPS solved my problem.
I also got different bugs on different android versions, all solved by the GPS-thing. 
Does anyone have an idea why flutter_blue does correlate with the GPS settings? It **has** something to do with permissionshandling..

RafCoder90, 2023-03-06:
Is this enough to grant permission?
<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
...
I have noticed that several third party apps also cannot find the device

Lokendra16, 2023-03-27:
Make sure that device Gps location is enable, if the Gps is disable then you won't be able to scan the device.

thebeastxdxd, 2023-04-06:
Looks like enabling GPS location for our app worked! We did not need to enable GPS when we built from a windows machine...
it only happened when we started building and debugging on macOS. Very weird, I guess when building on mac it really cares about permissions.
Anyways thanks guys

larsien, 2023-05-26:
in my case, I added below permission then it worked
I found this from changelog(https://pub.dev/packages/flutter_blue_plus/changelog#140)
```
<uses-permission android:name="android.permission.BLUETOOTH_SCAN"
  android:usesPermissionFlags="neverForLocation" />
```

larsien, 2023-05-26:
oh my... 
in release mode built apk , it still not work (no scan results... )
only debug build apk work. 
how to solve this?