pauldemarco/flutter_blue #337: BLE devices always disconnected on Android devices

abdussamad0, 2019-08-09:
BLE devices always disconnected on **Android** devices but easily connected on iOS devices. i have **Android 7 version** devices.

Fleximex, 2019-08-12:
Can you please elaborate?

Is the problem that you can't connect to BLE devices or that when you are connected you lose connection?

abdussamad0, 2019-08-12:
No i can not connect any BLE device via android instead when i use iOS devices and i am using example code in this repo ..

Fleximex, 2019-08-13:
but you can you find devices on Android?

Can you post the logs when you try to connect to a device?

andersunloc, 2019-08-16:
Is this the same issue I have found? Basically when you connect and listen for state you first get a 'disconnected' states passed to the listener. We had code disconnecting when we received this, so we have to ignore it the first time (unless a connect/connecting has happened). Does not happened on iOS, and happens now and then on androids.

Fleximex, 2019-08-16:
Can you both share your code where you connect to the device and add the listener to the device state?
I don'get disconnected states after I connect with my code.

andersunloc, 2019-08-16:
_connStateSub = _device.state.listen(connectionStateChanged);
await _device.connect(autoConnect: false);

abdussamad0, 2019-08-16:
> Is this the same issue I have found? Basically when you connect and listen for state you first get a 'disconnected' states passed to the listener. We had code disconnecting when we received this, so [...]

Yes i am getting 'disconnected' status then how can i ignore this because whenever i received this it does not got to state subscription ..

andersunloc, 2019-08-16:
I simply keep a flag hasConnected, which i set on connected, and then I only care about disconnects after hasConnected is true. Might not solve everything, but should take you pass the first one. Also I discovered a 'disconnect' and then a second 'disconnect' when there was an error, so simply ignoring the first message if its a disconnect solves it for me.

FWeissenb, 2019-08-16:
That's a quick fix but should be fixed in flutter_blue @pauldemarco 
I get btw the same behavior. 

```
I/flutter ( 9235): BLEHelper | Connect C9:11:3C:40:17:8A
D/BluetoothManager( 9235): getConnectedDevices
D/BluetoothGatt( 9235): connect() - device: C9:11:3C:40:17:8A, auto: false
D/BluetoothGatt( 9235): registerApp()
D/BluetoothGatt( 9235): registerApp() - UUID=fe07323f-67f9-4941-9df2-95b0ee2a5ed8
D/BluetoothGatt( 9235): onClientRegistered() - status=0 clientIf=8
D/BluetoothManager( 9235): getConnectionState()
D/BluetoothManager( 9235): getConnectedDevices
D/BluetoothManager( 9235): getConnectionState()
D/BluetoothManager( 9235): getConnectedDevices
I/flutter ( 9235): BLEHelper | State: BluetoothDeviceState.disconnected
D/BluetoothGatt( 9235): onClientConnectionState() - status=0 clientIf=8 device=C9:11:3C:40:17:8A
D/FlutterBluePlugin( 9235): [onConnectionStateChange] status: 0 newState: 2
I/flutter ( 9235): BLEHelper | State: BluetoothDeviceState.connected 
```

apinprastya, 2019-09-06:
I got this error using Android 6.0

`
D/BluetoothGatt( 4684): onClientConnectionState() - status=0 clientIf=6 device=70:39:3A:57:BC:48
D/BluetoothDevice( 4684): mAddress: 70:39:3A:57:BC:48
D/FlutterBluePlugin( 4684): [onConnectionStateChange] status: 0 newState: 2
D/BluetoothDevice( 4684): mAddress: 70:39:3A:57:BC:48
W/BluetoothGatt( 4684): Unhandled exception in callback
W/BluetoothGatt( 4684): java.lang.RuntimeException: Methods marked with @UiThread must be executed on the main thread. Current thread: Binder_2
W/BluetoothGatt( 4684):         at io.flutter.embedding.engine.FlutterJNI.ensureRunningOnMainThread(FlutterJNI.java:794)
W/BluetoothGatt( 4684):         at io.flutter.embedding.engine.FlutterJNI.dispatchPlatformMessage(FlutterJNI.java:684)
W/BluetoothGatt( 4684):         at io.flutter.embedding.engine.dart.DartMessenger.send(DartMessenger.java:80)
W/BluetoothGatt( 4684):         at io.flutter.embedding.engine.dart.DartExecutor.send(DartExecutor.java:189)
W/BluetoothGatt( 4684):         at io.flutter.view.FlutterNativeView.send(FlutterNativeView.java:155)
W/BluetoothGatt( 4684):         at io.flutter.plugin.common.MethodChannel.invokeMethod(MethodChannel.java:98)
W/BluetoothGatt( 4684):         at io.flutter.plugin.common.MethodChannel.invokeMethod(MethodChannel.java:84)
W/BluetoothGatt( 4684):         at com.pauldemarco.flutterblue.FlutterBluePlugin$8.onConnectionStateChange(FlutterBluePlugin.java:757)
W/BluetoothGatt( 4684):         at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:187)
W/BluetoothGatt( 4684):         at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:70)
W/BluetoothGatt( 4684):         at android.os.Binder.execTransact(Binder.java:458)
`

zachdolan16, 2019-10-01:
I'm still experiencing this issue on Android:

Console output when trying to connect on Android: (state changes to disconnected, then connected 4 seconds later)

```
D/ViewRootImpl@bcbcf5f[MainActivity](23659): ViewPostIme pointer 0
D/ViewRootImpl@bcbcf5f[MainActivity](23659): ViewPostIme pointer 1
I/flutter (23659): trying to connect to B8:27:EB:41:B5:2B
D/BluetoothManager(23659): getConnectedDevices
D/BluetoothAdapter(23659): STATE_ON
D/BluetoothGatt(23659): connect() - device: B8:27:EB:41:B5:2B, auto: true
D/BluetoothAdapter(23659): isSecureModeEnabled
D/BluetoothGatt(23659): registerApp()
D/BluetoothGatt(23659): registerApp() - UUID=4b50ebf0-8a0d-4364-a7c0-fc3170ed1ccb
D/BluetoothManager(23659): getConnectionState()
D/BluetoothManager(23659): getConnectedDevices
D/BluetoothGatt(23659): onClientRegistered() - status=0 clientIf=5
D/BluetoothManager(23659): getConnectionState()
D/BluetoothManager(23659): getConnectedDevices
I/flutter (23659): 2019-10-01 09:30:54.090478 Device State: BluetoothDeviceState.disconnected
D/BluetoothGatt(23659): onClientConnectionState() - status=0 clientIf=5 device=B8:27:EB:41:B5:2B
D/FlutterBluePlugin(23659): [onConnectionStateChange] status: 0 newState: 2
I/flutter (23659): 2019-10-01 09:30:58.711746 Device State: BluetoothDeviceState.connected
D/BluetoothGatt(23659): onConnectionUpdated() - Device=B8:27:EB:41:B5:2B interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt(23659): onConnectionUpdated() - Device=B8:27:EB:41:B5:2B interval=39 latency=0 timeout=500 status=0
```

Console output when trying to connect on iOS:
```
flutter: BluetoothState.on
flutter: Bluetooth is on
didConnectPeripheral
flutter: trying to connect to 26FC02D5-1B5D-D1D5-D2D0-40CA2EFD1126
flutter: 2019-10-01 09:33:34.324716 Device State: BluetoothDeviceState.connected
```

I will try and make a solution around using a flag like @andersunloc suggested. Although a proper fix would be nice.

alejo9719, 2020-02-26:
I want to show the user a loading icon while the device is connecting. But since FlutterBlue first passes a "disconnected" state after requesting connection instead of a "connecting" state, I am only able to show a "connected" icon or no icon.

figureai, 2023-01-13:
Any one to fix it？