pauldemarco/flutter_blue #338: FlutterBlue.instance.scanResults shows stale results

daniels20000, 2019-08-13:
Hi,
i hope it won't be redundant. I'm currently facing problem with FlutterBlue.instance.scanResults stream. I'm trying to display/process List<ScanResult> and it shows stale results which are no longer available or in range. That's crucial for my application cause I try to make an app which opens doors and there will be potentially dozens of BLE enabled devices.

Fleximex, 2019-08-13:
Are you saving the results in a Map? For example `Map<DeviceIdentifier, ScanResult> scanResults = Map();`. Because then you can just clear the list before every scan `scanResults.clear();` And the stale (old) results are gone.

daniels20000, 2019-08-13:
I'm not saving results in a map. My intend is to scan all the time without need to refresh. User is suppose to walk through the building and depending upon rssi threshold app should connect to BLE device and communicate with it.

Fleximex, 2019-08-13:
Your results should be saved somewhere. How are you using the results if you don't have them stored somewhere?

Scanning can be done asynchronous. So you could keep scanning endlessly and then every (i.e.) 3 seconds just clear the scan result list.
Or maybe better for your situation. Save the results in a list (i.e. called 'current list') and then after 3 seconds copy it to another list (i.e. called 'check list'). Then compare both lists. And (after 3 seconds) if a result from check list is not present in current list you remove it from check list.
The only thing is that you need to use check list so you are 3 seconds behind the actual scan results.

daniels20000, 2019-08-13:
I think I'm gonna put that stream of scan results `Observable<ScanResult>(FlutterBlue.instance.scan(
      withServices: [Guid(myUUID)],
      scanMode: ScanMode(2)))
      .bufferTime(Duration(milliseconds: 500)).distinct((p,n)=> p.isEmpty&&n.isEmpty)
      .asyncMap((value) => value.toSet().toList()
    ..sort((a, b) => a.device.id.id.compareTo(b.device.id.id)))` which every 500ms yields me fresh List of buffered ScanResults  into Bloc and if any rssi will be under threshold  I'll attempt to connect to it.

giuseppe-v, 2019-10-04:
The Flutter Blue code actually clear the ` _scanResults ` list just on scan start:

```
Stream<ScanResult> scan({
    ScanMode scanMode = ScanMode.lowLatency,
    List<Guid> withServices = const [],
    List<Guid> withDevices = const [],
    Duration timeout,
  }) async* {
    var settings = protos.ScanSettings.create()
      ..androidScanMode = scanMode.value
      ..serviceUuids.addAll(withServices.map((g) => g.toString()).toList());

    if (_isScanning.value == true) {
      throw Exception('Another scan is already in progress.');
    }
    ....
    // Clear scan results list
    _scanResults.add(<ScanResult>[]);

    try {
      await _channel.invokeMethod('startScan', settings.writeToBuffer());
    } catch (e) {
      print('Error starting scan.');
      _stopScanPill.add(null);
      _isScanning.add(false);
      throw e;
    }
    ....
  } 
```

 The list is no longer cleaned during the scan. The items are just added or updated, so it contains all devices that are no longer available:
```
yield* Observable(FlutterBlue.instance._methodStream
            .where((m) => m.method == "ScanResult")
            .map((m) => m.arguments))
        .takeUntil(Observable.merge(killStreams))
        .doOnDone(stopScan)
        .map((buffer) => new protos.ScanResult.fromBuffer(buffer))
        .map((p) {
      final result = new ScanResult.fromProto(p);
      final list = _scanResults.value;
      int index = list.indexOf(result);
      if (index != -1) {
        list[index] = result;
      } else {
        list.add(result);
      }
      _scanResults.add(list);
      return result;
    });
```

giuseppe-v, 2019-10-04:
Could be useful a public method on FlutterBlue class that we can use to reset the scanResults Stream

pauldemarco, 2019-10-18:
I'm a bit hesitant to expose list manipulation methods on the API, as it could get a bit crowded.

This might be better handled in the end application.

I'm open to checking out PR's for this.

daniels20000, 2019-10-22:
How about a method which sets a timeout for every device after which it would be removed unless it would be still in range. I try to avoid clearing list completely and rebuilding it up from scratch.

itsgk92, 2020-05-12:
yes, this actually becomes an issue when we listen to scanResults asynchronously. Say the start and stop scan methods are invoked only on widget creation and dispose, listening to scanResults stream sends stale data (i.e) sends devices that are just discovered from beginning of scan even though they are not in range anymore.

Pavel-Gorokhov, 2021-02-24:
Temporary decision

`await FlutterBlue.instance.startScan(timeout: Duration(milliseconds: 1));
    await Future.delayed(Duration(milliseconds: 3));

    FlutterBlue.instance.scanResults.listen((snapshot) {
      // do smth
    });
    await FlutterBlue.instance.startScan(timeout: Duration(seconds: 4));`

jakubkrnac, 2022-07-27:
any updates? @daniels20000

gamenic-babin, 2023-07-04:
> Temporary decision
>
> ```
> await FlutterBlue.instance.startScan(timeout: Duration(milliseconds: 1));
>     await Future.delayed(Duration(milliseconds: 3)); [...]

Having same issue , i cant detect if device is already out of range due to it

gamenic-babin, 2023-07-04:
> yes, this actually becomes an issue when we listen to scanResults asynchronously. Say the start and stop scan methods are invoked only on widget creation and dispose, listening to scanResults stream [...]

yes this was the problem i am facing how do solve it . currently i want to remove device that are not in range but could not due to stale data