pauldemarco/flutter_blue #339: bluetooth 5 futures

devabacus, 2019-08-13:
Do you have a plan for supporting bluetooth 5 futures, especially in of coded phy layer (long range)
here's android studio [sdk info](https://source.android.com/devices/bluetooth/ble_advertising)

horikang, 2019-09-16:
How can I use the PHY 2M feature?

drenghel, 2019-10-17:
I am too interested in Bt5 functionalities !  
I'm guessing that it's not a simple thing to have them work in a cross-platform fashion (as I didn't found any on the web)

drenghel, 2019-10-17:
Extra question, (tiny bit off the scope of this repo) in the case the lib doesn't support it, do you guys think it's reasonable to write it ourselves with https://flutter.dev/docs/development/platform-integration/platform-channels ?

JoeBakalor, 2019-10-18:
@horikang There is no direct access to update the PHY configuration on the iOS side but you can enable the 2M PHY on iOS by performing a PHY update request from the peripheral side and iOS will accept the request as long as it supports that feature. This should work on Android as well for devices that support the 2M PHY