pauldemarco/flutter_blue #341: Cannot connect to passkey-required/bonding peripherals

kcyte2, 2019-08-16:
I am trying to use Flutter Blue to connect to Nordic Semiconductor peripheral device (acting as server). The Nordic peripheral device is currently running code that requires the client to enter passkey before the client can read/write to custom characteristics. In another Bluetooth app I can connect and bond with the Nordic peripheral device. After I bond with the Nordic peripheral device I also get a passkey request on my phone when I try to read a custom characteristic.  

However, in Flutter Blue example app when I scan the Nordic peripheral device I can see it but the "Connect" button is greyed out and I cannot click the connect button. There is also no passkey request. Is there anyway to fix this or is bonding and passkey request just not supported on Flutter Blue? Thanks!

Wassim24, 2020-01-07:
Hello @kcyte2 !
Have you ever got a reponse about this issue ? I trying to solve the same problem, but can't figure out if this is supported or not by flutter_blue.

Thank you in advance

kcyte2, 2020-01-08:
I think it was related to the Bluetooth mode that the Nordic Semiconductor Device was advertising. Changing it on the Nordic device side fixed it, but I can't remember specifically what the variable was. But after I changed it I could connect with Flutter Blue.

Wassim24, 2020-01-08:
Thank you @kcyte2  for your quick repsonse !

We'il try and see if we can do the same ;)

braveheart007, 2020-01-18:
@Wassim24  
issue had been solved or not? my bluetooth pheripheral can be connected by flutter_blue app. But I didn't found pair and bondding information on flutter_blue app.

Wassim24, 2020-01-18:
@braveheart007 
It seems that the bonding and pairing is handled by the android bluetooth stack. We changed some code to activate the bonding and pairing on our BLE peripheral and it worked right away.

magicleon94, 2021-01-11:
@Wassim24 what did you change?