pauldemarco/flutter_blue #342: On Android device nothing returns from "device.discoverServices()" and it does not complete.

earyzhe, 2019-08-17:
When trying to discover services on an Andriod device the future dos not complete.

Sometimes crashes with error: ` Fatal signal 11 (SIGSEGV), code 128`

[image]

#After calling the function these are the logs.

D/BluetoothGatt(19283): discoverServices() - device: 6E:38:B4:58:F5:9C
D/BluetoothAdapter(19283): getLeState() returning 12
D/BluetoothLeScanner(19283): startScan
D/BluetoothAdapter(19283): getLeState() returning 12
D/BluetoothAdapter(19283): STATE_ON
D/BluetoothAdapter(19283): getLeState() returning 12
D/BluetoothLeScanner(19283): onClientRegistered() - status=0 clientIf=9
D/BluetoothLeScanner(19283): mClientIf=0
I/BluetoothLeScanner(19283): startRegisteration: mLeScanClients={com.pauldemarco.flutterblue.FlutterBluePlugin$2@47c618d=android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper@c71ac90}
E/flutter (19283): [ERROR:flutter/lib/ui/ui_dart_state.cc(148)] Unhandled Exception: UnimplementedError
E/flutter (19283): null
D/BluetoothDevice(19283): getType: type = 2
D/BluetoothDevice(19283): getType: type = 0
D/BluetoothDevice(19283): getType: type = 0
D/BluetoothDevice(19283): getType: type = 2
D/BluetoothDevice(19283): getType: type = 2
D/BluetoothDevice(19283): getType: type = 0
D/BluetoothDevice(19283): getType: type = 0
D/BluetoothDevice(19283): getType: type = 2
D/BluetoothDevice(19283): getType: type = 0
D/BluetoothDevice(19283): getType: type = 2
D/BluetoothDevice(19283): getType: type = 0
I/art     (19283): Do partial code cache collection, code=31KB, data=30KB
I/art     (19283): After code cache collection, code=31KB, data=30KB
I/art     (19283): Increasing code cache capacity to 128KB
D/BluetoothDevice(19283): getType: type = 2
D/BluetoothDevice(19283): getType: type = 2
D/BluetoothDevice(19283): getType: type = 0
D/BluetoothDevice(19283): getType: type = 0
D/BluetoothDevice(19283): getType: type = 2
D/BluetoothDevice(19283): getType: type = 0
D/BluetoothDevice(19283): getType: type = 2
D/BluetoothDevice(19283): getType: type = 0
D/BluetoothDevice(19283): getType: type = 0
D/BluetoothDevice(19283): getType: type = 2
D/BluetoothDevice(19283): getType: type = 0
D/BluetoothDevice(19283): getType: type = 0
D/BluetoothDevice(19283): getType: type = 2
D/BluetoothDevice(19283): getType: type = 0
D/BluetoothDevice(19283): getType: type = 0
D/BluetoothDevice(19283): getType: type = 2

ricksondpenha, 2019-09-14:
DiscoverServices returns a future now. Check the latest version

ride4sun, 2020-03-27:
I see the same problem and I use await but I don't get an exception. I am wondering if I have to be connected to the device before I can call that method? Any help?

boskokg, 2020-03-27:
I am not sure if this can be of any reference...
[stackoverflow link](https://stackoverflow.com/a/31303972/3647184)

ride4sun, 2020-04-09:
@pauldemarco I think that can be close because the state was on but not connected. Your last checking should lead in this case to an exception