pauldemarco/flutter_blue #344: Flutter Blue classes need their constructors for fixed devices

Fleximex, 2019-08-19:
A few versions back the constructors of BluetoothDevice, BluetoothCharacteristic and BluetoothDescriptor where removed: https://github.com/pauldemarco/flutter_blue/pull/277/commits/01cd27f80d798717604c25ebbd8fa088777fa0ec
I don't really know why this was done as there are many case where you don't need to scan for a device and you know its MAC address beforehand.
You might even imagine cases where Bluetooth Devices are not advertising themselves to be found when scanning, but are open for connect requests.

Changes proposed:

**bluetooth_device.dart**
```
BluetoothDevice(
   {@required this.id,
         @required this.name,
         @required this.type,});
```
**bluetooth_characteristic.dart**
```
BehaviorSubject<List<int>> _value = BehaviorSubject.seeded([]);

BluetoothCharacteristic(
   {@required this.uuid,
      @required this.deviceId,
      @required this.serviceUuid,
      this.secondaryServiceUuid,
      @required this.descriptors,
      @required this.properties});
```
**bluetooth_descriptor.dart**
```
BluetoothDescriptor(
   {@required this.uuid,
      @required this.deviceId,
      @required this.serviceUuid,
      @required this.characteristicUuid});
```

louwy001, 2019-08-19:
You can use /gen/flutterblue.pb.dart BluetoothDevice.fromJson to construct a protos device, then BluetoothDevice.fromProto(protos.BluetoothDevice p) construct your device. Others are similar, I think.

Fleximex, 2019-08-20:
Sure you could use that, but is that the proper way? Wouldn't a constructor which matches the fromProto (in terms of parameters) be more logical? And what is the reason to omit a constructor for these classes?

koleto, 2019-08-27:
I absolutely agree with @Fleximex! There is even more scenarios where the default constructors are needed! [01cd27f](https://github.com/pauldemarco/flutter_blue/commit/01cd27f80d798717604c25ebbd8fa088777fa0ec) is a really bad idea! There should be an easy way to construct BluetoothDevice for fixed devices, and not just ugly work around...

koleto, 2019-08-28:
In my case I need to be able to serialise and deserialise a DeviceModel to a database.
@louwy001 proto. BluetoothDevice.fromJson not work properly, but this is what I found instead:

 ```
import 'package:flutter_blue/flutter_blue.dart';
import 'package:flutter_blue/gen/flutterblue.pb.dart' as proto;

...

  BluetoothDevice toBluetoothDevice() {
    proto.BluetoothDevice p = proto.BluetoothDevice.create();

    p.name = name;
    p.type = proto.BluetoothDevice_Type.LE;
    p.remoteId = bleno;

    return BluetoothDevice.fromProto(p);
  }
```
The fields `name` and `bleno` are part of the `DeviceModel` stored to the DB.  The deserialization method `toBluetoothDevice` ... kind of ugly but works for me.

aytunch, 2019-11-05:
I would like to be able to save the connected device and auto connect after the first connect.