pauldemarco/flutter_blue #346: Error: flutter_blue:compileDebugJavaWithJavac'.

earyzhe, 2019-08-19:
# Problem
When trying to compile the app on Android devices An error its thrown:
Execution failed for task ':flutter_blue:compileDebugJavaWithJavac

**Things I have tried**
- Migrating to AndroidX in Android Studio via "Refactor" > "Migrate to AndroidX"
- Updating dependencies in app/build.gradle:
   - compile 'com.android.support:support-compat:28.0.0'
   - compile 'com.android.support:appcompat-v7:28.0.0'
   - compile 'com.android.support:multidex:1.0.0'

# LOGS
`
/Users/earyzhe/dev/SDKs/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.4.2/android/src/main/java/com/pauldemarco/flutterblue/FlutterBluePlugin.java:30: error: cannot find symbol
import android.support.v4.app.ActivityCompat;
                             ^
  symbol:   class ActivityCompat
  location: package android.support.v4.app
/Users/earyzhe/dev/SDKs/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.4.2/android/src/main/java/com/pauldemarco/flutterblue/FlutterBluePlugin.java:31: error: package android.support.v4.content does not exist
import android.support.v4.content.ContextCompat;
                                 ^
/Users/earyzhe/dev/SDKs/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.4.2/android/src/main/java/com/pauldemarco/flutterblue/FlutterBluePlugin.java:162: error: cannot find symbol
                if (ContextCompat.checkSelfPermission(registrar.activity(), Manifest.permission.ACCESS_COARSE_LOCATION)
                    ^
  symbol:   variable ContextCompat
  location: class FlutterBluePlugin
/Users/earyzhe/dev/SDKs/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.4.2/android/src/main/java/com/pauldemarco/flutterblue/FlutterBluePlugin.java:164: error: cannot find symbol
                    ActivityCompat.requestPermissions(
                    ^
  symbol:   variable ActivityCompat
  location: class FlutterBluePlugin
Note: /Users/earyzhe/dev/SDKs/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.4.2/android/src/main/java/com/pauldemarco/flutterblue/FlutterBluePlugin.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: /Users/earyzhe/dev/FlutterProjects/battery_saver/build/flutter_blue/generated/source/proto/debug/javalite/com/pauldemarco/flutterblue/Protos.java uses unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
4 errors

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':flutter_blue:compileDebugJavaWithJavac'.
> Compilation failed; see the compiler error output for details.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org
`

BUILD FAILED in 8s
*******************************************************************************************
The Gradle failure may have been because of AndroidX incompatibilities in this Flutter app.
See https://goo.gl/CP92wY for more information on the problem and how to fix it.
*******************************************************************************************
Gradle task assembleDebug failed with exit code 1`

# VERBOSE
`
[        ] [        ] ->
+file:///Users/earyzhe/dev/SDKs/flutter/.pub-cache/hosted/pub.dartlang.org/collection-1.14.11/lib/src/combined_wrappers/combined_list.da
rt
[        ] [        ] ->
+file:///Users/earyzhe/dev/SDKs/flutter/.pub-cache/hosted/pub.dartlang.org/collection-1.14.11/lib/src/combined_wrappers/combined_map.dar
t
[        ] [        ] ->
+file:///Users/earyzhe/dev/SDKs/flutter/.pub-cache/hosted/pub.dartlang.org/collection-1.14.11/lib/src/comparators.dart
[        ] [        ] ->
+file:///Users/earyzhe/dev/SDKs/flutter/.pub-cache/hosted/pub.dartlang.org/collection-1.14.11/lib/src/equality.dart
[        ] [        ] ->
+file:///Users/earyzhe/dev/SDKs/flutter/.pub-cache/hosted/pub.dartlang.org/collection-1.14.11/lib/src/equality_map.dart
[... 996 lines cut ...]
#7      Future._propagateToListeners (dart:async/future_impl.dart:668:32)
#8      Future._complete (dart:async/future_impl.dart:473:7)
#9      _SyncCompleter.complete (dart:async/future_impl.dart:51:12)
#10     _AsyncAwaitCompleter.complete (dart:async-patch/async_patch.dart:28:18)
#11     _completeOnAsyncReturn (dart:async-patch/async_patch.dart:294:13)
#12     runCommandAndStreamOutput (package:flutter_tools/src/base/process.dart)
#13     _asyncThenWrapperHelper.<anonymous closure> (dart:async-patch/async_patch.dart:77:64)
#14     _rootRunUnary (dart:async/zone.dart:1132:38)
#15     _CustomZone.runUnary (dart:async/zone.dart:1029:19)
#16     _FutureListener.handleValue (dart:async/future_impl.dart:126:18)
#17     Future._propagateToListeners.handleValueCallback (dart:async/future_impl.dart:639:45)
#18     Future._propagateToListeners (dart:async/future_impl.dart:668:32)
#19     Future._completeWithValue (dart:async/future_impl.dart:483:5)
#20     Future._asyncComplete.<anonymous closure> (dart:async/future_impl.dart:513:7)
#21     _rootRun (dart:async/zone.dart:1124:13)
#22     _CustomZone.run (dart:async/zone.dart:1021:19)
#23     _CustomZone.runGuarded (dart:async/zone.dart:923:7)
#24     _CustomZone.bindCallbackGuarded.<anonymous closure> (dart:async/zone.dart:963:23)
#25     _microtaskLoop (dart:async/schedule_microtask.dart:41:21)
#26     _startMicrotaskLoop (dart:async/schedule_microtask.dart:50:5)
#27     _runPendingImmediateCallback (dart:isolate-patch/isolate_patch.dart:116:13)
#28     _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:173:5)

earyzhe@MacBook-Pro-5:~/dev/FlutterProjects/battery_saver$ 
`

pauldemarco, 2019-10-18:
Please try with the latest version 0.6.3

If you are still experiencing issues, try to upgrade your project files to the latest flutter template by deleting the `android` and `ios` folders, and running `flutter create .`

Note: Please make sure to backup any of your modified files in the android and iOS folders before performing the above.

maheshwara-bwd, 2019-10-20:
Im also facing the same issue. Please help.

scripta55, 2019-10-22:
also having the same issue, i thought this was caused by permissions, but i am guessing not now

jcbrtl, 2019-10-29:
I'm facing the same problem, exactly.

GauravPatni, 2019-11-05:
I am also facing exactly the same issue after updating with latest package  ^0.6.3+1

VitorHugoOli, 2019-11-07:
I used the solution proposed by @GauravPatni with one addition proposed in 

https://github.com/flutter/flutter/issues/27254#issuecomment-477474444

This solves all the problems

alejo9719, 2019-12-02:
I'm using the latest Flutter template and facing the issue. To make it work, I had to downgrade to 0.6.2 like @GauravPatni says; 0.6.3 and 0.6.3+1 don't work.