pauldemarco/flutter_blue #348: Inconsistent API for enabling indication

YuraCobain, 2019-08-20:
According to FlutterBlue API in order to enable indication bit for characteristic the following function should be used:

setNotifyValue Sets notifications or indications on the characteristic.

However the function accept single "bool notify". 
Moreover it looks like that indication is not implemented at all.

mbn18, 2019-10-04:
Yes, I also stumbled on the same issue.
To use indication I need to pass bool to `setNotifyValue(bool notify)`.
And if so, true or false?

When setting `true` it is registered to notify. But passing `false` seems to do nothing while I expected it to register indicate.

This is the code I played with:

```dart
  onPressed: () async {
    await snapshot.data.setNotifyValue(true);
    snapshot.data.value.listen((s) => print(s));
  },
```

So, is `indicate` supported?
If that the case, where is my mistake?
And also, it might be clearer if there where 2 methods (like `subscribIndicate`) for each that mask the bool issue.

Thanks

plmmanwu, 2020-08-03:
same issue, how to set indication enable

plmmanwu, 2020-08-03:
got it

d-silveira, 2020-11-17:
@plmmanwu what does got it mean, you got it working, can you explain how?

Looking at the code of Android part of the lib I can see that it checks whether the device supports indicate, in which case it sets it, otherwise it sets notify if it's supported.

But for me using this lib with a notify characteristic works flawlessly, however with an indicate characteristic it simply does not return any value, other than an initial empty list, leading me to believe that it's not sending the ack which indicate requires! Is this true @pauldemarco?

Any tips on how to get it to work?

d-silveira, 2020-11-19:
It seemed to be a glitch, it's working now!
Thanks

maddionatoplus, 2021-06-01:
I use this to enable indication: #858 (write to the specific descriptor [0x02, 0x00]

it works on Android but it crash on iOS.

Did you find any other solution?