pauldemarco/flutter_blue #349: How to clear "scanResults"

GauravPatni, 2019-08-20:
HI,

Is there any way to clear "**scanResults**" ? 

It require when we disconnect device & goto connect screen again ,where we call  '**FindDevicesScreen**' then it shows old devices as **snapshot.data** of **FlutterBlue.instance.scanResults** stream gives the last result.

Note that while calling "scan" scanResults list get clears because it internally calls **_scanResults.add(<ScanResult>[]);**

Fleximex, 2019-08-20:
Maybe we can add a clearList method to the class FlutterBlue. Or change the scan method:

Change
```
if (timeout != null) {
    killStreams.add(Observable.timer(null, timeout));
}
```
to
```
if (timeout != null) {
    killStreams.add(Observable.timer(clearList, timeout).listen(
        (clearList) {
            if (clearList)
            {
                _scanResults = BehaviorSubject.seeded([]);
            }
        }));
}
```
`clearList` being a boolean you can pass to the scan method.

jeanmatthieud, 2019-08-20:
Or at least add a 'lastScanDate' timestamp to the device to be able to filter devices.

GauravPatni, 2019-08-21:
Thanks for the reply .
But is there any way to clear scan results without editing package internal code ? as it is difficult to maintain if package version updates....

928PJY, 2020-10-04:
So is there an easy way to get the latest available device list? 
From my usage, when I keep scanning all the device, if I turn off my ble device and turn on it again, since it is not a new device, the listen callback function doesn't get invoked immediately, it looks wired.

LipJ01, 2023-08-24:
I know this is an old issue but any chance there is now a way to do this?? When I call repeatedly call startScan(timeout:X) I'm still using the listener from my original scan and it doesn't clear the results. I can't turn bluetooth off and on because I'm trying to maintain multiple connections. Any wizdom appreciated.