pauldemarco/flutter_blue #350: Restoring connected device state

derzhavets, 2019-08-22:
Hi
I'm trying to do following. I'm using iOS CB State Preservation and Restoration to restore BLE connection if application was terminated being in background. The problem is that when iOS relaunches Flutter application and restores BLE connection with device, I can't neither find device on scan (because it's already connected), nor receive data from this device (because plugin haven't gone through the routine of connecting-discovering-subscribing).

I tried to disconnect from device (to reconnect again), but iOS immediately suspends the application (because it's in background).

Any ideas how can I fully restore connection state within the plugin, when the app launching being already connected to BLE device?

Thanks in advance!!

valtbarbos, 2020-01-03:
Hey, this can help u => https://github.com/pauldemarco/flutter_blue/pull/191

JuniorJPDJ, 2021-01-16:
Any news about it?