pauldemarco/flutter_blue #352: BLE device need authentication

stonedMoose, 2019-08-25:
I try to use flutter_blue to read heartrate value from my Xiaomi MiBand 3.

But when I connect to the device, I'm deconnected every 30s... Reading this [medium article](https://medium.com/@yogeshojha/i-hacked-xiaomi-miband-3-and-here-is-how-i-did-it-43d68c272391), I understood that I need to authenticate to my MiBand by sending a specific byte sequence.

Exploring the FlutterBlue Api I did not find any way to send a raw byte sequence to a given bluetooth device. Am I missing something ? If not, it could be awesome to add this functionnality to the library.

Thanks in advance to any person willing to help.

pauldemarco, 2019-10-18:
Hi @stonedMoose 
You can send a raw byte sequence using the BluetoothCharaceristic's instance method `.write()`, which takes a List<int>.

```dart
await characteristic.write([0x01, 0x02]);
```

I recommend asking your question on StackOverflow, where it can be indexed and used as a reference later on.

aytunch, 2019-10-26:
@stonedMoose I am having the exact same problem. My connection with Mi band drops every 30 seconds. Could you find a way to overcome this?

claytonsands, 2020-10-02:
> @stonedMoose I am having the exact same problem. My connection with Mi band drops every 30 seconds. Could you find a way to overcome this?
> 
> heart rate service is 0x180D
> and Heart rate Measurement Characteristic is 0x2A37
> 
> what do we need to write and in where? Do we need to do anything about the Descriptor 0x2902?

it's not only write.
It is necessary to create the entire authentication process with the miband.

see https://leojrfs.github.io/writing/miband2-part1-auth/

I also need help to implement this.