pauldemarco/flutter_blue #353: Under IOS, the device can not automatically reconnect after disconnection. May I help you, please?

tanliubin, 2019-08-27:
// Connect to device
    deviceConnection = _flutterBlue.connect(device, autoConnect: true).listen((data){
      print(data);
    });

gaurav92-gs, 2020-01-02:
@tanliubin did you got any solution for this auto connect issue? If yes then please let me know. Because I am getting same problem at my end too.

valtbarbos, 2020-01-05:
To reproduce:

1) Device is out-range
2) Bluetooth radio being turned off and back on
3) Airplane mode turned on then off

In this scenario we need to try reconnect, however we fall into the following exception: https://github.com/pauldemarco/flutter_blue/blob/00f57b55cf2f7e08fd2be838f0b24c455feaf166/ios/Classes/FlutterBluePlugin.m#L278

didDisconnectPeripheral is not called when this state change (in out-range) occours
https://github.com/pauldemarco/flutter_blue/blob/00f57b55cf2f7e08fd2be838f0b24c455feaf166/ios/Classes/FlutterBluePlugin.m#L387

valtbarbos, 2020-01-05:
> To reproduce:
>
>     1. Device is out-range
>
>     2. Bluetooth radio being turned off and back on
>
>     3. Airplane mode turned on then off
>
>
> In this scenario we need to try reconnect, however we fall [...]

I found a way... is not completed but is a begin

I'm not a iOS develper so... things are crazy here 

didDisconnectPeripheral will be called and we can call a re-connection... but, all services, characteristics, and characteristic descriptors a peripheral become invalidated. In the didConnectPeripheral I clear helper arrays and call a new [peripheral discoverServices:nil]. The notify chars are keep.

The next step is cancelPeripheralConnection when we want an intentional disconnection.

- (void)centralManager:(CBCentralManager *)central didDisconnectPeripheral:(CBPeripheral *)peripheral error:(NSError *)error {
  NSLog(@"didDisconnectPeripheral");
  // Unregister self as delegate for peripheral, not working #42
  peripheral.delegate = self;
  
  // Try to reconnect
  [self.centralManager connectPeripheral:peripheral options:nil];

  // Send connection state
  [_channel invokeMethod:@"DeviceState" arguments:[self toFlutterData:[self toDeviceStateProto:peripheral state:peripheral.state]]];
}

- (void)centralManager:(CBCentralManager *)central didConnectPeripheral:(CBPeripheral *)peripheral {
  NSLog(@"didConnectPeripheral");
  // Register self as delegate for peripheral
  peripheral.delegate = self;

  // Clear helper arrays and [peripheral discoverServices:nil];
  [_servicesThatNeedDiscovered removeAllObjects];
  [_characteristicsThatNeedDiscovered removeAllObjects ];
  [peripheral discoverServices:nil];
  
  // Send initial mtu size
  uint32_t mtu = [self getMtu:peripheral];
  [_channel invokeMethod:@"MtuSize" arguments:[self toFlutterData:[self toMtuSizeResponseProto:peripheral mtu:mtu]]];
  
  // Send connection state
  [_channel invokeMethod:@"DeviceState" arguments:[self toFlutterData:[self toDeviceStateProto:peripheral state:peripheral.state]]];
}

Drkstr, 2020-01-23:
Do we have any updates on this.

VinniciusJesus, 2021-04-27:
Do we have any updates on this?

DanielPiflaks, 2021-08-24:
Any updates? Works for me on Android but not on IOS

DanielPiflaks, 2021-09-22:
> Meybe someone helps:
> Documentation for scanResults says that "When a scan is first started, an empty list is emitted."
> This is not true for the case when performing reconnection: we first reseive non empty result list, and after that we start receiving results starting from empty list.
> 
> So, the first received result is bad and should not be used.
> 
> So we made a workaround and just called skip first:
> .scanResults.skip(1).listen

I'm not sure how skipping the first list of scanned devices solves the issue of auto connection on iOS.

On Android, which there auto connection works fine, I'm not calling for a scan after the first time I found the device, but instead it just reconnects automatically after the device advertises again. 
I expect that on iOS it will work the same, so why do I need to scan again? @boskokg