pauldemarco/flutter_blue #354: ios, After setNotifyValue, the device does not change the real-time feedback value, but there is one in the printed log.

tanliubin, 2019-08-27:
flutter: connected!
flutter: BluetoothDeviceState.connected
didDiscoverServices
Found service: 53A20001-58DF-4E45-AC57-573028548B13
didDiscoverCharacteristicsForService
didDiscoverDescriptorsForCharacteristic
didDiscoverDescriptorsForCharacteristic
peripheral uuid:4BC36C6D-FC56-2019-3D56-860658F9FFC8
service uuid:53a20001-58df-4e45-ac57-573028548b13
uuid: 53a20003-58df-4e45-ac57-573028548b13 **value: <060000a5>**
uuid: 53a20002-58df-4e45-ac57-573028548b13 value: (null)
didUpdateNotificationStateForCharacteristic
didUpdateValueForCharacteristic 4BC36C6D-FC56-2019-3D56-860658F9FFC8
uuid: 53a20003-58df-4e45-ac57-573028548b13 value: **<020006de adbf99b1 6e63>**
didUpdateValueForCharacteristic 4BC36C6D-FC56-2019-3D56-860658F9FFC8
uuid: 53a20003-58df-4e45-ac57-573028548b13 value: <060000a5>
uuid: 53a20003-58df-4e45-ac57-573028548b13 value: <060000a5>

value: <020006de adbf99b1 6e63>， This is what I want, but if I listen in device. onValueChanged (notify_Characteristic). listen ((s){}), I can't get the return value, so I have to connect the device for the second time to display it.

bruceo-beflex, 2019-09-18:
any one same issue with this?

simondukaric, 2022-06-29:
Same issue here...