pauldemarco/flutter_blue #358: character.value observable dies after a few responses

markwilde11, 2019-09-04:
I am migrating from the 0.5 version. Everything was running properly when subscribing to a BLE character notification observable. After implementing the new API (I like it) the observable dies after a few notifications (between 4 and 20 notifactions). 

I created a new vanilla flutter app and implemented the basic code below to connect to the first available device and find the service / characteristic to connect to. After a couple of seconds the notification stream just stops without any logging in the debug console.

`
  void _incrementCounter() async {
    var discoveryStream = await FlutterBlue.instance
      .scan(
        // scan for all the known service types (Calypso / Screen)
        withServices: [
            new Guid('7A2C5500-C492-4B71-BA1B-000000000001'),
            new Guid('0000180D-0000-1000-8000-00805F9B34FB'),
            new Guid('0000180A-0000-1000-8000-00805F9B34FB')
          ]
      )
      .firstWhere((ScanResult result) => result != null && result.device != null);
    var device = discoveryStream.device;

    await device.connect();
    var calypsoServices = await device.discoverServices();
    var service = calypsoServices.firstWhere((service) => service.uuid.toString() == '0000180d-0000-1000-8000-00805f9b34fb');

    var char = service.characteristics.firstWhere(( BluetoothCharacteristic char) => char.uuid.toString() == '00002a39-0000-1000-8000-00805f9b34fb');
    await char.setNotifyValue(true);

    char.value.listen((list){
      setState(() {
        _counter++;
      });
    });
`

When I look in the Debug Console i have the following output

`
D/FlutterBluePlugin(23483): [onServicesDiscovered] count: 4 status: 0
D/BluetoothGatt(23483): setCharacteristicNotification() - uuid: 00002a39-0000-1000-8000-00805f9b34fb enable: true
D/BluetoothGatt(23483): onConnectionUpdated() - Device=F3:9F:8B:04:5B:54 interval=156 latency=0 timeout=400 status=0
D/FlutterBluePlugin(23483): [onDescriptorWrite] uuid: 00002902-0000-1000-8000-00805f9b34fb status: 0
D/FlutterBluePlugin(23483): [onCharacteristicChanged] uuid: 00002a39-0000-1000-8000-00805f9b34fb
D/FlutterBluePlugin(23483): [onCharacteristicChanged] uuid: 00002a39-0000-1000-8000-00805f9b34fb
D/FlutterBluePlugin(23483): [onCharacteristicChanged] uuid: 00002a39-0000-1000-8000-00805f9b34fb
D/FlutterBluePlugin(23483): [onCharacteristicChanged] uuid: 00002a39-0000-1000-8000-00805f9b34fb
D/FlutterBluePlugin(23483): [onCharacteristicChanged] uuid: 00002a39-0000-1000-8000-00805f9b34fb
D/FlutterBluePlugin(23483): [onCharacteristicChanged] uuid: 00002a39-0000-1000-8000-00805f9b34fb
D/FlutterBluePlugin(23483): [onCharacteristicChanged] uuid: 00002a39-0000-1000-8000-00805f9b34fb
D/FlutterBluePlugin(23483): [onCharacteristicChanged] uuid: 00002a39-0000-1000-8000-00805f9b34fb
D/FlutterBluePlugin(23483): [onCharacteristicChanged] uuid: 00002a39-0000-1000-8000-00805f9b34fb
D/FlutterBluePlugin(23483): [onCharacteristicChanged] uuid: 00002a39-0000-1000-8000-00805f9b34fb
D/FlutterBluePlugin(23483): [onCharacteristicChanged] uuid: 00002a39-0000-1000-8000-00805f9b34fb

`

Can anyone help pinpoint the issue? I suspect there is something wrong in the Flutter_Blue as the 0.5 version was working without any issues.

koleto, 2019-09-17:
I have absolutely the same problem here after I tried to upgrade to v. 0.6.+ 
What is interesting is that the ` print("Received: $data");` outputs the complete response but the print statement below `print('BUFFER: $buffer');` only a fraction of it.  

```   dart 

 StringBuffer buffer = new StringBuffer();

 var subscription = _readFileChar.value.transform(utf8.decoder)
        .listen((data) {
      print("Received: $data");
      buffer.write(data);
    }, onDone: () => print(buffer.toString()), onError: (e) => print(e));

    await _readFileChar.write(utf8.encode(recordName),
        withoutResponse: false);

    print('BUFFER: $buffer');
```

Dzunior, 2019-10-15:
Same here - sometimes it takes 10 seconds sometimes more (30 seconds) and it stops. Reconnecting BLE and connecting again shows that BLE data is properly sent (and characteristic changes).

markwilde11, 2019-11-16:
After a renewed attempt I found that my Bluetooth hardwarde was failing. Due to a faulty powermode the device disconnected after 10 attempts. Closing the issue

closed 2019-11-16 by markwilde11