pauldemarco/flutter_blue #361: Handling existing connections

FWeissenb, 2019-09-09:
Hi,

I have following problem: In my application, I want to connect to a BLE device. 
I have a list with devices and if a user clicks a list item I create a device and connect to it. 

In the connect method I listen to the state:
````
subscriptionHandleState = _currentDevice.state.listen(_handleConnection);
````

And in handleConnection I start discovering (and register notifications), handle disconnect, whatever...

My problem:
If a device is still connected and a user clicks another list item I want to disconnect and reconnect to the device (same device or another device). 

My disconnect:
```
await subscriptionHandleState?.cancel();
subscriptionHandleState = null;
await _currentDevice?.disconnect();  
```
I cancel the subscription handler because I dont't want to execute the _handleConnection with the disconnect event (because of it reconnects the current device).

This works fine with iOS. The old connection is canceled and the new connection is started.

But if I try this with Android my _handleConnection is called twice. That means in my situation that I start discovery and register notification twice and that I get some faults while reading values.
It works if I make a delay after my disconnect procedure
```
 await Future.delayed(const Duration(milliseconds: 1200), () {}); 
```

raphael-bmec-co, 2019-10-11:
Hi, same issue. 

I'm looking into :

`_deviceConnectStream = _bluetoothDevice.connect(timeout: Duration(seconds: _CONN_TIMEOUT_S)).timeout(Duration(seconds: _CONN_TIMEOUT_S)).asStream().listen((_){});`

I'll let you know how that works out.

raphael-bmec-co, 2019-10-11:
Okay this seems to work by giving you a handle to cancel the listener on the connect. Let me know if you need more info.

Maybe the library should be modified to return a StreamSubscription from the .connect method.