pauldemarco/flutter_blue #373: How to filter devices by serviceUuids?

0x77dev, 2019-09-19:
Like I have serviceUuid: daebb240-b041-11e4-9e45-0002a5d5c51b and i want to search only for these devices?

mortenboye, 2019-09-19:
You can set the withServices parameter:
`flutterBlue.startScan(scanMode: ScanMode.balanced, withServices: [Guid("yourSpecificServiceUUIDString")], timeout: Duration(seconds: 30)).catchError((error) {
      print("error starting scan $error");
    });`

Dzunior, 2019-09-30:
Strangely, when I setup withServices like in the example above, no device is seen. Without it I can connect to required device with that service guid (so guid is set correctly). Is it working properly in other's projects?

ericstephen, 2019-10-10:
If I am not mistaken unless you are specifically exposing that Service UUID in the advertisement data you cannot use it as a parameter to isolate your scanResult.

pauldemarco, 2019-10-18:
You can also use dart's built-in filter functionality for streams:
```dart
services.where((s) => s.uuid == someUuid)
```
In the future, please consider using a public forum like StackOverflow to ask questions.

closed 2019-10-18 by pauldemarco