pauldemarco/flutter_blue #378: Register notification

FWeissenb, 2019-09-23:
Hi,
What should happen if I register a notification? In my opinion, it should only trigger the callback method if the device triggers the notification. But in current flutter_blue I get a callback if I still register my notification.

With following code:
```
Future<StreamSubscription> _registerNotify(String serviceUUID, String characteristicUUID, void onData(List<int> data), {bool read = false, String debugText}) async {
    StreamSubscription retVal;
    if(debugText == null || debugText.isEmpty) debugText = "($serviceUUID | $characteristicUUID)";

     for (var service in _currentServices) {
      if (service.uuid != Guid(serviceUUID)) {
        continue;
      }
      for (var c in service.characteristics) {
        if (c.uuid == Guid(characteristicUUID)) {
          debugPrint(_debugPrefix + "$debugText found");

          if(read) {
             await c.read();
             debugPrint(_debugPrefix + "$debugText read");
          }
          
          debugPrint(_debugPrefix + "$debugText register notify");
          await c.setNotifyValue(true);
          retVal = c.value.listen(onData);
        
          if(read) {
             onData(val);
          }
        }
      }
    }

    return retVal;
  }
```

I thought if I call this method with read=false I will only get the onData if my device sends a notification. And if I use read=true I trigger my onData with the current value (not with the notification).
But onData is called every time. With read=false I get null (or last read) value and with read=true its called twice.
I hope someone can help me to understand my fault.

FWeissenb, 2019-09-23:
If I change 

```
bluetooth_characteristic.dart

  BehaviorSubject<List<int>> _value;
  Stream<List<int>> get value => Observable.merge([
        _value.stream,
        _onValueChangedStream,
      ]);
```

to 

```
  BehaviorSubject<List<int>> _value;
  Stream<List<int>> get value => Observable.merge([
        _onValueChangedStream,
      ]);
```

I get my expected behavior. 
The question is how can I avoid triggering the callback with the current value. I only need it if something changes (I don't want to filter this in my application with flags).

votruk, 2019-10-09:
I suppose that this issue is connected with this one: https://github.com/pauldemarco/flutter_blue/issues/311
I made a pull request to fix that but sadly, it seems that the creator of this repo does not merge any updates.

closed 2020-04-02 by pauldemarco