pauldemarco/flutter_blue #386: [iOS] fatal error: 'flutter_blue/FlutterBluePlugin.h' file not found

chrisdrobison, 2019-10-09:
I got your plugin working once somehow but, now I can't get it to build, even with a new project.

Steps:
1. Use beta flutter, v1.9.1+hotfix.4
1. `flutter create project`
2. Add the following to pubspec.yaml. Doing this in VSCode will automatically run `flutter pub get`
  `flutter_blue: ^0.6.2`
3. `flutter build ios`
  Fails with: ` /usr/local/bin/bash: line 3: /Users/chris/dev/cognuro/Pods/!ProtoCompiler/protoc: No such file or directory`
3. `ln -s ios/Pods Pods` - failes

```
Xcode's output:
↳
    4 warnings generated.
    /Users/chris/Tools/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.6.2/ios/Classes/FlutterBluePlugin.m:545:49: warning: 'CBManagerState' is only available on iOS 10.0 or newer
    [-Wunguarded-availability]
    - (ProtosBluetoothState*)toBluetoothStateProto:(CBManagerState)state {
                                                    ^
    In module 'CoreBluetooth' imported from /Users/chris/Tools/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.6.2/ios/Classes/FlutterBluePlugin.h:6:
    /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS13.0.sdk/System/Library/Frameworks/CoreBluetooth.framework/Headers/CBManager.h:32:28: note:
    'CBManagerState' has been marked as being introduced in iOS 10.0 here, but the deployment target is iOS 8.0.0
    typedef NS_ENUM(NSInteger, CBManagerState) {
                               ^
    /Users/chris/Tools/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.6.2/ios/Classes/FlutterBluePlugin.m:545:1: note: annotate 'toBluetoothStateProto:' with an availability attribute
    to silence this warning
    - (ProtosBluetoothState*)toBluetoothStateProto:(CBManagerState)state {
    ^
                                                                          API_AVAILABLE(ios(10.0))
    /Users/chris/Tools/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.6.2/ios/Classes/FlutterBluePlugin.m:106:52: warning: values of type 'NSUInteger' should not be used as format
    arguments; add an explicit cast to 'unsigned long' instead [-Wformat]
        NSLog(@"getConnectedDevices periphs size: %d", [periphs count]);
                                                  ~~   ^~~~~~~~~~~~~~~
                                                  %lu  (unsigned long)
    /Users/chris/Tools/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.6.2/ios/Classes/FlutterBluePlugin.m:74:37: warning: 'CBManagerStateUnsupported' is only available on iOS 10.0 or
    newer [-Wunguarded-availability]
        if(self.centralManager.state != CBManagerStateUnsupported && self.centralManager.state != CBManagerStateUnknown) {
                                        ^~~~~~~~~~~~~~~~~~~~~~~~~
    In module 'CoreBluetooth' imported from /Users/chris/Tools/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.6.2/ios/Classes/FlutterBluePlugin.h:6:
    /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS13.0.sdk/System/Library/Frameworks/CoreBluetooth.framework/Headers/CBManager.h:32:28: note:
    'CBManagerState' has been marked as being introduced in iOS 10.0 here, but the deployment target is iOS 8.0.0
    typedef NS_ENUM(NSInteger, CBManagerState) {
                               ^
    /Users/chris/Tools/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.6.2/ios/Classes/FlutterBluePlugin.m:74:37: note: enclose 'CBManagerStateUnsupported' in an @available check to
    silence this warning
        if(self.centralManager.state != CBManagerStateUnsupported && self.centralManager.state != CBManagerStateUnknown) {
                                        ^~~~~~~~~~~~~~~~~~~~~~~~~
    /Users/chris/Tools/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.6.2/ios/Classes/FlutterBluePlugin.m:74:95: warning: 'CBManagerStateUnknown' is only available on iOS 10.0 or newer
    [-Wunguarded-availability]
        if(self.centralManager.state != CBManagerStateUnsupported && self.centralManager.state != CBManagerStateUnknown) {
                                                                                                  ^~~~~~~~~~~~~~~~~~~~~
    In module 'CoreBluetooth' imported from /Users/chris/Tools/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.6.2/ios/Classes/FlutterBluePlugin.h:6:
    /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS13.0.sdk/System/Library/Frameworks/CoreBluetooth.framework/Headers/CBManager.h:32:28: note:
    'CBManagerState' has been marked as being introduced in iOS 10.0 here, but the deployment target is iOS 8.0.0
    typedef NS_ENUM(NSInteger, CBManagerState) {
                               ^
    /Users/chris/Tools/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.6.2/ios/Classes/FlutterBluePlugin.m:74:95: note: enclose 'CBManagerStateUnknown' in an @available check to silence
    this warning
        if(self.centralManager.state != CBManagerStateUnsupported && self.centralManager.state != CBManagerStateUnknown) {
                                                                                                  ^~~~~~~~~~~~~~~~~~~~~
    /Users/chris/Tools/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.6.2/ios/Classes/FlutterBluePlugin.m:80:37: warning: 'CBManagerStatePoweredOn' is only available on iOS 10.0 or
    newer [-Wunguarded-availability]
        if(self.centralManager.state == CBManagerStatePoweredOn) {
                                        ^~~~~~~~~~~~~~~~~~~~~~~
    In module 'CoreBluetooth' imported from /Users/chris/Tools/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.6.2/ios/Classes/FlutterBluePlugin.h:6:
    /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS13.0.sdk/System/Library/Frameworks/CoreBluetooth.framework/Headers/CBManager.h:32:28: note:
    'CBManagerState' has been marked as being introduced in iOS 10.0 here, but the deployment target is iOS 8.0.0
    typedef NS_ENUM(NSInteger, CBManagerState) {
                               ^
    /Users/chris/Tools/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.6.2/ios/Classes/FlutterBluePlugin.m:80:37: note: enclose 'CBManagerStatePoweredOn' in an @available check to
    silence this warning
        if(self.centralManager.state == CBManagerStatePoweredOn) {
                                        ^~~~~~~~~~~~~~~~~~~~~~~
    5 warnings generated.
    Building AOT snapshot in release mode (ios-release)...          
    Building App.framework for arm64...

    Building App.framework for armv7...
    Building AOT snapshot in release mode (ios-release)...             14.3s
    Built to build/aot/.
    warning: parsing line table prologue at offset 0x6f697463 found unsupported version 0x00
    warning: line table parameters mismatch. Cannot emit.
    note: while processing /Users/chris/dev/cognuro/build/aot/armv7/snapshot_assembly.o
    Project /Users/chris/dev/cognuro built and packaged successfully.
    /Users/chris/dev/cognuro/ios/Runner/GeneratedPluginRegistrant.m:6:9: fatal error: 'flutter_blue/FlutterBluePlugin.h' file not found
    #import <flutter_blue/FlutterBluePlugin.h>
            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    /Users/chris/dev/cognuro/ios/Runner/GeneratedPluginRegistrant.m:6:9: note: did not find header 'FlutterBluePlugin.h' in framework 'flutter_blue' (loaded from
    '/Users/chris/dev/cognuro/build/ios/Release-iphoneos/flutter_blue')
    1 error generated.
    error: the following command failed with exit code 1 but produced no further output
    CompileC
    /Users/chris/Library/Developer/Xcode/DerivedData/Runner-akubqslycmaxdqdmtwulxlsejbko/Build/Intermediates.noindex/Runner.build/Release-iphoneos/Runner.build/Objects-normal/arm64/GeneratedPlugi
    nRegistrant.o /Users/chris/dev/cognuro/ios/Runner/GeneratedPluginRegistrant.m normal arm64 objective-c com.apple.compilers.llvm.clang.1_0.compiler
    note: Using new build systemnote: Planning buildnote: Constructing build description
```

If I do `flutter clean` the warning go away, but the error remains.

Sivakorn2540, 2019-10-09:
i have problem like you.

vodovozovge, 2019-10-11:
I have the same issue. 
looks like the problem is with `Protos` subspec in cocoapods Podspec file
```
  s.subspec 'Protos' do |ss|
    ss.source_files = 'gen/**/*.pbobjc.{h,m}'
    ss.header_mappings_dir = 'gen'
    ss.requires_arc = false
    ss.dependency 'Protobuf'
  end
```

`ss.header_mappings_dir = 'gen'` line messes the visibility of the header files in the pods project.

I commented it out and compiled the project without the errors. 
```
  s.subspec 'Protos' do |ss|
    ss.source_files = 'gen/**/*.pbobjc.{h,m}'
    #ss.header_mappings_dir = 'gen'
    ss.requires_arc = false
    ss.dependency 'Protobuf'
  end
```
Works for my project, but could introduce some unintended side effects though

KlemenAndrasic, 2019-10-14:
Where do you find the file?

vodovozovge, 2019-10-14:
it's in the `.pub-cache` subfolder in the `flutter` main folder

in my case the path is `~/develop/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.6.2/ios/flutter_blue.podspec`

KlemenAndrasic, 2019-10-14:
@vodovozovge thanks it now builds just fine. I just hope they fix this as soon as possible. :) Thx again for the help :)

pauldemarco, 2019-10-18:
This should be resolved in the latest [version 0.6.3](https://pub.dev/packages/flutter_blue).
Please try it out and let me know if it works for you.
Thanks!

hanabi1224, 2019-10-18:
@pauldemarco No it's not working for me.  I think the problem is still the subspec part, since you have included generated pb files in source control, why not just remove the entire subspec part?

hanabi1224, 2019-10-18:
@pauldemarco  Hmmm..  after removing !ProtoCompiler it gives more errors that it cannot found protobuf built in types like Any etc.  Actually 0.6.2 works perfectly for me by just commenting out the single line ss.header_mappings_dir = 'gen'

pauldemarco, 2019-10-18:
@hanabi1224

Try again after removing Podfile and Podfile.lock.

I have a working version with those changes in PR 405, but I’m hesitant to add protoc back into it, since it seems the path resolving doesn’t work for everyone.

hanabi1224, 2019-10-18:
@pauldemarco the path thing was not configured properly before, I have fixed it in #403 , and have also verified the fix with CI build, please take a look

softmarshmallow, 2019-10-19:
Modifing the file still wont work for me. any updates?

hanabi1224, 2019-10-19:
@softmarshmallow If you made in place update of podspec in package cache with change 403 manually, try flutter clean then build

ag84ark, 2019-10-21:
Try one of this two:
https://github.com/pauldemarco/flutter_blue/issues/406#issuecomment-544501458
https://github.com/pauldemarco/flutter_blue/issues/406#issuecomment-544530920

thermoChas, 2019-10-28:
The only thing that worked for me was removing the iOS project, creating a new one and using Objective-C instead of Swift.

zwells, 2019-10-30:
You have to create a bridging header in xcode, see my steps for the solution here:
https://github.com/fengqiangboy/flutter-nordic-dfu/issues/4#issuecomment-532506690

Open the project with XCode. Then choose File -> New -> File -> Swift File.
A dialog will be displayed when creating the swift file(Since this file is deleted, any name can be used.). XCode will ask you if you wish to create Bridging Header, click yes.
Make sure you have use_frameworks! in the Runner block, in ios/Podfile。
Make sure you have SWIFT_VERSION 4.2 selected in you XCode -> Build Settings
Do flutter clean
Go to your ios folder, delete Podfile.lock and Pods folder and then execute pod install --repo-update

chrisdrobison, 2019-11-05:
@zwells The bridging header already exists in new swift flutter projects

zwells, 2019-11-05:
Sorry, I'm using Flutter 1.5 in order to use an over-the-air updates library that depends on flutter_blue 0.5.0

toupilsner, 2019-11-07:
Same issue here. New clean project

ericstumper, 2019-11-07:
The changes by hanabi1224 in #403 fix the issue for me on an objective-c based project with latest flutter stable 1.9.1+hotfix5. I also created a bridging header and have use_frameworks enabled and SWIFT_VERSION set to 4.2 in my podspec.

chrisdrobison, 2019-11-17:
It looks like the Flutter team has merged in a change to the plugin template to address some of these issues: https://github.com/flutter/flutter/pull/44324. Might be worth adding them to this repo.

caiovncius, 2019-11-21:
I needed do downgrade to `0.6.1` and comment the line `ss.header_mappings_dir = '.'`. Worked to me.

Doctor summary (to see all details, run flutter doctor -v):
[✓] Flutter (Channel stable, v1.9.1+hotfix.6, on Mac OS X 10.15.1 19B88, locale pt-BR)
 
[✓] Android toolchain - develop for Android devices (Android SDK version 28.0.3)
[✓] Xcode - develop for iOS and macOS (Xcode 11.2.1)

aytunch, 2019-11-24:
I can not run my project on iOS because of this same error. What is the cause? is there a workaround?

sergey-shuyskiy, 2019-11-28:
> I needed do downgrade to `0.6.1` and comment the line `ss.header_mappings_dir = '.'` in `flutter_blue.pubspec`. Worked to me.
> 
> Doctor summary (to see all details, run flutter doctor -v):
> [✓] Flutter (Channel stable, v1.9.1+hotfix.6, on Mac OS X 10.15.1 19B88, locale pt-BR)
> 
> [✓] Android toolchain - develop for Android devices (Android SDK version 28.0.3)
> [✓] Xcode - develop for iOS and macOS (Xcode 11.2.1)

thanks! this is working for 0.6.3+1

caiovncius, 2019-11-28:
I tried that. Worked to me too.

gpfister, 2020-05-18:
Hi,

I'm trying to build my iOS app using flutter_blue 0.7.2 and it still fails. Are there any plans to fix this issue ?

marcmathys, 2020-05-18:
It works for me.

gpfister, 2020-05-18:
Hi,

I'm getting the following output:

Failed to build iOS app
Error output from Xcode build:
↳
    ** BUILD FAILED **

Xcode's output:
↳
    4 warnings generated.
    While building module 'flutter_blue' imported from /Users/gpfister/Development/Workspaces/Flutter/oskey.io/flutter-oskey-dev/ios/Runner/GeneratedPluginRegistrant.m:34:
    In file included from <module-includes>:1:
    /Users/gpfister/Development/Workspaces/Flutter/oskey.io/flutter-oskey-dev/build/ios/Release-iphoneos/flutter_blue/flutter_blue.framework/Headers/flutter_blue-umbrella.h:13:
    9: fatal error: 'FlutterBluePlugin.h' file not found
    #import "FlutterBluePlugin.h"
            ^~~~~~~~~~~~~~~~~~~~~
    1 error generated.
    /Users/gpfister/Development/Workspaces/Flutter/oskey.io/flutter-oskey-dev/ios/Runner/GeneratedPluginRegistrant.m:34:9: fatal error: could not build module 'flutter_blue'
    @import flutter_blue;
     ~~~~~~~^~~~~~~~~~~~
    2 errors generated.
    note: Using new build system
    note: Building targets in parallel
    note: Planning build
    note: Constructing build description

Encountered error while building for device.

fengqiangboy, 2020-05-25:
> I have the same issue.
> looks like the problem is with `Protos` subspec in cocoapods Podspec file
>
> ```
>   s.subspec 'Protos' do |ss|
>     ss.source_files = 'gen/**/*.pbobjc.{h,m}' [...]

Thx, Also works for me!

elanpang, 2020-05-27:
> I have the same issue.
> looks like the problem is with `Protos` subspec in cocoapods Podspec file
>
> ```
>   s.subspec 'Protos' do |ss|
>     ss.source_files = 'gen/**/*.pbobjc.{h,m}' [...]

works for me, commented this line in the file

>  ~/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.7.2/ios/flutter_blue.podspec 
then run pod install again

KlemenAndrasic, 2020-05-27:
This is a solution that works but still doesn't solve the problem when using CI tools.

charlesomer, 2020-06-04:
Is this still being looked at? I'm using 0.7.2 and having what I believe is the same issue. So far I've been able to run the app in the iOS simulator but haven't been able to run on a physical device. I don't think commenting out code is the best solution since this should work from the docs. If anyone has any other ideas then please could you let me know.

rudeboyas, 2020-06-27:
I'm facing the same issue. someone, please help me to solve this issue.

justprodev, 2020-07-17:
> I'm facing the same issue. someone, please help me to solve this issue.

Updating cocoapods by 'sudo gem install cocoapods' & flutter pub upgrade fixed that issue for me

robinjohn, 2020-09-09:
Hi @chrisdrobison , could you let know if this issue is still valid for you ?

Following below comment from @pauldemarco , I was able to get past the similar build error that you faced. 
https://github.com/pauldemarco/flutter_blue/issues/406#issuecomment-545993733
Basically had to delete all the pods* files & directories, comment podspec line to use static lib(.a) instead of framework files and do a clean before building it again.

Help close the issue if you were able to build successfully. Thanks.

(flutter_blue version: 0.7.2)

chrisdrobison, 2020-09-16:
I'm not developing the app any more that required this and I think the person that is when with another solution so I couldn't say if this is an issue still or not.

jvigneshcs, 2020-12-18:
> @hanabi1224
> 
> Try again after removing Podfile and Podfile.lock.
> 
> I have a working version with those changes in PR 405, but I’m hesitant to add protoc back in, since it seems the path resolving doesn’t work for everyone.

@pauldemarco , Can you please include this in ReadMe?

meridorkobi, 2021-03-27:
I have the same issue.
In 0.6.3 & 0.7.3

Users/builder/programs/flutter_2_0_3/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.6.3+1/ios/Classes/FlutterBluePlugin.m:106:53: warning: values of type 'NSUInteger' should not be used as format arguments; add an explicit cast to 'unsigned long' instead [-Wformat]
        NSLog(@"getConnectedDevices periphs size: %lu", [periphs count]);
                                                  ~~~   ^~~~~~~~~~~~~~~
                                                  %lu   (unsigned long)

chrisdrobison, 2021-03-30:
For the benefit of everyone looking at this issue, in our project we’ve moved on. When we started this was the only BT plug-in available and now there are a few more well supported ones. I would look around at this point.

gpfister, 2021-03-30:
I second Chris comment. This package is not under maintenance, you need to either migrate to another one, or (and that would be my recommendation) build one yourself.

The reason is that Bluetooth is tied to hardware, package dependencies become a liability when the maintainer doesn't have your device best interests in mind, contrary to using a package to access an API, the package been maintained by the API provider is rather safe.

This is the direction I going to, and I want to keep control on it when Apple/Google upgrade their Bluetooth API.

I can appreciate the marketing behind Flutter is to promise a smooth multi device integration and a large community maintained package list, however many people have voiced that it has been actually a problem with these frameworks...

yuriboeira11tx, 2021-04-06:
really, have any suggestions?

chrisdrobison, 2021-04-06:
We've had success with this one: https://pub.dev/packages/flutter_ble_lib

yuriboeira11tx, 2021-04-06:
I tried to use this package, but I was not successful following the
tutorial, do you have any projects that I can be inspired by?

LongXiangGuo, 2021-05-17:
Remove the sub podSpec, it works well.

```
  s.dependency 'Flutter'
  s.platform = :ios, '8.0'
  s.framework = 'CoreBluetooth'
  
  # Fix the subspec `Protos` could not referenced in mobile-connected
  s.requires_arc = ['Classes/**/*.m']
  s.dependency "Protobuf", '~> 3.12.0'

  # s.subspec "Protos" do |ss|
  #   ss.source_files = "gen/*.pbobjc.{h,m}", "gen/**/*.pbobjc.{h,m}"
  #   ss.header_mappings_dir = "gen"
  #   ss.requires_arc = false
  #   ss.dependency "Protobuf", '~> 3.11.4'
  # end
```

mensur-durakovic, 2021-12-25:
this is still happening on version 0.8.0

[image]

Anyone knows the fix?

timtheteh, 2022-03-17:
Anyone knows the fix to this issue? I am facing the same problem as mensur-durakovic