pauldemarco/flutter_blue #387: iOS: flutter run failure with latest environment

espresso3389, 2019-10-09:
I'm facing the following error during flutter run or flutter build ios:

```
Unable to run command 'PBXCp Flutterblue.pbobjc.h' - this target might include its own product.
Unable to run command 'Touch flutter_blue.framework' - this target might include its own product.
```

```
[✓] Flutter (Channel master, v1.10.15-pre.23, on Mac OS X 10.15 19A582a, locale ja-JP)
    • Flutter version 1.10.15-pre.23 at /Users/kawasaki/flutter                                                            • Framework revision 440753bae8 (10 hours ago), 2019-10-08 21:33:17 -0700                                              • Engine revision 0d749933bc
• Dart version 2.6.0 (build 2.6.0-dev.0.0 f4a72bfc64)                                                                                                                                   
[✓] Xcode - develop for iOS and macOS (Xcode 11.1)                                                                         • Xcode at /Applications/Xcode.app/Contents/Developer                                                                  • Xcode 11.1, Build version 11A1027
• CocoaPods version 1.8.3                                                                   
```

I've also tested flutter stable but the behavior is same.

I suspect Xcode 11.1 or CocoaPods 1.8.3 is the cause but I cannot determine which one is the cause of the issue.

espresso3389, 2019-10-09:
Downgrading CocoaPods to 1.7.5 seems to workaround the issue for me.

julienzarka, 2019-10-17:
I have the same issue - same fix...

pauldemarco, 2019-10-18:
I've compiled and ran the example app successfully with the following setup:

```
[✓] Flutter (Channel master, v1.10.15-pre.163, on Mac OS X 10.14.6 18G95, locale
    en-US)
    • Flutter version 1.10.15-pre.163 at /Users/pauldemarco/flutter
    • Framework revision f53b32eb23 (7 hours ago), 2019-10-17 17:42:40 -0700
    • Engine revision 508146f0de
    • Dart version 2.6.0 (build 2.6.0-dev.8.0 f020ce5d23)

[✓] Xcode - develop for iOS and macOS (Xcode 10.3)
    • Xcode at /Applications/Xcode.app/Contents/Developer
    • Xcode 10.3, Build version 10G8
    • CocoaPods version 1.8.4

! Doctor found issues in 2 categories.
```

Please try the latest version 0.6.3 and let me know if that solves the issue.

bradd-pktlzyer, 2019-10-21:
I saw the same issues when I was working with #406  this weekend, @espresso3389 @julienzarka did you generate your flutter project with swift support?

espresso3389, 2019-10-21:
> I saw the same issues when I was working with #406 this weekend, @espresso3389 @julienzarka did you generate your flutter project with swift support?

Yes, I'm using Swift for iOS.

bradd-pktlzyer, 2019-10-21:
@espresso3389 try removing `use_frameworks!` from Podfile and see if that helps.

AAverin1994, 2019-10-22:
@bradd-pktlzyer I removed use_frameworks!. It doesn't work for me.

pauldemarco, 2019-10-24:
Hi all, I've created a dedicated Discord channel so we can debug these problems more efficiently:

[Discord Invite Link](https://discord.gg/K8qKgX)

ramisadek, 2020-01-07:
is anyone solved this yet ???

julienzarka, 2020-01-07:
I pushed a branch that works fine with me and I got positive feedback.
It is available to be merged.

oakmad, 2020-05-05:
Is there a definitive fix for this aside from down grading cocoapods? 

Using version 0.7.2, our project has swift support. 

`[✓] Flutter (Channel stable, v1.12.13+hotfix.9, on Mac OS X 10.15.4 19E287, locale en-US)
    • Flutter version 1.12.13+hotfix.9 at /Users/adamoakman/flutter
    • Framework revision f139b11009 (5 weeks ago), 2020-03-30 13:57:30 -0700
    • Engine revision af51afceb8
    • Dart version 2.7.2
[✓] Android toolchain - develop for Android devices (Android SDK version 28.0.3)
    • Android SDK at /Users/adamoakman/Library/Android/sdk
    • Android NDK location not configured (optional; useful for native profiling support)
    • Platform android-29, build-tools 28.0.3
    • Java binary at: /Applications/Android Studio.app/Contents/jre/jdk/Contents/Home/bin/java
    • Java version OpenJDK Runtime Environment (build 1.8.0_212-release-1586-b4-5784211)
    • All Android licenses accepted.

[✓] Xcode - develop for iOS and macOS (Xcode 11.4.1)
    • Xcode at /Applications/Xcode.app/Contents/Developer
    • Xcode 11.4.1, Build version 11E503a
    • CocoaPods version 1.9.1`