pauldemarco/flutter_blue #389: Update readme to include IOS 13 crash troubleshooting guidelines

hanabi1224, 2019-10-10:
https://developer.apple.com/documentation/corebluetooth

Important

Your app will crash if its Info.plist doesn’t include usage description keys for the types of data it needs to access, or it crashes. To access Core Bluetooth APIs on apps linked on or after iOS 13, include the NSBluetoothAlwaysUsageDescription key. In iOS 12 and earlier, include NSBluetoothPeripheralUsageDescription to access Bluetooth peripheral data.

mbn18, 2019-10-16:
For `IOS 13` I had to add to add this lines to `info.plist`.
I think it should be added to the install description.

```xml
<dic>
...
	<key>NSBluetoothAlwaysUsageDescription</key>
	<string>A meesage</string>
	<key>NSBluetoothPeripheralUsageDescription</key>
	<string>A message</string>
...
</dic>
```

happy49sky, 2019-10-28:
@mbn18 Thank you so much ! You save me ...