pauldemarco/flutter_blue #391: Reconnection error

SietseAchterop, 2019-10-11:
Hello,

I don't know whether this is the proper forum for this, but hopefully someone can point me in the right direction.

I am trying to connect a flutter app with an android phone to a arduino clone, but am often getting the following error:
```
E/flutter ( 2113): [ERROR:flutter/lib/ui/ui_dart_state.cc(148)] Unhandled Exception: PlatformException(already_connected, connection with device already exists,\
 null)
```

The question is of course, why this error. I don't see a reconnection occurring.
Maybe it can be ignored, but the app is not working, so it maybe is better to understand this error first.
The error occurs when I tap in the Textfield mentioned below.

And by not working I mean that if I type a command in the textfield, it gets send to the arduino clone, which answers it
with the proper reply. I conclude that because I have seen it working with an android app and on Linux using the bluepy library.
The listen only yields the string that that I typed in, and that often repeatedly. I never see the reply from the bluno.

I have to mention that the arduino clone is a Bluno m0, which uses a TI cc2540 with custom firmware.
Notification cannot properly be set using the 0x2902 descriptor. Notification is on by default.
I hope that this is not a problem with flutter_blue.
Please see the code description below.

   Thanks in advance,
       Sietse

==============
Now to the code.
I use the flutter_blue example with basically a single change, that is I now have another
version of `_buildServiceTiles()`.

The changes are as follows:
I made the DeviceScreen stateful with
  `  final BluetoothDevice device;`
as state.

Then, instead of the call to _buildServicesTiles I select the correct characteristic from the service
and call _buildReply with that characteristic:

```
               List<BluetoothService> servs = snapshot.data
                    .where((s) => (s.uuid.toString() == myservice))
                    .toList();
                if (servs.length == 0) {
                  return Text("No services found yet.");
                } else {
                  List<BluetoothCharacteristic> characs =
                      servs[0].characteristics;
                  if (characs[0].uuid.toString() != mycharacteristic)
                    print("Wrong characteristic, shouldn't happen!");

                  return Column( children: _buildReply(characs[0]), );
                }

```
And finally the _buildReply function:

```

String _result = "Nothing yet";
List<Widget> _buildReply(BluetoothCharacteristic c) {
    c.value.listen((v) {
      _result = utf8.decode(v);
      if (v.length > 0) {
        print("Answer: " + _result);
        print("   " + v.length.toString() + "  " + v[0].toString());
        setState(() {
        });
      }
    }, onDone: () {
      print("Task Done");
    }, onError: (error) {
      print("Some Error");
    });

    return <Widget>[
      TextField
        decoration: InputDecoration(hintText: "Enter command"),
        onSubmitted: (String str) {
          str += "\r";
          print("The string send is " + str);
          c.write(utf8.encode(str), withoutResponse: false);

          //setState(() {});
        },
      ),
      FlatButton(
        child: Text(_result),
      )
    ];
  }
```

And here a fragment of the logging with the error:
```

D/BluetoothManager( 2113): getConnectionState()
D/BluetoothManager( 2113): getConnectedDevices
E/flutter ( 2113): [ERROR:flutter/lib/ui/ui_dart_state.cc(148)] Unhandled Exception: PlatformException(already_connected, connection with device already exists,\
 null)
[38;5;244mE/flutter ( 2113): #0      StandardMethodCodec.decodeEnvelope[39;49m
[38;5;244mE/flutter ( 2113): #1      MethodChannel.invokeMethod[39;49m
E/flutter ( 2113): <asynchronous suspension>
[38;5;248mE/flutter ( 2113): #2      BluetoothDevice.connect[39;49m
E/flutter ( 2113): <asynchronous suspension>
[38;5;248mE/flutter ( 2113): #3      FindDevicesScreen.build.<anonymous closure>.<anonymous closure>.<anonymous closure>.<anonymous closure>[39;49m
[38;5;244mE/flutter ( 2113): #4      MaterialPageRoute.buildPage[39;49m
[38;5;244mE/flutter ( 2113): #5      _ModalScopeState.build.<anonymous closure>[39;49m

                  ............

[38;5;244mE/flutter ( 2113): #41     ComponentElement.performRebuild[39;49m
[38;5;244mE/flutter ( 2113): #42     Element.rebuild[39;49m
E/flutter ( 2113): #43     StatefulElement.update (package:flutter/src/widgets/framework.dart:4
D/BluetoothManager( 2113): getConnectionState()
D/BluetoothManager( 2113): getConnectedDevices

```

IAmSarthakVerma, 2019-10-14:
Hey @SietseAchterop What you can do here is that you check for any connected devices and then disconnect them. I have written a function that does just that, here it is:

```
checkConnectedDevices() async{

    connectedDevices = await flutterBlue.connectedDevices.whenComplete((){
      print(connectedDevices);
      for(int i = 0; i < connectedDevices.length; i++){
        connectedDevices[i].disconnect();
      }
    });

  }
```

closed 2023-01-05 by SietseAchterop

corteXnzt, 2023-06-16:
> Thanks @IAmSarthakVerma for the reply! I can't find a proper place to call it, and the DeviceScreen is only called when the device is already connected. And nothing is done there that should trigger a new connection. So if I succeed in using this fix, I only am covering up the actual problem, I'm afraid. Thanks again.
> 
> PS. Any idea why notifications are not being shown?

Hello @SietseAchterop , did you fix this? I am already connected to the device. I get connection with device already exists when I click the text field that I added which is weird. Why would ble reconnect each time I  click the text field? Whole UI freezes after that.

SietseAchterop, 2023-06-21:
No, I moved away from the bluno because the bluetooth implementation is not correct, non compliant. It only seems to work with the given android example code.

corteXnzt, 2023-06-21:
I'm using different board, running Nordic UART service on nrf52DK. Yeah they have closed source. Thanks for the reply @SietseAchterop