pauldemarco/flutter_blue #402: error: package androidx.core.app does not exist

Jbz797, 2019-10-18:
Since the 0.6.3 update, i have this error message and i can't build my app :

`/home/jeaden/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.6.3/android/src/main/java/com/pauldemarco/flutter_blue/FlutterBluePlugin.java:43: error: package androidx.core.app does not exist
import androidx.core.app.ActivityCompat;`

gitjohnmoon1234, 2019-10-18:
Do you have history version to download ? such as v 0.6.2   

[image]

Jbz797, 2019-10-18:
Okay, but what should I do?

gitjohnmoon1234, 2019-10-18:
> Okay, but what should I do?
could you send me a history version? whitch can run without androidx

kamilk91, 2019-10-18:
Yes its todays error. Just use older version and wait for author response.

Jbz797, 2019-10-18:
ok thanks

KevinRohn, 2019-10-18:
> Okay, but what should I do?

Just use an older version in the pubspec.yaml flutter_blue: ^0.6.2

kamilk91, 2019-10-18:
@KevinRohn  that was my advice, but i have problem now.. I tried to use 0.6.3, and when i switched to 0.6.2 and pub get i have

```
  location: class FlutterBluePlugin
C:\Users\kamil\Desktop\flutter\.pub-cache\hosted\pub.dartlang.org\flutter_blue-0.6.3\android\src\main\java\com\pauldemarco\flutter_blue\FlutterBluePlugin.java:152: error: cannot find symbol
                    ActivityCompat.requestPermissions(
                    ^
  symbol:   variable ActivityCompat
  location: class FlutterBluePlugin
Note: C:\Users\kamil\Desktop\flutter\.pub-cache\hosted\pub.dartlang.org\flutter_blue-0.6.3\android\src\main\java\com\pauldemarco\flutter_blue\FlutterBluePlugin.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: C:\Users\kamil\Documents\bluetoothmanageriqos\build\flutter_blue\generated\source\proto\debug\javalite\com\pauldemarco\flutter_blue\Protos.java uses unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
4 errors

FAILURE: Build failed with an exception.
```

Why its strill trying to use 0.6.3?

Jbz797, 2019-10-18:
I have downgraded successful. I put a fixed version in my pubspec.yaml : `flutter_blue: 0.6.2`.
Have you tried a `flutter clean` ?

kamilk91, 2019-10-18:
@Jbz797  yes i tried both, clean and change pubspec. i tried even 0.5.0 and delete folder flutter_blue.0.6.3 from cache

kamilk91, 2019-10-18:
Okay it was my mistake, i had to use ```flutter_blue: 0.6.2``` instead of ```flutter_blue: ^0.6.2 ```

pauldemarco, 2019-10-18:
Could it be because your project has outdated Android project files?

Run flutter upgrade 
Remove android and iOS folders in your project
Run “flutter create .” within your project to get new android project files

wadekb, 2019-10-21:
I just created a new Flutter project in Android Studio 3.5.1 and pulled in 0.6.3 and got the error.  Version from flutter doctor -

Flutter (Channel stable, v1.9.1+hotfix.5

alexcode9, 2019-10-21:
Same for me...
0.6.2 works, 0.6.3 breaks project with these errors
even on a clean new project created using `flutter create --androidx`

vincentsong, 2019-10-22:
@pauldemarco I got she same issue when try to build Android version

maheshwara-bwd, 2019-10-22:
@pauldemarco I too got the same issue in 0.6.3,
I also tried the same by running your example by cloning the flutter_blue repo, Still the same issue

KevinRohn, 2019-10-22:
Same problem here. 0.6.3 doesn't work for Android. I downgraded to 0.6.2 which works fine.

ElArtista, 2019-10-22:
Yep same here

MarkusPayne, 2019-10-22:
Can confirm same issue with 0.6.3.  I also tried cloning the repo and trying the example app.  0.6.2 does work for me.

yudhiApp, 2019-10-22:
I am new to flutter and have a fresh installation of android studio and flutter. I tried everthing mentioned in this thread but I get all kind many errors. I did what they instructed here https://github.com/pauldemarco/flutter_blue/issues/125, it did not work too. 

Can anyone summarise how to get flutter_blue to work step by step from scratch?

maheshwara-bwd, 2019-10-23:
Please try adding the required androidx dependencies in gradle file

android/build.gradle:

dependencies {
    implementation 'androidx.appcompat:appcompat:1.1.0'
    implementation 'androidx.core:core:1.1.0'
}

yudhiApp, 2019-10-23:
This might help others. I do not receive this error anymore **"package androidx.core.app does not exist**  
I have managed to make the example work as follows:
1. I created a new flutter project in android studio.
2. I deleted the test folder and deleted the main.dart in the lib folder.
3. I copied the main.dart and widgets.dart files found in "example/lib/" from the master repository to my lib folder.
4. Next I added   flutter_blue: 0.6.2 underneath cupertino_icons: ^0.1.2 in the dependencies in the pubspec.yaml and clicked packages get.
5. In android > app > build.gradle, I changed  minSdkVersion 16 to 19. 
6. Run it on my phone, got a couple of deprecation messages. The app runs fine and I can connect to the BLE device.

I need to mention that the new project I created was not automatically created in androidx and I did not migrate to androidx using Refactor.
I am new o flutter and the app dev world, will not migrating to androidx be a problem?

Jbz797, 2019-10-23:
The version of 'flutter_blue' that is a problem is version "0.6.3". And you have installed the "0.6.2".

Jbz797, 2019-10-25:
Some news about this issue ?

rjcasson, 2019-10-25:
I made a pull request (#416), but you can quickly fix this issue by adding the following line to the dependencies in android/build.grade of the flutter_blue plugin:

`implementation "androidx.core:core:1.1.0"`

00believers00, 2019-10-26:
> I made a pull request (#416), but you can quickly fix this issue by adding the following line to the dependencies in android/build.grade of the flutter_blue plugin:
> 
> `implementation "androidx.core:core:1.1.0"`

it's work!!
thanks

PhilColbert, 2019-10-26:
Thanks and thanks for paul for creating this plugin, nice one :)

KevinRohn, 2019-10-26:
Thank you this works for me!

CedYF, 2019-10-26:
I get could not find implementation @rjcasson

PhilColbert, 2019-10-26:
Maybe added in the wrong place , I did first.

Make sure its in the flutter blue package and not your own.

 in dir :- flutter\.pub-cache\hosted\pub.dartlang.org\flutter_blue-0.6.3+1\android

bus710, 2019-10-30:
The solution sounds like....

In ~/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.6.3+1/android/build.gradle,

At the line of 80, I could find: 
```
dependencies {
     implementation 'com.google.protobuf:protobuf-lite:3.0.1'
     implementation "androidx.core:core:1.1.0" // Add this line there.
}
```

pauldemarco, 2019-10-31:
Please follow the instructions here and let me know your results:
https://github.com/pauldemarco/flutter_blue/issues/415#issuecomment-548186492

jiyangjjj, 2019-11-05:
> @KevinRohn that was my advice, but i have problem now.. I tried to use 0.6.3, and when i switched to 0.6.2 and pub get i have
>
> ```
>   location: class FlutterBluePlugin [...]

I download 0.6.2 and changed the url in .packages to use 0.6.2.

tomcatvr, 2019-11-05:
> Please follow the instructions here and let me know your results:
> [#415 (comment)](https://github.com/pauldemarco/flutter_blue/issues/415#issuecomment-548186492)

Hi,
I've deleted and recreated the android project as mentioned in the link
Fluttersi: Channel master, v1.10.15-pre.404,
My flutter_blue build.gradle plugin is now:
dependencies {
    implementation 'com.google.protobuf:protobuf-lite:3.0.1'
    implementation 'androidx.core:core:1.1.0'
    implementation 'androidx.appcompat:appcompat:1.1.0'
}

But I still get an error:
error: incompatible types: Activity cannot be converted to AppCompatActivity
wich points to line 83 of FlutterBluePlugin.java
81    FlutterBluePlugin(Registrar r){
82        this.registrar = r;
83        this.activity = r.activity();

error is: 
incompatible types:
Required: androidx.appcompat.app.AppCompatActivity
Found: android.app.Activity

r.activity() is an Activity while this.activity is declared as AppCompatActivity
If at line 64 (og FlutterBluePlugin.java) I declare activity as Activity
before
private final AppCompatActivity activity;
now
private final Activity activity;

it works but I'm not sure if is the right way

bus710, 2019-11-05:
@tomcatvr,
Just want to know that if there is the pubspec.lock file.
I had the same error even after I followed the steps @pauldemarco provided. 
That was because the instruction introduces the steps from ground but from an existing project.

tomcatvr, 2019-11-05:
@bus710 
Yes, there is the pubspec.loc file.
I've recreated the project without it but still getting the same problem

bus710, 2019-11-05:
@tomcatvr,
I mean... you may delete the lock file if you try building from an existing project.

Anyway, I tried by myself as the instruction provided (#415) and this is the end result:

```
$ flutter build apk
You are building a fat APK that includes binaries for android-arm, android-arm64, android-x64.
If you are deploying the app to the Play Store, it's recommended to use app bundles or split the APK
to reduce the APK size.
    To generate an app bundle, run:
        flutter build appbundle --target-platform android-arm,android-arm64,android-x64
        Learn more on: https://developer.android.com/guide/app-bundle
    To split the APKs per ABI, run:
        flutter build apk --target-platform android-arm,android-arm64,android-x64 --split-per-abi
        Learn more on:
        https://developer.android.com/studio/build/configure-apk-splits#configure-abi-split
Note: /home/bus710/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.6.3+1/android/src/main/java/com/pauldemarco/flutter_blue/FlutterBluePlugin.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.                    
Note: /home/bus710/Desktop/aa/build/flutter_blue/generated/source/proto/release/javalite/com/pauldemarco/flutter_blue/Protos.java uses unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.                      
Removed unused resources: Binary resource data reduced from 41KB to 34KB: Removed 17%
Running Gradle task 'assembleRelease'...                                
                                                   
Running Gradle task 'assembleRelease'... Done                     130.0s (!)
✓ Built build/app/outputs/apk/release/app-release.apk (15.3MB).
```

My environment: 

```
$ flutter --version

Flutter 1.10.15-pre.408 • channel master • https://github.com/flutter/flutter.git
Framework • revision 7d03371610 (72 minutes ago) • 2019-11-05 08:57:25 -0800
Engine • revision 7b968ff95f
Tools • Dart 2.7.0
```

tomcatvr, 2019-11-06:
@bus710 
Thanks for the info. I've recreted once more the project and this time worked!
Maybe after deleting pubspec.lock I did not call flutter clean and so I was using a previous version of the library which used AppCompatActivity (but is only an hypothesis)

closed 2020-04-02 by pauldemarco