pauldemarco/flutter_blue #406: Problem building for iOS

ag84ark, 2019-10-19:
I am trying to build for iOS and i can not. I have the latest version of flutter_blue, I have created a new project and just added this plugin and when I try to build I get the error below.
I was using 0.6.2 version in my project and was working fine on Android, then tried it on iOS with no success, then I tried to update the version to 0.6.3 with no luck. Then I created a new project just to be sure that was nothing from my project that made the problem, and I got the same problem.
I looked through the issues, and could not get the problem resolved. Please help.
Thank you very much.

Flutter doctor:
`
Doctor summary (to see all details, run flutter doctor -v):
[✓] Flutter (Channel master, v1.10.15-pre.186, on Mac OS X 10.15 19A602, locale en-FR)
[✓] Android toolchain - develop for Android devices (Android SDK version 28.0.3)
[✓] Xcode - develop for iOS and macOS (Xcode 11.1)
[✓] Android Studio (version 3.5)
[✓] IntelliJ IDEA Ultimate Edition (version 2019.2.3)
[✓] VS Code (version 1.39.2)
[✓] Connected device (1 available)

• No issues found!
`

`
Running pod install...                                              1.6s
Running Xcode build...                                                  
                                                   
Xcode build done.                                           10.1s
Failed to build iOS app
Error output from Xcode build:
↳
    2019-10-19 16:35:17.770 xcodebuild[10186:1800491] [MT] PluginLoading: Required plug-in compatibility UUID 92CB09D8-3B74-4EF7-849C-99816039F0E7 for plug-in at path '~/Library/Application Support/Developer/Shared/Xcode/Plug-ins/MarmaladeGCC.xcplugin' not
    present in DVTPlugInCompatibilityUUIDs
    2019-10-19 16:35:17.770 xcodebuild[10186:1800491] Failed to load plugin at: /Users/cojogeo/Library/Application Support/Developer/Shared/Xcode/Plug-ins/MarmaladeGCC.xcplugin, skipping.  Reason for failure: *** -[__NSPlaceholderDictionary
    initWithObjects:forKeys:count:]: attempt to insert nil object from objects[0]
    ** BUILD FAILED **

Xcode's output:
↳
    While building module 'protobuf' imported from /Users/cojogeo/development/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.6.3/ios/gen/Flutterblue.pbobjc.m:13:
    In file included from <module-includes>:1:
    In file included from /Users/cojogeo/flutterApps/bluetoothtest1/ios/Pods/Target Support Files/Protobuf/Protobuf-umbrella.h:28:
    In file included from /Users/cojogeo/flutterApps/bluetoothtest1/ios/Pods/Protobuf/objectivec/GPBProtocolBuffers.h:44:
    /Users/cojogeo/flutterApps/bluetoothtest1/ios/Pods/Protobuf/objectivec/GPBWellKnownTypes.h:44:10: fatal error: 'google/protobuf/Any.pbobjc.h' file not found
     #import "google/protobuf/Any.pbobjc.h"
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    1 error generated.
    /Users/cojogeo/development/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.6.3/ios/gen/Flutterblue.pbobjc.m:13:10: fatal error: could not build module 'protobuf'
     #import "GPBProtocolBuffers_RuntimeSupport.h"
      ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    2 errors generated.
    note: Using new build systemnote: Planning buildnote: Constructing build description
`

ag84ark, 2019-10-19:
As an update, I managed to make the app work with the example provided on flutter_blue.
Maybe I can get some directions on something that needs to be done for iOS apps.
Again, thank!

NuclearGear, 2019-10-19:
> As an update, I managed to make the app work with the example provided on flutter_blue.
> Maybe I can get some directions on something that needs to be done for iOS apps.
> Again, thank!

Hi @ag84ark , could u pls tell me how to solve this problem?

thermoChas, 2019-10-20:
> fatal error:
>     could not build module 'protobuf'
>      #import "GPBProtocolBuffers_RuntimeSupport.h"

I am still getting this with a fresh install of 0.6.3. Running macOS 10.15 and iOS 13.1.2. Any ideas on this one?

ag84ark, 2019-10-21:
> > As an update, I managed to make the app work with the example provided on flutter_blue.
> > Maybe I can get some directions on something that needs to be done for iOS apps.
> > Again, thank!
> 
> Hi @ag84ark , could u pls tell me how to solve this problem?

I did not fix the problem, I just wanted to say if I try to create a project from the files in the example folder of flutter_blue, that compiles, but in my project or a new project that just has flutter_blue 0.6.x as a dependency, I can not build for iOS.
So I wanted to know what the author did extra on the example he provided so I can do it in my project as well to make it work.
Hope I explained better this time.

bradd-pktlzyer, 2019-10-21:
Ran into the same issue(Android worked for me as well), i also tried creating a brand new flutter project and included flutter_blue(0.6.3) to see if I could rule out my project, that did not work, I get the exact same errors as @ag84ark did using current stable flutter.

`Doctor summary (to see all details, run flutter doctor -v):
[✓] Flutter (Channel stable, v1.9.1+hotfix.5, on Mac OS X 10.14.6 18G95, locale en-US)
[✓] Android toolchain - develop for Android devices (Android SDK version 29.0.2)
[✓] Xcode - develop for iOS and macOS (Xcode 11.1)
[✓] Android Studio (version 3.5)
[✓] VS Code (version 1.39.0)
[✓] Connected device (1 available)

• No issues found!`

ag84ark, 2019-10-21:
Hi guys, again with an update!
I have managed to make the build work with something that I don't recommend :).
As I said earlier the example project from flutter_blue is working good on iOS. So what I did as an experiment, I copied all the files from the working example iOS folder and paste it into my own iOS folder overwriting all the files, and to my surprise the project compiled for iOS as well. Now I have some errors in the app, but that I think is my fault :)
One thing that I noticed that I think could maybe have something to do with my problem is that when I created the project I enabled swift for iOS but after I overwrote my original files from my limited objective C/swift knowledge the project went to ObjC. Maybe that is the problem for others as well.
I am now trying to recreate the iOS with ObjC as the default iOS lang to see if I still encounter this problem.

I hope that my solution helps you guys, or maybe helps you find the exact problem of why it isn't working.
I am still waiting for a good solution as this is not the best solution for the future of the project.
Waiting for some feedback from you
@bradd-pktlzyer @thermoChas @NuclearGear

ag84ark, 2019-10-21:
Steps to take to see if it's working for you as well.
1. Delete the iOS folder in your app
2. run: `flutter create -i objc` . from the **root** of the project folder 
This will change to ObjC from Swift.
After this compiled without any problems. 
The problem that I was getting when creating a new project was because now by default flutter enables Swift instead of ObjC.

Note to author: @pauldemarco project does not compile on iOS if Swift is enabled over ObjC in flutter.

bradd-pktlzyer, 2019-10-21:
@ag84ark Finally found a workaround for me, maybe give some others more clues. I was able to get things working with 0.6.3 by commenting out
`use_frameworks!`
from my Podfile. I noticed that @pauldemarco does not have this in his example Podfile as he probably generated Podfile without swift.

vincentsong, 2019-10-22:
@bradd-pktlzyer , thanks for this workaround, but if my project has other dependencies which require use_frameworks, then it will be failed.

The root cause is the preprocessor [GPB_USE_PROTOBUF_FRAMEWORK_IMPORTS]

I will try to fix it by adding it to podspec
```
'GCC_PREPROCESSOR_DEFINITIONS' => ['$(inherited)', 'GPB_USE_PROTOBUF_FRAMEWORK_IMPORTS=1']
```

NuclearGear, 2019-10-22:
> @bradd-pktlzyer , thanks for this workaround, but if my project has other dependencies which require use_frameworks, then it will be failed.
> 
> The root cause is the preprocessor [GPB_USE_PROTOBUF_FRAMEWORK_IMPORTS]
> 
> I tried to fix it by adding it to podspec, and it's working for my project
> 
> ```
> 'GCC_PREPROCESSOR_DEFINITIONS' => ['$(inherited)', 'GPB_USE_PROTOBUF_FRAMEWORK_IMPORTS=1']
> ```

hi @vincentsong , could u pls tell me how to solve this problem? which file should be edited? thx!

vincentsong, 2019-10-22:
@NuclearGear  you could put it into Podfile or  podspec

ChenChao-Android, 2019-10-23:
Can you describe in detail how to modify podfile?thx!@

> hi @vincentsong , could u pls tell me how to solve this problem? which file should be edited? thx!

ChenChao-Android, 2019-10-23:
hi @vincentsong , could u pls tell me  how to modify podfile?

pauldemarco, 2019-10-24:
Hi all, I've created a dedicated Discord channel so we can debug these problems more efficiently:

[Discord Invite Link](https://discord.gg/K8qKgX)

pauldemarco, 2019-10-24:
**Compiling flutter_blue on a Swift enabled Flutter project**

Flutter projects using Swift for the iOS platform may have trouble compiling with flutter_blue:
```
While building module 'flutter_blue' imported from
    /Users/paul/my_app/ios/Runner/GeneratedPluginRegistrant.m:10:
    In file included from <module-includes>:1:
    /Users/paul/my_app/build/ios/Debug-iphoneos/flutter_blue/flutter_blue.framework/Headers/fl
    utter_blue-umbrella.h:13:9: fatal error: 'FlutterBluePlugin.h' file not found
    #import "FlutterBluePlugin.h"
            ^~~~~~~~~~~~~~~~~~~~~
    1 error generated.
    /Users/paul/my_app/ios/Runner/GeneratedPluginRegistrant.m:10:9: fatal error: could not
    build module 'flutter_blue'
    @import flutter_blue;
     ~~~~~~~^~~~~~~~~~~~
    2 errors generated.
    note: Using new build systemnote: Planning buildnote: Constructing build description
```

From your project's root folder:

Clean everything:
```
rm -rf ios/Pods
rm ios/Podfile
rm ios/Podfile.lock
rm -rf ~/Library/Developer/Xcode/DerivedData/
flutter clean
flutter packages upgrade
```

Regenerate the Podfile with:
```
flutter run
```

Open `ios/Podfile` and comment the following line:
```
#use_frameworks!
```

The project should now run successfully with:
```
flutter run
```

CedYF, 2019-10-26:
I can't find that line in my ios podfile?

KevinRohn, 2019-10-28:
> Hi all, I've created a dedicated Discord channel so we can debug these problems more efficiently:
> 
> [Discord Invite Link](https://discord.gg/K8qKgX)

The invite link is invalid.

pauldemarco, 2019-10-28:
Here’s one that doesn’t expire:

https://discord.gg/Yk5Efra

ekuleshov, 2019-10-30:
@pauldemarco the removal of `use_frameworks!` is breaking ios build for me due to several other plugins requiring it. Are there other alternatives?

pauldemarco, 2019-10-31:
@ekuleshov At the moment, the alternative would be to make sure your flutter project was generated with objective-c for the ios side.  You can regenerate by deleting the ios folder (assuming you made no custom changes), and regenerating with:
```
flutter create -i objc .
```

I am working on finding an actual solution for using flutter_blue with projects generated to use swift.

KevinRohn, 2019-10-31:
Thank you for letting us know.

mikaelj, 2019-12-12:
> 
> Open `ios/Podfile` and comment the following line:
> 
> ```
> #use_frameworks!
> ```

Unfortunately - this is only a workaround. I have dependencies that *require* use_frameworks!.

Are you planning on fixing the issue?

ekuleshov, 2019-12-12:
> I am working on finding an actual solution for using flutter_blue with projects that use swift.

Any progress or update on a fix for this?

The related flutter framework issue https://github.com/flutter/flutter/issues/16049 has someone's notes on manually forcing swift (or objc) headers to be generated, which might be related to this, but I can't follow the steps

zoref, 2019-12-30:
Hi,
I add a similar error. It couldn't find "Flutterblue.pbobjc.h". After tryin all the solutions above with no luck, I ended up editing ios/Pods/Headers/Public/flutter_blue/flutter_blue-umbrella.h and replacing 

`#import "Flutterblue.pbobjc.h"`
by
`#import "gen/Flutterblue.pbobjc.h"`

and it seems to work.

boskokg, 2019-12-31:
just install the right cocoa and it will work.

syedadeel2, 2020-01-05:
> Steps to take to see if it's working for you as well.
>
> 1. Delete the iOS folder in your app
> 2. run: `flutter create -i objc` . from the **root** of the project folder
>    This will change to ObjC [...]

This actually works for me after wasting 2 days. Thank you

ekuleshov, 2020-01-05:
It works only if your project doesn't have swift code AND not using any swift-based plugins. If project uses swift plugins this solution is at no help. I believe @pauldemarco is aware

TzuHanLiang, 2020-01-10:
```open ios/Runner.xcworkspaces```

go to Pods > Build Phases  change ```flutter_blue-umbralla.h``` & ```FlutterBluePlugin.h``` to Public

then open Flutterblue.pbobjc.h
replacing

#import "gen/Flutterblue.pbobjc.h"
by
#import "Flutterblue.pbobjc.h"

TzuHanLiang, 2020-01-10:
open ```ios/Runner.xcworkspaces```
go to ```Pods > Build Phases``` 
move ```flutter_blue-umbralla.h``` & ```FlutterBluePlugin.h``` to Public Group
then edit ```Flutterblue.pbobjc.h```
replacing ```#import "gen/Flutterblue.pbobjc.h"```
by ```#import "Flutterblue.pbobjc.h"```

There you go!

JohnGorter, 2020-01-10:
Hi all, 

still having the error:    

While building module 'flutter_blue' imported from /.../Runner/GeneratedPluginRegistrant.m:10:
    In file included from <module-includes>:1:
    /.../build/ios/Debug-iphoneos/flutter_blue/flutter_blue.framework/Headers/flutter_blue-umbrella.h:13:9: fatal error: 'FlutterBluePlugin.h' file not found
    #import "FlutterBluePlugin.h"

when using CodeMagic to build the project... I've tried almost every step described above. Locally it builds fine but when using CI/CD like codemagic it still fails with this error. 

Does anyone have any insights about WHY this fails?

bradd-pktlzyer, 2020-01-10:
@JohnGorter I am building on CodeMagic, currently my workaround is I have a post clone script that uninstalls all of cocoapods and installs version 1.7.5 every build, this is working well for me. I am using flutter stable  and flutter_blue: 0.6.3+1. 

`gem uninstall cocoapods --all
gem install cocoapods -v 1.7.5`

JohnGorter, 2020-01-21:
that worked! Thanks very much

KevinRohn, 2020-01-22:
@bradd-pktlzyer  Wow thank you so much!

claudioafelix, 2020-01-27:
I am able to run a debug mode application, but if try to build a release in XCode got this error:

ld: warning: directory not found for option '-L /Library/Developer/Xcode/DerivedData/Runner-bsbiahtqtxjlsjbcltenmajuttqn/ArchiveIntermediates/Runner/BuildProductsPath/Release-iphoneos/shared_preferences'
ld: library not found for -lProtobuf
clang: error: linker command failed with exit code 1 (use -v to see invocation)

Doctor summary (to see all details, run flutter doctor -v):
[✓] Flutter (Channel stable, v1.12.13+hotfix.5, on Mac OS X 10.15.2 19C57, locale pt-BR)
 
[✓] Android toolchain - develop for Android devices (Android SDK version 29.0.2)
[✓] Xcode - develop for iOS and macOS (Xcode 11.3.1)
[✓] Android Studio (version 3.5)
[✓] VS Code (version 1.41.1)
[✓] Connected device (1 available)

• No issues found!

debasiskoley, 2020-02-27:
```
While building module 'flutter_blue' imported from /Users/debasis/works/flutter/projects/ble_wifi/ios/Runner/GeneratedPluginRegistrant.m:10:
In file included from <module-includes>:1:
/Users/debasis/Library/Developer/Xcode/DerivedData/Runner-gfkbodfpfhhsnqepptisvjdnimxi/Build/Products/Debug-iphoneos/flutter_blue/flutter_blue.framework/Headers/flutter_blue-umbrella.h:13:9: fatal error: 'FlutterBluePlugin.h' file not found
#import "FlutterBluePlugin.h"
        ^~~~~~~~~~~~~~~~~~~~~
1 error generated.
/Users/debasis/works/flutter/projects/ble_wifi/ios/Runner/GeneratedPluginRegistrant.m:10:9: fatal error: could not build module 'flutter_blue'
@import flutter_blue;
 ~~~~~~~^~~~~~~~~~~~
2 errors generated.
```

shaybc, 2020-03-03:
it was a real pain to get it going, i had to collect bits of answers to pass the run stage, but eventually got it to work, here is a complete ordered list of what i did:

(basically i started with a flutter hello world)

`
1. flutter create blue_test
2. cd blue_test
3. add all your code and dependencies to pubspec.yaml (without flutter_blue)
4. flutter clean (in case it is another run or existing project)
5. flutter packages get
6. flutter create -i objc .
7. flutter run (make sure it runs properly without the flutter_blue dependency)
8. add to pubspec.yaml:
     flutter_blue: ^0.6.3+1
9. sudo gem uninstall cocoapods --all
10. sudo gem install cocoapods
11. pod cache clean --all
12. in the Podfile (found under ios) add this at the top of the file:
       platform :ios, '10.0'
13. make sure in xcode - Runner->project->Runner->info->Configurations-> debug,release and profile are set to none, close the xcode
14. cd ios
15. pod install
16. cd ..
17. add to info.plist (under ios/Runner after the <dict> tag):
      
      <key>NSLocationWhenInUseUsageDescription</key>
      <string>Reason why app needs location</string>
      
      
      <key>NSLocationAlwaysAndWhenInUseUsageDescription</key>
      <string>Reason why app needs location</string>
      
      <key>NSLocationAlwaysUsageDescription</key>
      <string>Reason why app needs location</string>
      
      
      <key>NSBluetoothAlwaysUsageDescription</key>
      <string>Reason why app needs bluetooth</string>
18. flutter run
`

if for some reason this flow still wont work for you (because you use swift code, or have different dependecies that cross versions...) - these are the steps others did as well (but wasn't required for my case), you can try some of them or all of them, they can't hurt the ios project:

`
* in the project root folder run:
   ln -s ios/Pods Pods

* from xcode, make sure Runner->project->Runner->info->deployment target is 10.0

* make sure project Runner->Targets->Runner->Signing & Capabilities has a team selected

* [this solution](https://github.com/pauldemarco/flutter_blue/issues/406#issuecomment-572910336)

* note: this tip can break your project, so be carefull and backup
   - cd ios 
   - edit ios/Podfile
   - remove the line: use_frameworks!
`

hope this will help someone

ride4sun, 2020-03-04:
@shaybc I just never see a podfile in the IOS folder under you line item 12.
The file never gets generated.  Any idea?

shaybc, 2020-03-18:
> @shaybc I just never see a podfile in the IOS folder under you line item 12.
> The file never gets generated. Any idea?

assuming:
- you are building the project from a Mac and not from a PC
- you have CocoaPods installed (if not then run this command from terminal:) `sudo gem install cocoapods`
- you added your project dependencies to the `pubspec.yaml` file (after the `dependencies:` section with 2 spaces indention before each dependency)

then when you do `flutter build ios` (from the root project directory) you should see the created files in ios folder:
```
- Podfile
- Podfile.lock
```

if you still don't see it, then you can do this:
```
1. manually add a Pod file
2. cd ios
3. pod deintegrate
4. pod setup
5. pod install
```

Podfile example:
```
# Uncomment this line to define a global platform for your project
platform :ios, '10.0'

# CocoaPods analytics sends network stats synchronously affecting flutter build latency.
ENV['COCOAPODS_DISABLE_STATS'] = 'true'

project 'Runner', {
  'Debug' => :debug,
  'Profile' => :release,
  'Release' => :release,
}

def parse_KV_file(file, separator='=')
  file_abs_path = File.expand_path(file)
  if !File.exists? file_abs_path
    return [];
  end
  generated_key_values = {}
  skip_line_start_symbols = ["#", "/"]
  File.foreach(file_abs_path) do |line|
    next if skip_line_start_symbols.any? { |symbol| line =~ /^\s*#{symbol}/ }
    plugin = line.split(pattern=separator)
    if plugin.length == 2
      podname = plugin[0].strip()
      path = plugin[1].strip()
      podpath = File.expand_path("#{path}", file_abs_path)
      generated_key_values[podname] = podpath
    else
      puts "Invalid plugin specification: #{line}"
    end
  end
  generated_key_values
end

target 'Runner' do
  use_frameworks!
  use_modular_headers!
  
  # Flutter Pod

  copied_flutter_dir = File.join(__dir__, 'Flutter')
  copied_framework_path = File.join(copied_flutter_dir, 'Flutter.framework')
  copied_podspec_path = File.join(copied_flutter_dir, 'Flutter.podspec')
  unless File.exist?(copied_framework_path) && File.exist?(copied_podspec_path)
    # Copy Flutter.framework and Flutter.podspec to Flutter/ to have something to link against if the xcode backend script has not run yet.
    # That script will copy the correct debug/profile/release version of the framework based on the currently selected Xcode configuration.
    # CocoaPods will not embed the framework on pod install (before any build phases can generate) if the dylib does not exist.

    generated_xcode_build_settings_path = File.join(copied_flutter_dir, 'Generated.xcconfig')
    unless File.exist?(generated_xcode_build_settings_path)
      raise "Generated.xcconfig must exist. If you're running pod install manually, make sure flutter pub get is executed first"
    end
    generated_xcode_build_settings = parse_KV_file(generated_xcode_build_settings_path)
    cached_framework_dir = generated_xcode_build_settings['FLUTTER_FRAMEWORK_DIR'];

    unless File.exist?(copied_framework_path)
      FileUtils.cp_r(File.join(cached_framework_dir, 'Flutter.framework'), copied_flutter_dir)
    end
    unless File.exist?(copied_podspec_path)
      FileUtils.cp(File.join(cached_framework_dir, 'Flutter.podspec'), copied_flutter_dir)
    end
  end

  # Keep pod path relative so it can be checked into Podfile.lock.
  pod 'Flutter', :path => 'Flutter'

  # Plugin Pods

  # Prepare symlinks folder. We use symlinks to avoid having Podfile.lock
  # referring to absolute paths on developers' machines.
  system('rm -rf .symlinks')
  system('mkdir -p .symlinks/plugins')
  plugin_pods = parse_KV_file('../.flutter-plugins')
  plugin_pods.each do |name, path|
    symlink = File.join('.symlinks', 'plugins', name)
    File.symlink(path, symlink)
    pod name, :path => File.join(symlink, 'ios')
  end
end

# Prevent Cocoapods from embedding a second Flutter framework and causing an error with the new Xcode build system.
install! 'cocoapods', :disable_input_output_paths => true

post_install do |installer|
  installer.pods_project.targets.each do |target|
    target.build_configurations.each do |config|
      config.build_settings['ENABLE_BITCODE'] = 'NO'
    end
  end
end

```

read more ideas here: [https://stackoverflow.com/a/53492243/530884](url)

ride4sun, 2020-03-19:
@debasiskoley 
I did read somewhere to add a deepclean script like this:

pod repo update
rm ios/Podfile
rm ios/Podfile.lock
rm -rf ios/Pods/
rm .flutter-plugins
rm .packages
rm pubspec.lock
flutter packages get
flutter clean

After that the pod file got created.

aam074, 2020-03-31:
@pauldemarco 0.7.0 still does not build on iOS, same error about gen/Flutterblue.pbobjc.h. Any chances for a fix?

b055man, 2020-03-31:
@pauldemarco 0.7.0 still does not build on iOS, same error about gen/Flutterblue.pbobjc.h. Any chances for a fix?

pauldemarco, 2020-03-31:
@b055man Are you available for a discord chat sometime today so I can investigate the issue?
http://discord.gg/Yk5Efra

b055man, 2020-04-01:
@pauldemarco joined.

ride4sun, 2020-04-01:
There was a problem with xcode which just got release  - make sure you update

pauldemarco, 2020-04-01:
@ride4sun any links to the xcode problem?

pauldemarco, 2020-04-01:
I just committed some code and built the example project and swift project successfully, without any workarounds. Please test your projects by targeting the master branch of flutter in your pubspec, like so:
```
dependencies:
  flutter_blue:
    git: git://github.com/pauldemarco/flutter_blue.git
```
Let me know the results.

Thanks!

ride4sun, 2020-04-02:
> @ride4sun any links to the xcode problem?

https://github.com/flutter/flutter/issues/49504#issuecomment-592767041

prashant3285, 2020-04-03:
@pauldemarco build now working on iOS

b055man, 2020-04-03:
@pauldemarco yup, I'm able to build as well. Thanks!

ride4sun, 2020-04-09:
@pauldemarco I think that can be closed

ekuleshov, 2020-04-09:
I still can't make ios build work when pubspec has mix of ios and swift plugin dependencies

ride4sun, 2020-04-09:
If u take flutter_blue out of the picture can u build than?

pauldemarco, 2020-04-09:
@ekuleshov let me know when you’re available for a discord session so we can tackle the problem in faster iterations.

ekuleshov, 2020-04-15:
With @pauldemarco help my project is also building for ios with mix of objc and swift plugin dependencies (namely flutter_blue and flutter_tss).

Quick check list:

- make sure to use flutter stable - `flutter channel stable`
- the latest cocoapods 1.9.1
- Xcode 11.4
- re-generate ios with swift code - `rm -rf ios` and `flutter create -i swift .`

LiJiyaoisrunning, 2020-04-24:
> @JohnGorter I am building on CodeMagic, currently my workaround is I have a post clone script that uninstalls all of cocoapods and installs version 1.7.5 every build, this is working well for me. I [...]

@bradd-pktlzyer Thanks a lot You save me from hell !!

gujianxing, 2020-05-12:
> Steps to take to see if it's working for you as well.
>
> 1. Delete the iOS folder in your app
> 2. run: `flutter create -i objc` . from the **root** of the project folder
>    This will change to ObjC [...]

It works for me, create app in Objc, not swift.

decolon, 2020-08-20:
I also have this problem.  

When I build with Android, I have no problems.  When I build without flutter_blue in my pubspec, no problems.  As soon as I add flutter_blue in my pubspec, and build for my iPhone, I get this error 

```
Running "flutter pub get" in nobobo_pantry...
Launching lib/main.dart on iPhone (4) in debug mode...
Automatically signing iOS for device deployment using specified development team in Xcode project: 8R4KTRC86F
Running pod install...
Error: CocoaPods's specs repository is too out-of-date to satisfy dependencies.
To update the CocoaPods specs, run:
  pod repo update

Exception: Error running pod install
CocoaPods' output:
↳
  Preparing

Analyzing dependencies

Inspecting targets to integrate
  Using `ARCHS` setting to build architectures of target `Pods-Runner`: (``)

Finding Podfile changes
  A flutter_blue
  - Flutter
  - camera
  - cloud_firestore
  - firebase_auth
  - firebase_core
  - firebase_ml_vision
  - firebase_storage
  - flutter_exif_rotation
  - path_provider
  - vibration

Fetching external sources
-> Fetching podspec for `Flutter` from `Flutter`
-> Fetching podspec for `camera` from `.symlinks/plugins/camera/ios`
-> Fetching podspec for `cloud_firestore` from `.symlinks/plugins/cloud_firestore/ios`
-> Fetching podspec for `firebase_auth` from `.symlinks/plugins/firebase_auth/ios`
-> Fetching podspec for `firebase_core` from `.symlinks/plugins/firebase_core/ios`
-> Fetching podspec for `firebase_ml_vision` from `.symlinks/plugins/firebase_ml_vision/ios`
-> Fetching podspec for `firebase_storage` from `.symlinks/plugins/firebase_storage/ios`
-> Fetching podspec for `flutter_blue` from `.symlinks/plugins/flutter_blue/ios`
-> Fetching podspec for `flutter_exif_rotation` from `.symlinks/plugins/flutter_exif_rotation/ios`
-> Fetching podspec for `path_provider` from `.symlinks/plugins/path_provider/ios`
-> Fetching podspec for `vibration` from `.symlinks/plugins/vibration/ios`

Resolving dependencies of `Podfile`
  CDN: trunk Relative path: CocoaPods-version.yml exists! Returning local because checking is only perfomed in repo update
  CDN: trunk Relative path: all_pods_versions_0_3_5.txt exists! Returning local because checking is only perfomed in repo update
  CDN: trunk Relative path: Specs/0/3/5/Firebase/6.30.0/Firebase.podspec.json exists! Returning local because checking is only perfomed in repo update
  CDN: trunk Relative path: Specs/0/3/5/Firebase/6.30.0/Firebase.podspec.json exists! Returning local because checking is only perfomed in repo update
  CDN: trunk Relative path: Specs/0/3/5/Firebase/6.30.0/Firebase.podspec.json exists! Returning local because checking is only perfomed in repo update
  CDN: trunk Relative path: Specs/0/3/5/Firebase/6.30.0/Firebase.podspec.json exists! Returning local because checking is only perfomed in repo update
  CDN: trunk Relative path: Specs/0/3/5/Firebase/6.30.0/Firebase.podspec.json exists! Returning local because checking is only perfomed in repo update
  CDN: trunk Relative path: Specs/0/3/5/Firebase/6.30.0/Firebase.podspec.json exists! Returning local because checking is only perfomed in repo update
  CDN: trunk Relative path: Specs/0/3/5/Firebase/6.30.0/Firebase.podspec.json exists! Returning local because checking is only perfomed in repo update
  CDN: trunk Relative path: all_pods_versions_6_3_6.txt exists! Returning local because checking is only perfomed in repo update
  CDN: trunk Relative path: Specs/6/3/6/FirebaseAuth/6.8.0/FirebaseAuth.podspec.json exists! Returning local because checking is only perfomed in repo update
  CDN: trunk Relative path: all_pods_versions_e_c_d.txt exists! Returning local because checking is only perfomed in repo update
  CDN: trunk Relative path: Specs/e/c/d/Protobuf/4.0.0-rc1/Protobuf.podspec.json exists! Returning local because checking is only perfomed in repo update
[!] CocoaPods could not find compatible versions for pod "Protobuf":
  In snapshot (Podfile.lock):
    Protobuf (= 3.13.0, ~> 3.12)

  In Podfile:
    flutter_blue (from `.symlinks/plugins/flutter_blue/ios`) was resolved to 0.0.1, which depends on
      flutter_blue/Protos (= 0.0.1) was resolved to 0.0.1, which depends on
[... 63 lines cut ...]
    • Platform android-30, build-tools 30.0.0
    • Java binary at: /Applications/Android Studio.app/Contents/jre/jdk/Contents/Home/bin/java
    • Java version OpenJDK Runtime Environment (build 1.8.0_242-release-1644-b3-6222593)
    • All Android licenses accepted.

[✓] Xcode - develop for iOS and macOS (Xcode 11.6)
    • Xcode at /Applications/Xcode.app/Contents/Developer
    • Xcode 11.6, Build version 11E708
    • CocoaPods version 1.9.3

[✓] Android Studio (version 4.0)
    • Android Studio at /Applications/Android Studio.app/Contents
    • Flutter plugin version 48.1.2
    • Dart plugin version 193.7361
    • Java version OpenJDK Runtime Environment (build 1.8.0_242-release-1644-b3-6222593)

[✓] VS Code (version 1.46.1)
    • VS Code at /Applications/Visual Studio Code.app/Contents
    • Flutter extension version 3.12.0

[✓] Connected device (1 available)
    • iPhone (4) (mobile) • 5ae9753c41cae0609366ed62da85ce52106a6082 • ios • iOS 13.6

• No issues found!
```

anakin-jin, 2020-09-17:
> @bradd-pktlzyer Thanks a lot You save me from hell !!

 this is the correct way 

later i installed the 1.8.0, it works well too.

[✓] Flutter (Channel master, 1.22.0-10.0.pre.121, on Mac OS X 10.15.6 19G2021, locale en-CN)
[✓] Android toolchain - develop for Android devices (Android SDK version 29.0.2)
[✓] Xcode - develop for iOS and macOS (Xcode 11.6)
[✓] Chrome - develop for the web
[✓] Android Studio (version 4.0)
[✓] IntelliJ IDEA Ultimate Edition (version 2019.2.1)
[✓] VS Code (version 1.48.2)
[✓] Connected device (3 available)

dangabdolfo, 2020-10-13:
Hi every one.
I found another woraround which permit to keep use_frameworks! uncommented. Because in my case, if I comment 'use_frameworks!', that cause some issues with other plugins.

The workaround consist to insert in  your Podfile this lines:
`pre_install do |installer|
  installer.pod_targets.each do |pod|
    if (pod.name.downcase.include? "protobuf") || (pod.name.downcase.include? "fluter_blue")
      def pod.static_framework?;
          true
      end
      def pod.build_as_static_framework?;
          true
      end
    end
  end
end`

End for Xcode 12, add this lines in the post _install

`target.build_configurations.each do |bc|
      bc.build_settings['IPHONEOS_DEPLOYMENT_TARGET'] = '11.0'
    end`