pauldemarco/flutter_blue #407: iOS build problems, help me, I

ZhangChengmin99, 2019-10-20:
Xcode's output:
↳
    While building module 'protobuf' imported from /Users/zhangchengmin/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.6.3/ios/gen/Flutterblue.pbobjc.m:13:
    In file included from <module-includes>:1:
    In file included from /Users/zhangchengmin/AndroidStudioProjects/flutter_app/ios/Pods/Target Support Files/Protobuf/Protobuf-umbrella.h:28:
    In file included from /Users/zhangchengmin/AndroidStudioProjects/flutter_app/ios/Pods/Protobuf/objectivec/GPBProtocolBuffers.h:44:
    /Users/zhangchengmin/AndroidStudioProjects/flutter_app/ios/Pods/Protobuf/objectivec/GPBWellKnownTypes.h:44:10: fatal error: 'google/protobuf/Any.pbobjc.h' file not found
     #import "google/protobuf/Any.pbobjc.h"
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    1 error generated.
    /Users/zhangchengmin/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.6.3/ios/gen/Flutterblue.pbobjc.m:13:10: fatal error: could not build module 'protobuf'
     #import "GPBProtocolBuffers_RuntimeSupport.h"
      ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    2 errors generated.
    2 errors generated.
    /Users/zhangchengmin/AndroidStudioProjects/flutter_app/ios/Runner/GeneratedPluginRegistrant.m:6:9: fatal error: 'flutter_blue/FlutterBluePlugin.h' file not found
    #import <flutter_blue/FlutterBluePlugin.h>
            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    /Users/zhangchengmin/AndroidStudioProjects/flutter_app/ios/Runner/GeneratedPluginRegistrant.m:6:9: note: did not find header 'FlutterBluePlugin.h' in framework 'flutter_blue' (loaded from '/Users/zhangchengmin/AndroidStudioProjects/flutter_app/build/ios/Debug-iphonesimulator/flutter_blue')
    1 error generated.
    fatal error: lipo: can't open input file: /Users/zhangchengmin/AndroidStudioProjects/flutter_app/build/ios/Debug-iphonesimulator/Runner.app/Frameworks/flutter_blue.framework/flutter_blue (No such file or directory)
    fatal error: lipo: can't open input file: /Users/zhangchengmin/AndroidStudioProjects/flutter_app/build/ios/Debug-iphonesimulator/Runner.app/Frameworks/flutter_blue.framework/flutter_blue (No such file or directory)
    Failed to extract x86_64 for /Users/zhangchengmin/AndroidStudioProjects/flutter_app/build/ios/Debug-iphonesimulator/Runner.app/Frameworks/flutter_blue.framework/flutter_blue. Running lipo -info:
    fatal error: lipo: can't open input file: /Users/zhangchengmin/AndroidStudioProjects/flutter_app/build/ios/Debug-iphonesimulator/Runner.app/Frameworks/flutter_blue.framework/flutter_blue (No such file or directory)
    Command PhaseScriptExecution failed with a nonzero exit code
    note: Using new build systemnote: Planning buildnote: Constructing build description

ZhangChengmin99, 2019-10-20:
my flutter_blue 's version is 0.6.3

ZhangChengmin99, 2019-10-20:
[✓] Flutter (Channel stable, v1.9.1+hotfix.6-pre.1, on Mac OS X 10.15 19A602,
locale zh-Hans-CN)

[✓] Android toolchain - develop for Android devices (Android SDK version 29.0.0)
[✓] Xcode - develop for iOS and macOS (Xcode 11.1)
[✓] Android Studio (version 3.5)
[✓] VS Code (version 1.39.2)
[✓] Connected device (1 available)

vincentsong, 2019-10-21:
Because Xcode building system try to flat the path of header files, you could change from this below

```
#import "google/protobuf/Any.pbobjc.h"
```
to
```
#import "Any.pbobjc.h"
```

ZhangChengmin99, 2019-10-21:
> Because Xcode building system try to flat the path of header files, you could change the path below
> 
> ```
> #import "google/protobuf/Any.pbobjc.h"
> ```
> 
> to
> 
> ```
> #import "Any.pbobjc.h"
> ```

I follow your tips and solve some problems, but there is remaining an error in Users/zhangchengmin/AndroidStudioProjects/flutter_app/ios/Runner/GeneratedPluginRegistrant.m. This is an auto-generated file. The error's detail is below.

Xcode's output:
↳
    /Users/zhangchengmin/AndroidStudioProjects/flutter_app/ios/Runner/GeneratedPluginRegistrant.m:6:9: fatal error: 'flutter_blue/FlutterBluePlugin.h' file not found
    #import <flutter_blue/FlutterBluePlugin.h>
            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    /Users/zhangchengmin/AndroidStudioProjects/flutter_app/ios/Runner/GeneratedPluginRegistrant.m:6:9: note: did not find header 'FlutterBluePlugin.h' in framework 'flutter_blue' (loaded from '/Users/zhangchengmin/AndroidStudioProjects/flutter_app/build/ios/Debug-iphonesimulator/flutter_blue')
    1 error generated.
    note: Using new build systemnote: Planning buildnote: Constructing build description

And when I try to delete the flutter_blue/ in  '#import <flutter_blue/FlutterBluePlugin.h>' , it will regenerates.

vincentsong, 2019-10-21:
Then it's another issue, have you tried to clean and rebuild?

ZhangChengmin99, 2019-10-21:
> Then it's another issue, have you tried to clean and rebuild?

I will try to clean and rebuild after 3 hours. Thank you very much.

ZhangChengmin99, 2019-10-21:
> Then it's another issue, have you tried to clean and rebuild?

I try to clean and rebuild, but it remains the same error.

Xcode's output:
↳
    /Users/zhangchengmin/AndroidStudioProjects/flutter_app/ios/Runner/GeneratedPluginRegistrant.m:6:9: fatal error: 'flutter_blue/FlutterBluePlugin.h' file not found
    #import <flutter_blue/FlutterBluePlugin.h>
            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    /Users/zhangchengmin/AndroidStudioProjects/flutter_app/ios/Runner/GeneratedPluginRegistrant.m:6:9: note: did not find header 'FlutterBluePlugin.h' in framework 'flutter_blue' (loaded from '/Users/zhangchengmin/AndroidStudioProjects/flutter_app/build/ios/Debug-iphonesimulator/flutter_blue')
    1 error generated.
    note: Using new build systemnote: Planning buildnote: Constructing build description

ag84ark, 2019-10-21:
Try this: https://github.com/pauldemarco/flutter_blue/issues/406#issuecomment-544501458

ZhangChengmin99, 2019-10-21:
> Try this: [#406 (comment)](https://github.com/pauldemarco/flutter_blue/issues/406#issuecomment-544501458)

Thanks. The example project finally works.

pauldemarco, 2019-10-24:
Hi all, I've created a dedicated Discord channel so we can debug these problems more efficiently:

[Discord Invite Link](https://discord.gg/K8qKgX)

ZhangChengmin99, 2019-10-24:
Thank you I had solved my problems. Thank you by the same.

| |
张成敏
|
|
邮箱：18547083863@163.com
|
On 10/24/2019 20:40, Paul DeMarco wrote:

Hi all, I've created a dedicated Discord channel so we can debug these problems more efficiently:

Discord Invite Link

pauldemarco, 2019-10-24:
**Compiling flutter_blue on a Swift enabled Flutter project**

Flutter projects using Swift for the iOS platform may have trouble compiling with flutter_blue:
```
While building module 'flutter_blue' imported from
    /Users/paul/my_app/ios/Runner/GeneratedPluginRegistrant.m:10:
    In file included from <module-includes>:1:
    /Users/paul/my_app/build/ios/Debug-iphoneos/flutter_blue/flutter_blue.framework/Headers/fl
    utter_blue-umbrella.h:13:9: fatal error: 'FlutterBluePlugin.h' file not found
    #import "FlutterBluePlugin.h"
            ^~~~~~~~~~~~~~~~~~~~~
    1 error generated.
    /Users/paul/my_app/ios/Runner/GeneratedPluginRegistrant.m:10:9: fatal error: could not
    build module 'flutter_blue'
    @import flutter_blue;
     ~~~~~~~^~~~~~~~~~~~
    2 errors generated.
    note: Using new build systemnote: Planning buildnote: Constructing build description
```

From your project's root folder:

Clean everything:
```
rm -rf ios/Pods
rm ios/Podfile
rm ios/Podfile.lock
rm -rf ~/Library/Developer/Xcode/DerivedData/
flutter clean
flutter packages upgrade
```

Regenerate the Podfile with:
```
flutter run
```

Open `ios/Podfile` and comment the following line:
```
#use_frameworks!
```

The project should now run successfully with:
```
flutter run
```

closed 2019-10-24 by pauldemarco

Fjnova, 2020-11-16:
I fixed by using local library

Steps:

1. Download flutter blue version by git tag
2. Put in beside flutter project
3. Include flutter into flutter project (pubspec.yaml) with

```
use   flutter_blue:
    path: ../flutter_blue-version
```