pauldemarco/flutter_blue #410: Why it takes a long time to connect Bluetooth devices

houyongfeng1984, 2019-10-21:
await bluetoothDevice.connect();

This statement takes a long time to execute, sometimes tens of seconds

houyongfeng1984, 2019-10-22:
connectSubscription = scanResult.device.state.listen((connectState) async {
              if(connectState==BluetoothDeviceState.connected){
                var endnow = new DateTime.now();
                var end = endnow.millisecondsSinceEpoch;
                time = (end -start)~/1000;
                print('开锁时间:$time');
                List<BluetoothService> services = await scanResult.device.discoverServices();
                services.forEach((service) async {
                  if(service.uuid.toString().toLowerCase()=='0000fee7-0000-1000-8000-00805f9b34fb'){
                    var characteristics = service.characteristics;
                    writeChar = characteristics[1];
                    readChar = characteristics[0];
                    await characteristics[0].setNotifyValue(true);
                    enterOrder=0x01;
                    infoSubscription = characteristics[0].value.listen((value){
                      listenChangeData(value);
                    });//监听变化的流
                  }
                });
              }else{
                print('蓝牙状态:$connectState');
              }
            });
            await scanResult.device.connect(autoConnect: false);

After debugging, I can use the above method to connect quickly. Is this feasible

vincentsong, 2019-10-22:
I suggest you put the state listener after connect action, and use a try on to catch timeout error

houyongfeng1984, 2019-10-22:
yes!
    I use monitor to execute interactive commands when the connection is successful!Set the auto connect parameter to false, and the connection speed will be faster

like this:
scanSubscription = flutterBlue.scan(withServices: [guid],timeout: Duration(seconds: 5)
).listen((scanResult) async {
  // do something with scan result
  if(scanResult.device.name!=''){
    if(scanResult.device.name.contains("C420EA3F2C84589A")){
      bluetoothDevice = scanResult.device;
      print('${bluetoothDevice.name} found! rssi: ${scanResult.rssi}');
      flutterBlue.isScanning.listen((data){
        print(data);
      });
      flutterBlue.stopScan();
      var now = new DateTime.now();
      var start = now.millisecondsSinceEpoch;
      connectSubscription = scanResult.device.state.listen((connectState) async {
        if(connectState==BluetoothDeviceState.connected){

          List<BluetoothService&gt; services = await scanResult.device.discoverServices();
          services.forEach((service) async {
            if(service.uuid.toString().toLowerCase()=='0000fee7-0000-1000-8000-00805f9b34fb'){
              var characteristics = service.characteristics;
              writeChar = characteristics[1];
              readChar = characteristics[0];
              await characteristics[0].setNotifyValue(true);
              enterOrder=0x01;
              infoSubscription = characteristics[0].value.listen((value){
                var endnow = new DateTime.now();
                var end = endnow.millisecondsSinceEpoch;
                time = (end -start)~/1000;
                print('开锁时间:$time');
                listenChangeData(value);
              });//监听变化的流
            }
          });
        }else{
          print('蓝牙状态:$connectState');
        }
      });
      await scanResult.device.connect(autoConnect: false);
    }
  }

yiguozhen0510, 2019-11-14:
i also get this problem.can you share your project?thank you!!

LiangJinMing66, 2019-11-18:
Set autoConnect: false try

adadion, 2020-02-03:
I'm still having this problem. Sometime, it takes long time to connect.
Set autoConnect to false doesn't seem to help.

BananaHemic, 2021-01-15:
I'm also experiencing a longer-than-expected time to connect to the device. @pauldemarco Do you have any ideas what might be going on?

adadion, 2021-01-15:
Based on my testing:
iPhone 7 = instant connect (under 1000ms)
Redmi Note 4X = sometimes took ages, mostly in about 1-3 secs
Realme 3 = mostly took ages even not gonna connect
Samsung M10 = mostly in about 1-3 secs