pauldemarco/flutter_blue #414: How to Send Advertising ?

ShadowOfWhite, 2019-10-25:
I can receive the Advertisement,but how to Send Advertising ?

ericstephen, 2019-10-28:
Not sure you understand the nature of Advertisement Data or I am missing something. You don't send advertisement data. It's something you listen for. If you want to write something to a device you will need to set up services and characteristics in order to perform writes.

ShadowOfWhite, 2019-10-31:
> Not sure you understand the nature of Advertisement Data or I am missing something. You don't send advertisement data. It's something you listen for. If you want to write something to a device you will need to set up services and characteristics in order to perform writes.

Obviously,i can listen Advertisement Data ,but I also should be able to send Advertisement data.Im an Android Devolper,in the Android, call BluetoothLeAdvertiser#startAdvertising can send  advertisement data,so i wish this awesome plugin support send advertisement data,thanks.

yiguozhen0510, 2019-11-26:
How can I receive the Advertisement Data?what,s the method?

ShadowOfWhite, 2019-11-26:
@yiguozhen0510 
// Listen to scan results
var subscription = flutterBlue.scanResults.listen((scanResult) {
    // do something with scan result
    device = scanResult.device;
    print('${device.name} found! rssi: ${scanResult.rssi}');
});
scanResult is the Advertisement Data

yiguozhen0510, 2019-12-04:
Hello:
    I am grad to here your e-mail and thank you very much.The Advertisement Data for Android we can get it,but IOS.and now We find the preblem beacuse the blueDevice send data for a mistake,and blueDevice doesn't return the data by  standard format，as a result we can't get the  Advertisement Data.But angway,Thanks a lot.Good Luck!

Akilesh30, 2020-08-06:
how to broadcast my own id to other devices?

closed 2020-08-18 by ShadowOfWhite