pauldemarco/flutter_blue #415: Example program not compiling [androidx-migration]

epietrowicz, 2019-10-25:
I've followed all of the migration procedures outlined by the link in the error message and am still having issue compiling the example Android program. 

Is there a workaround? 

*******************************************************************************************
The Gradle failure may have been because of AndroidX incompatibilities in this Flutter app.
See https://goo.gl/CP92wY for more information on the problem and how to fix it.
*******************************************************************************************

ewechyetan, 2019-10-27:
I ran

git clone https://github.com/pauldemarco/flutter_blue.git
cd flutter_blue\example
then
flutter run

and I'm getting the package androidx.core.app does not exist error

Shouldn't the latest have the project migrated to androidx and run?

tiagoaleff, 2019-10-27:
I had the same problem, so I opened the "android" folder with Android Studio and on the lines that was giving problem and hit alt + enter, so Android Studio suggested putting depedence on the project, after that the project compiled, I don't know if it was the right approach, but it worked.

KevinRohn, 2019-10-28:
The issue is still open, as you can see.

borisllona, 2019-10-29:
I think we are all having the same problem, does anyone know any other example we can test just to get in touch with the funcionality of the library?

epietrowicz, 2019-10-29:
@borisllona I've been following "That Project"'s video projects of BLE Flutter development. His BLE Joypad example has been a useful place for me to start. It complies on my machine without any problems. 

https://github.com/0015/IdeasNProjects

https://www.youtube.com/watch?v=fQz7aajslJc&t=876s

borisllona, 2019-10-29:
> @borisllona I've been following "That Project"'s video projects of BLE Flutter development. His BLE Joypad example has been a useful place for me to start. It complies on my machine without any problems.
> 
> https://github.com/0015/IdeasNProjects
> 
> https://www.youtube.com/watch?v=fQz7aajslJc&t=876s

Usefull ! Thanks mate.

pauldemarco, 2019-10-30:
These compilation bugs are very dev-environment dependent, stop into the discord if you'd like some help getting up and running.

pauldemarco, 2019-10-31:
I worked with a developer for a bit today that was having this issue; the solution is to use the latest version of flutter and update your project files.

Again, the following instructions might help, but if not, join the discord for some troubleshooting:
```
flutter channel master
flutter upgrade
flutter doctor -v
```
Make sure you are using at least Flutter version 1.10.15

Navigate to your project's folder:
```
rm -rf android
flutter create .
```
Change minSdkVersion to 19 in android/app/build.gradle:
```
defaultConfig {
        ...
        minSdkVersion 19
        ...
}
```

The project should now successfully compile:
```
flutter clean
flutter build apk
```

tomcatvr, 2019-11-11:
@pauldemarco 
thanks for the instructions but using flutter channel master can be dangerous as the matter of fact not all plugins used in an application may be compatible with master channel which is not raccomanded to be uses in a production stage

jonhunterking, 2019-11-11:
I tried the solution @pauldemarco suggested on recommendation from @tomcatvr (Issue #434 ) and it would not work. I was able to change channel to master but when I tried to upgrade I got the error: 
"Because flutter_tools depends on build_runner >=1.7.2 which requires SDK version >=2.6.0-dev.8.0 <3.0.0, version solving failed.
Error (1): Unable to 'pub upgrade' flutter tool. Retrying in five seconds.."

Now when I try and run my project I get tons of errors like the following: "../../Downloads/flutter_windows_v1.9.1+hotfix.6-stable/flutter/packages/flutter/lib/src/services/text_editing.dart:13:29: Error: Type 'TextRange' not found.
class TextSelection extends TextRange {"

I tried to revert back to v1.10.16 and I get a similar error from above: 
"Because flutter_tools depends on build_runner >=1.7.2 which requires SDK version >=2.6.0-dev.8.0 <3.0.0, version solving failed.
Error (1): Unable to 'pub upgrade' flutter tool. Retrying in five seconds... (7 tries left)"

Seems like I got myself stuck. Anyone know how to get Flutter Blue working and how to get myself back into a working state?

lsaudon, 2019-12-09:
I have this error on 
- flutter_blue: ^0.6.3+1
- flutter: 1.9.1+hotfix.6

```
C:\src\flutter\.pub-cache\hosted\pub.dartlang.org\flutter_blue-0.6.3+1\android\src\main\java\com\pauldemarco\flutter_blue\FlutterBluePlugin.java:43: error: package androidx.core.app does not exist
import androidx.core.app.ActivityCompat;
                        ^
C:\src\flutter\.pub-cache\hosted\pub.dartlang.org\flutter_blue-0.6.3+1\android\src\main\java\com\pauldemarco\flutter_blue\FlutterBluePlugin.java:44: error: package androidx.core.content does not exist
import androidx.core.content.ContextCompat;
                            ^
C:\src\flutter\.pub-cache\hosted\pub.dartlang.org\flutter_blue-0.6.3+1\android\src\main\java\com\pauldemarco\flutter_blue\FlutterBluePlugin.java:150: error: cannot find symbol
                if (ContextCompat.checkSelfPermission(activity, Manifest.permission.ACCESS_COARSE_LOCATION)
                    ^
  symbol:   variable ContextCompat
  location: class FlutterBluePlugin
C:\src\flutter\.pub-cache\hosted\pub.dartlang.org\flutter_blue-0.6.3+1\android\src\main\java\com\pauldemarco\flutter_blue\FlutterBluePlugin.java:152: error: cannot find symbol
                    ActivityCompat.requestPermissions(
                    ^
  symbol:   variable ActivityCompat
  location: class FlutterBluePlugin
Note: C:\src\flutter\.pub-cache\hosted\pub.dartlang.org\flutter_blue-0.6.3+1\android\src\main\java\com\pauldemarco\flutter_blue\FlutterBluePlugin.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: C:\Users\lucas\AndroidStudioProjects\flutter_app_with_ble\build\flutter_blue\generated\source\proto\debug\javalite\com\pauldemarco\flutter_blue\Protos.java uses unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
4 errors
*******************************************************************************************
The Gradle failure may have been because of AndroidX incompatibilities in this Flutter app.
See https://goo.gl/CP92wY for more information on the problem and how to fix it.
*******************************************************************************************
Finished with error: Gradle task assembleDebug failed with exit code 1
```

ride4sun, 2020-02-13:
Latest stable and I can not build the example: 
```
Launching lib\main.dart on moto g 7 in debug mode...
[!] Your app isn't using AndroidX.
    To avoid potential build failures, you can quickly migrate your app by following the steps on https://goo.gl/CP92wY.
Running Gradle task 'assembleDebug'...
C:\Users\ride4\.gradle\caches\transforms-1\files-1.1\support-compat-28.0.0.aar\4f7ab4c6fa2922a87fc36a5d679eb841\res\values\values.xml:133:5-70: AAPT: error: resource android:attr/fontVariationSettings not found.

C:\Users\ride4\.gradle\caches\transforms-1\files-1.1\support-compat-28.0.0.aar\4f7ab4c6fa2922a87fc36a5d679eb841\res\values\values.xml:133:5-70: AAPT: error: resource android:attr/ttcIndex not found.

C:\Projects\flutter_blue\example\build\app\intermediates\incremental\mergeDebugResources\merged.dir\values\values.xml:84: error: resource android:attr/fontVariationSettings not found.
C:\Projects\flutter_blue\example\build\app\intermediates\incremental\mergeDebugResources\merged.dir\values\values.xml:84: error: resource android:attr/ttcIndex not found.
error: failed linking references.

Failed to execute aapt
com.android.ide.common.process.ProcessException: Failed to execute aapt
	at com.android.builder.core.AndroidBuilder.processResources(AndroidBuilder.java:796)
	at com.android.build.gradle.tasks.ProcessAndroidResources.invokeAaptForSplit(ProcessAndroidResources.java:551)
	at com.android.build.gradle.tasks.ProcessAndroidResources.doFullTaskAction(ProcessAndroidResources.java:285)
	at com.android.build.gradle.internal.tasks.IncrementalTask.taskAction(IncrementalTask.java:109)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.gradle.internal.reflect.JavaMethod.invoke(JavaMethod.java:73)
	at org.gradle.api.internal.project.taskfactory.IncrementalTaskAction.doExecute(IncrementalTaskAction.java:50)
	at org.gradle.api.internal.project.taskfactory.StandardTaskAction.execute(StandardTaskAction.java:39)
	at org.gradle.api.internal.project.taskfactory.StandardTaskAction.execute(StandardTaskAction.java:26)
	at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter$1.run(ExecuteActionsTaskExecuter.java:131)
	at org.gradle.internal.operations.DefaultBuildOperationExecutor$RunnableBuildOperationWorker.execute(DefaultBuildOperationExecutor.java:300)
	at org.gradle.internal.operations.DefaultBuildOperationExecutor$RunnableBuildOperationWorker.execute(DefaultBuildOperationExecutor.java:292)
	at org.gradle.internal.operations.DefaultBuildOperationExecutor.execute(DefaultBuildOperationExecutor.java:174)
	at org.gradle.internal.operations.DefaultBuildOperationExecutor.run(DefaultBuildOperationExecutor.java:90)
	at org.gradle.internal.operations.DelegatingBuildOperationExecutor.run(DelegatingBuildOperationExecutor.java:31)
	at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter.executeAction(ExecuteActionsTaskExecuter.java:120)
	at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter.executeActions(ExecuteActionsTaskExecuter.java:99)
	at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter.execute(ExecuteActionsTaskExecuter.java:77)
	at org.gradle.api.internal.tasks.execution.OutputDirectoryCreatingTaskExecuter.execute(OutputDirectoryCreatingTaskExecuter.java:51)
	at org.gradle.api.internal.tasks.execution.SkipUpToDateTaskExecuter.execute(SkipUpToDateTaskExecuter.java:59)
	at org.gradle.api.internal.tasks.execution.ResolveTaskOutputCachingStateExecuter.execute(ResolveTaskOutputCachingStateExecuter.java:54)
	at org.gradle.api.internal.tasks.execution.ValidatingTaskExecuter.execute(ValidatingTaskExecuter.java:59)
	at org.gradle.api.internal.tasks.execution.SkipEmptySourceFilesTaskExecuter.execute(SkipEmptySourceFilesTaskExecuter.java:101)
	at org.gradle.api.internal.tasks.execution.FinalizeInputFilePropertiesTaskExecuter.execute(FinalizeInputFilePropertiesTaskExecuter.java:44)
	at org.gradle.api.internal.tasks.execution.CleanupStaleOutputsExecuter.execute(CleanupStaleOutputsExecuter.java:91)
	at org.gradle.api.internal.tasks.execution.ResolveTaskArtifactStateTaskExecuter.execute(ResolveTaskArtifactStateTaskExecuter.java:62)
	at org.gradle.api.internal.tasks.execution.SkipTaskWithNoActionsExecuter.execute(SkipTaskWithNoActionsExecuter.java:59)
	at org.gradle.api.internal.tasks.execution.SkipOnlyIfTaskExecuter.execute(SkipOnlyIfTaskExecuter.java:54)
	at org.gradle.api.internal.tasks.execution.ExecuteAtMostOnceTaskExecuter.execute(ExecuteAtMostOnceTaskExecuter.java:43)
	at org.gradle.api.internal.tasks.execution.CatchExceptionTaskExecuter.execute(CatchExceptionTaskExecuter.java:34)
	at org.gradle.api.internal.tasks.execution.EventFiringTaskExecuter$1.run(EventFiringTaskExecuter.java:51)
	at org.gradle.internal.operations.DefaultBuildOperationExecutor$RunnableBuildOperationWorker.execute(DefaultBuildOperationExecutor.java:300)
	at org.gradle.internal.operations.DefaultBuildOperationExecutor$RunnableBuildOperationWorker.execute(DefaultBuildOperationExecutor.java:292)
	at org.gradle.internal.operations.DefaultBuildOperationExecutor.execute(DefaultBuildOperationExecutor.java:174)
	at org.gradle.internal.operations.DefaultBuildOperationExecutor.run(DefaultBuildOperationExecutor.java:90)
[... cut ...]
Execution failed for task ':app:processDebugResources'.
> Failed to execute aapt

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 29s
AndroidX incompatibilities may have caused this build to fail. Please migrate your app to AndroidX. See https://goo.gl/CP92wY.
Finished with error: Gradle task assembleDebug failed with exit code 1
```

pauldemarco, 2020-04-02:
Flutter blue will now be pinned to the latest stable release of flutter.
If your project’s template files are up to date, there should be no issues when building for Android or iOS.

Please reopen this issue if that’s not the case.

closed 2020-04-02 by pauldemarco