pauldemarco/flutter_blue #418: Error importing androidx.core in FlutterBluePlugin.java

borisllona, 2019-10-30:
:\Users\boris\AppData\Roaming\Pub\Cache\hosted\pub.dartlang.org\flutter_blue-0.6.3+1\android\src\main\java\com\pauldemarco\flutter_blue\FlutterBluePlugin.java:43: error: package androidx.core.app does not exist
import androidx.core.app.ActivityCompat;
                        ^
C:\Users\boris\AppData\Roaming\Pub\Cache\hosted\pub.dartlang.org\flutter_blue-0.6.3+1\android\src\main\java\com\pauldemarco\flutter_blue\FlutterBluePlugin.java:44: error: package androidx.core.content does not exist
import androidx.core.content.ContextCompat;

I have this error and I've already imported android x and refractored. What should I do? I've tested with diferent versions of flutter ble too.

rjcasson, 2019-10-30:
@borisllona,  check out my comment here:  https://github.com/pauldemarco/flutter_blue/issues/402#issuecomment-546425715
Or my pull request here: #416

bus710, 2019-10-30:
Also, staying at 0.6.2 is not a bad idea (in pubspeck.yaml).
I had to delete the .flutter0plugins and pubspec.lock before **flutter pub get**.

It works for me but I am looking forward the next version if it comes with the solution of @rjcasson

borisllona, 2019-10-30:
> @borisllona, check out my comment here: [#402 (comment)](https://github.com/pauldemarco/flutter_blue/issues/402#issuecomment-546425715)
> Or my pull request here: #416

It worked ! I was starting going crazy and updating my own app gradle but I haven't realized that it has to be changed the library one.
Thanks so much!

closed 2019-10-30 by borisllona

pauldemarco, 2019-10-30:
@borisllona Chances are the problem is with outdated project files. Stop into the discord if you'd like some help getting things up and running.

pauldemarco, 2019-10-31:
Please follow the instructions here and let me know the results:
https://github.com/pauldemarco/flutter_blue/issues/415#issuecomment-548186492