pauldemarco/flutter_blue #422: Conect to specific Service and Charact

saulopef, 2019-10-31:
I need to conect to a specific service and charact in my product, a tried several ways to do this, but all tima a had some error, I don't want to make my user to chose the service and the charact manualy like the example code, I want to, for example:
Concect to an HR strap, in the 180D service, and 2A37 charact, and set notiy in this charact.
I tried, map the services of the device and compare if:
`s.uuid.toString().toUpperCase().substring(4, 8) ==myService`
but i always get a null value.
whats if the proper way to do this conection automaticaly?

bus710, 2019-11-01:
This has some overhead though, you may utilize it as your needs:

```
_bleData.targetChar = null;
    for (int i = 0; i < _bleData.servicesFound.length; i++) {
      for (BluetoothCharacteristic c
          in _bleData.servicesFound[i].characteristics) {
        try {
          await Future.delayed(Duration(milliseconds: 10));
          if (c.properties.read && c.uuid.toString().contains(targetUuid)) {
            _bleData.targetChar = c;
          }
        } catch (e) {
          debugPrint("Error!: " + e.toString());
          _bleActivityState = BLEActivityStatus.init;
          _doneHandler(Request.failed);
          return;
        }
      }
```

saulopef, 2019-11-02:
What is this "_bleData" object? How to access it?

saulopef, 2019-11-02:
Oh! my fault, I think I understand your example, I will try apply it.

bus710, 2019-11-02:
@saulopef,
Sorry for my laziness about the code.
**_bleCore** is a class as a layer to track down the state/data of flutter_blue object.
**targetUuid** is the known information if you already have a BLE peripheral that you designed.
The for loop doesn't need to iterate every time once you found the target char.
Please let me know if you have additional question or so.

saulopef, 2019-11-03:
hey, thanks for this, I tried to adapt to my actual code and result this:

`for (int i = 0; i < snapshot.data.length; i++) {
                                  if (snapshot.data[i].uuid
                                      .toString()
                                      .toUpperCase()
                                      .substring(4, 8)
                                      .contains(myService)) {
                                    for (BluetoothCharacteristic c
                                        in snapshot.data[i].characteristics) {
                                      try {
                                        if (c.properties.notify &&
                                            c.uuid
                                                .toString()
                                                .toUpperCase()
                                                .substring(4, 8)
                                                .contains(myCharact)) {
                                          AppPrefereces.charact = c;
                                          AppPrefereces.charact
                                              .setNotifyValue(true);
                                        }
                                      } catch (e) {
                                        debugPrint("Error!: " + e.toString());
                                      }
                                    }
                                  }
                                }`

and it's work very well.
but tell me more about this _bleCore, it seem to be the correct form to trate states and data.

bus710, 2019-11-03:
@saulopef,

Please check this out and suggest fixes if you want:
- https://gist.github.com/bus710/0bf083093dafa6cd8c3bf6f955a1a012

Also, please feel free to close this ticket if you want :sunglasses:

saulopef, 2019-11-03:
Oh nice, tks for your time, I will study your code.

closed 2019-11-03 by saulopef