pauldemarco/flutter_blue #423: Documentation faulty/unclear?

ferdiamg, 2019-11-01:
```
void _scanBLE() {
    FlutterBlue flutterBlue = FlutterBlue.instance;
    // Start scanning
    flutterBlue.startScan(timeout: Duration(seconds: 4));

    // Listen to scan results
    var subscription = flutterBlue.scanResults.listen((scanResult) {
      // do something with scan result
      device = scanResult.device;
      print('${device.name} found! rssi: ${scanResult.rssi}');
    });

    // Stop scanning
    flutterBlue.stopScan();
  }
```

Just straight up copy/pasting from the documentation gives several issues:
- variable device is not defined (should be `var device`)
- the getter 'device'/'.rssi' isn't defined for the class 'List<ScanResult>'.

Whats up?
Changing it to the version of the previous documentation it works:
```
// Listen to scan results
    var subscription = flutterBlue.scan().listen((scanResult) {
      // do something with scan result
      var device = scanResult.device;
      print('${device.name} found! rssi: ${scanResult.rssi}');
    });

```

ekuleshov, 2019-11-01:
Also, while scan is running you can start another scan and there isn't an easy way to check if another scan is in progress (e.g. started by a different screen in the app).

bus710, 2019-11-01:
To prevent multiple requests, I just made a bool flag to avoid it. 
When there is a request from upper layer, the ble layer checks the flag to earn the control.

I ended up meeting some issues with the service discovery since it hangs sometimes so that I used timeout for the await to quit the API.

ekuleshov, 2019-11-01:
If I'm not mistaken FlutterBlue plugin has a similar flag internally and it tosses an error when this flag is set.

I'm struggling with accessing that flag from multiple different screens. If someone could point me to some kind of example, I'd greatly appreciate that.

bus710, 2019-11-01:
It would be nice if we can read it if the underlying module/hardware is being used by a previous request or not. **Stream<bool> get isScanning => _isScanning.stream;** is the only thing that I can find as exposed to us. I made couple enums and state machines to track down the state/process/request. Since my app is pretty simple (1:1), I am okay for now.